dresden-elektronik/deconz-rest-plugin #7756: Third Reality 3RDS17BZ Door/Window Contact

Heiko1977, 2024-05-10:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Third Reality, Inc

### Manufacturer

Third Reality, Inc

### Model identifier

3RDS17BZ

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]
[image]
[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration
[image]

### On/Off
[image]
[image]

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Heiko1977, 2024-05-10:
Perfect, works fine.

Thank you very much !!!

Smanar, 2024-05-10:
No issue ?
Do you want to make the PR to submit the device or want I make it ?

Heiko1977, 2024-05-10:
I’ve added 4 of them into deconz. Phoscon added them without a problem. States are fine.

if you like, you can submit it. 

Thanks for your work.

Smanar, 2024-05-11:
>if you like, you can submit it.

I prefer, to have it under you name, I can explain how to do it if needed  ...

Heiko1977, 2024-05-11:
> > if you like, you can submit it.
> 
> I prefer, to have it under you name, I can explain how to do it if needed ...

That would be very nice. I’ve never done it before.

Smanar, 2024-05-12:
Easy:
- go on this page https://github.com/dresden-elektronik/deconz-rest-plugin/tree/master/devices/third_reality
- select "add file" (as name can use 3RDS17BZ_door_sensor.json
- The magic will happen, github will create a fork on your name, and switch to editor mode.
- Copy/paste the file, and make chnage you want.
- Save it
- Github will purpose you to make a PR automatically (but I m not sure, on my side I m using an application), else on your fork (the adress will be https://github.com/Heiko1977/deconz-rest-plugin) select "pull request" on follow github.

You PR will appear here https://github.com/dresden-elektronik/deconz-rest-plugin/issues

After that all change you will make on your fork, will be synchronised on this page, so if there is problem, no need to restart all.

Edit:
Have just see this page https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-create-a-pull-request-to-submit-a-new-Device-Description-File

closed 2024-06-10 by github-actions[bot]