dresden-elektronik/deconz-rest-plugin #7761: GLEDOPTO GL-C-008P output W doesn't work

joubbi, 2024-05-13:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

I just got two GLEDOPTO GL-C-008P's.

The controller has outputs marked W, C, B, G, R and V+.
According to the instructions, a single color LED is to be connected to W and V+.

I connected my LED strip to V+ and W on the controller. The LED turned on as soon as I connected power to the controller.
I then paired the controller in the Phoscon app.
As soon as the pairing is done, the LED turns off.
I am not able to turn it back on.

I can see the sliders for Brightness, Color and Saturation in the app.
Nothing happens when I move the sliders.

I measured the outputs of the controller with a multimeter.
I concluded that RGB works but not W.
There is no power on W no matter how I slide the sliders.
So I connected the LED strip to V+ and R( or B or G), which works.

Exact same behavior with the other controller.

Before buying them I had read that they should wok with DECONZ.
But all the reports were for LEDs with several colors.
I have a LED strip with only white.

### Steps to reproduce the behavior

1. Connect a single color LED strip to GLEDOPTO GL-C-008P to V+ and W. (LED turns on)
2. Go to lights in Phoscon app and pair.. (LED turns off)

### Expected behavior

I expect to be able to control the light.

### Environment

- Host system: PC
- Running method: Raspbian
- Firmware version: 26400500
- deCONZ version (not Home assistant Addon version!): 2.26.3
- Device: ConBee I
- Do you use an USB extension cable: no
- Is there any other USB or serial devices connected to the host system? No

ebaauw, 2024-06-04:
What does the controller look like in deCONZ?  Does it have one or two endpoints?  What is the source of your instructions?

GLEDOPTO have controllers with separate endpoints for W vs RGB; they can be used to control separate lightstrips.  deCONZ exposes these as two `lights` resources, one _Color Light_ and one _Dimmable Light_.  

I don’t think I’ve seen a five-channel controller with two endpoints.  In that case the white endpoint should be a _Color Temperature Light_.  With a single endpoint, it should be an _Extended Color Light_.  To drive the W (warm white) channel, you need to set the light’s `ct` to a high value (500).  Alternatively, try the C (cool white) channel with a low `ct` value (153).

joubbi, 2024-06-04:
The controller looks like one endpoint in deCONZ. I can control brightness, color, saturation and temperature.

I received a piece of paper with instructions together with the controller.
I can't find the exact same paper online. But this one has the same picture on page 2:
https://fccid.io/2AR8F-GL-C-008/User-Manual/user-manual-4146503
The picture marked P5 is the way I should connect it according to my understanding, V+ and W.

I have it connected to V+ and R now. It works. The R output is right next to V+ while W is furthest away. So it's actually better to connect the cables to V+ and R. So it's all good.

closed 2024-07-03 by github-actions[bot]