dresden-elektronik/deconz-rest-plugin #7775: Ikea Starkvind (air purifier) not providing the status thru REST-API after DECONZ update to v2.27.0-beta

kimosak, 2024-05-16:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

With DECONZ v2.27.0-beta airputifier stopped providing states thru API - there is only lastupdated, the rest is missing (deviceruntime, filterruntime, lastupdated, replacefilter, speed..)

Actual API output:
{"config":{"on":true,"reachable":true},"ep":1,"etag":"4a8fa5a53c922c754d4b69e3a23ce16a","lastannounced":"2024-04-01T15:30:16Z","lastseen":"2024-05-16T12:51Z","manufacturername":"IKEA of Sweden","modelid":"STARKVIND Air purifier table","name":"Cisticka vzduchu Ikea","state":{"lastupdated":"2024-05-14T18:23:08.819"},"swversion":"1.1.001","type":"ZHAAirPurifier","uniqueid":"50:32:5f:ff:fe:b4:ab:f8-01-fc7d"}

DECONZ attributes
[image]

### Steps to reproduce the behavior

It appears from latest DECONZ update

### Expected behavior

Fix the REST API for Ikea Air Purifier

### Environment

- Host system: (Raspberry Pi3b
- Running method: Raspbian
- Firmware version: (26780700)
- deCONZ version (not Home assistant Addon version!): (v2.27.0-beta)
- Device: RaspBee II

kimosak, 2024-05-16:
Not 100% sure (2.25.2?), is there any way how to check it?

Mimiix, 2024-05-16:
In that case,

please check:
https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7246

kimosak, 2024-05-16:
Will do, thank you very much :)

kimosak, 2024-05-17:
Have I missed something? I checked #7246, but I am none the wiser. I’ve tried to update the DDF for the STARKVIND Air purifier from the deCONZ Toolbox, but there was no change. Also, there are no items under the DDF editor in deCONZ
[image]

ebaauw, 2024-05-17:
Looks like the DDF wasn't applied?

kimosak, 2024-05-19:
Basic mode and Gold status in Control panel, should I try anything else?

Smanar, 2024-05-19:
Do the node have a little "DDF" displayed on titlebar ?
The "gold" for status in the editor is from you ?

How have you updated the DDF ?

kimosak, 2024-05-19:
No "DDF" icon on title bar
[image]

I've tried to update DDF thru Phoscon web app but no luck, still missing the states from device
[image]

Can I solve it by re-pairing of the device in deCONZ?

Zehir, 2024-05-19:
The Device integration page use DDF bundle, not the DDF json file.

Could you try with this DDF Bundle ?

https://ddf.cryonet.io/bundle/download/ada38fd25cbc1fc9b4da0b86e8af1b4ed659f1074470916bbd9ea6288ef448fc

kimosak, 2024-05-19:
I did it like first troubleshoot step and the Device integration report - Import successfull.  However, there was no change. The states are still not being reported through the RestAPI.

Zehir, 2024-05-19:
Any updates on the deconz gui ?

kimosak, 2024-05-19:
No change in GUI and DDF editor is still empty
[image]

Zehir, 2024-05-19:
No DDF or DDB on the node ?

kimosak, 2024-05-19:
What is DDB? I see onlu this on the node 
[image]
and the DDF Editor is empty, no content/device here

Zehir, 2024-05-19:
DDB is a DDF bundle, the new feature in 2.27.0, a new way to import DDF in a single file.

It's seems that is there something wrong in your installation.
Did you try to install the last stable version ?

kimosak, 2024-05-19:
Thank you @Zehir , downgrading deconz from 2.27.0-beta to 2.26.3 helped. Now I see the DDF and device reports the states thru rest-API. I will stay on this stable version now.

Zehir, 2024-05-19:
Yes the 2.27 need more love to work perfectly, like everything, works in the lab but not in production ;)

Smanar, 2024-05-20:
So there is something in Version 0.27 that prevent deconz use a DDF bundle AND json ?

manup, 2024-05-31:
Hi had anyone the chance to test if this is fixed with version v2.27.2-beta?

kimosak, 2024-06-04:
No, I am still on stable v2.26.3

kimosak, 2024-06-04:
Yes and it is still same, not able to controll device thru API like in 2.26.3

manup, 2024-06-11:
Does the device show the DDF label on the node in the deCONZ UI?

kimosak, 2024-06-11:
Yes, DDF is visible in deCONZ UI

manup, 2024-06-12:
Here in my test setup with v2.27.3-beta the device can be controlled via API and also reports changes done on the device back to the API.

```json
{
  "config": {
    "filterlifetime": 259200,
    "ledindication": true,
    "locked": false,
    "mode": "speed_2",
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "3f28aadd46c9bfed7234135ea277ed6b",
  "lastannounced": null,
  "lastseen": "2024-06-12T14:33Z",
  "manufacturername": "IKEA of Sweden",
  "modelid": "STARKVIND Air purifier",
  "name": "AirPurifier 343",
  "productid": "E2007",
  "state": {
    "deviceruntime": 37944,
    "filterruntime": 37944,
    "lastupdated": "2024-06-12T14:34:00.606",
    "replacefilter": false,
    "speed": 40
  },
  "swversion": "1.0.033",
  "type": "ZHAAirPurifier",
  "uniqueid": "0c:43:14:ff:fe:6e:65:55-01-fc7d"
}
```

Now setting a different speed:

```
curl -X PUT -d '{"mode":"speed_4"}' 127.0.0.1:8090/api/12345/sensors/343/config
```

```json
[{"success":{"/sensors/343/config/mode":"speed_4"}}]
```

Which also reports back as speed_4 and the damn thing is loud as hell:

```json
{
  "config": {
    "filterlifetime": 259200,
    "ledindication": true,
    "locked": false,
    "mode": "speed_4",
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "bd2f952d339277ac82fcbdbd8229c49c",
  "lastannounced": null,
  "lastseen": "2024-06-12T14:35Z",
  "manufacturername": "IKEA of Sweden",
  "modelid": "STARKVIND Air purifier",
  "name": "AirPurifier 343",
  "productid": "E2007",
  "state": {
    "deviceruntime": 37946,
    "filterruntime": 37946,
    "lastupdated": "2024-06-12T14:35:44.965",
    "replacefilter": false,
    "speed": 80
  },
  "swversion": "1.0.033",
  "type": "ZHAAirPurifier",
  "uniqueid": "0c:43:14:ff:fe:6e:65:55-01-fc7d"
}
```

Can you please share how your API output looks like when doing this?

manup, 2024-06-19:
Hi could anybody with the issue retest with v2.27.4-beta?

manup, 2024-06-28:
Closing for now please, please re-open if the issue can still be seen with v2.27.4-beta and up.

closed 2024-06-28 by manup