dresden-elektronik/deconz-rest-plugin #7776: Lost device after updating to 2.27

molnarti, 2024-05-16:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

My [Ikea Trafdri On/Off switch ](https://dresden-elektronik.github.io/deconz-rest-doc/devices/ikea/tradfri_on_off_switch_old_fw/)  has disappeared from Phoscon since updating to 2.27 (from 2.26.2). The device is not visible from Phoscon or Deconz and also cannot be re-paired. Home Assistant does not detect key presses for the button.

The device reappeares when downgrading back to 2.26.2 

### Steps to reproduce the behavior

1. Update Deconz to 2.27
2. Check list of switches in Phoscon
3. Ikea Tradfri On/Off switch is missing
4. Try to re-pair device by following the instructions provided by Deconz
5. Device is not detected.

### Expected behavior

Device should be visible after update.

### Environment

- Host system: Debian Linux
- Running method:  deconz-docker container 
- Firmware version: 26720700
- deCONZ version: 2.27
- Device: ConBee II
- Do you use an USB extension cable: no
- Is there any other USB or serial devices connected to the host system? If so: UPS

Zehir, 2024-05-17:
I just tried to add one on my network and it's seems to work fine.
Did you got a red light on the switch when you press the button 4 times ?
Any related logs in deconz ?

molnarti, 2024-05-17:
yes, there was a red light blinking, everything looked like the pairing should succeed. I was just thinking about replacing the battery, but had no spares at home, so just tried reverting to the previous version and suddenly it worked. in general it seems to me that I have issues pairing new devices to my Conbee lately. maybe its related to the fact that my conbee stick is in the basement and there is only one device connected to it directly, the rest goes to the mesh network.

MarioMarti77, 2024-05-17:
I have the same behavior than molnarti with Ikea TRADFRI on/off switches, downgrade to 2.26..2 show them again

Zehir, 2024-05-17:
What frimware version of the device do you have ?
You can download the latest at http://fw.ota.homesmart.ikea.net/global/GW1.0/01.21.031/bin/10005778-tradfri_onoff_controller-24.4.6-prod.ota.ota.signed
And then you can install them using the OTA update plugin. Takes about 2-3 hours per switch (can run multiple upgrades at once)

manup, 2024-05-21:
The On/Off switch has two DDFs which match to different firmware versions. This was introduced in v2.26.1-beta via PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7643

After the update it may take a while until the firmware version is received from the switch (the device should send this periodically to deCONZ), after that the correct DDF should be selected.

MaxVRAM, 2024-05-27:
I've had the same issue. Updating to `2.27` caused the IKEA switches completely disappear from the Deconz interface and would not pair again.

I didn't catch a "breaking changes" in the release notes. Is this expected behaviour?

Zehir, 2024-05-27:
> I've had the same issue. Updating to `2.27` caused the IKEA switches completely disappear from the Deconz interface and would not pair again.
> 
> I didn't catch a "breaking changes" in the release notes. Is this expected behaviour?

Are you on linux ?
It's not an expected behavior and will be fixed soonish

molnarti, 2024-05-27:
> What frimware version of the device do you have ? You can download the latest at [...]

updating to the linked fw solved the issue for me. but i have another switch still in the box, I am curious how it will pair still with the old fw

MaxVRAM, 2024-05-27:
> Are you on linux ?
> It's not an expected behavior and will be fixed soonish

Thanks for the clarification. I don't know what the OTA plugin is or if it's a known part of the update process, but figured I'd just roll back until I had time to investigate next week.

Yes, I'm running the docker image on Fedora 39 x86-64.