dresden-elektronik/deconz-rest-plugin #7777: Selected lines in OTA Update became blank

Zehir, 2024-05-17:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

In the OTA Update plugin, any device selected line became blank

### Steps to reproduce the behavior

Open the OTA Update tab and select a device

### Expected behavior

Expecting to have the line in blue but with the text

### Screenshots

Not selected;
[image]

Selected;
[image]

### Environment

- Host system: PC
- Running method: Windows
- Firmware version: (26720700)
- deCONZ version (not Home assistant Addon version!): (2.27.0)
- Device: ConBee II
- Do you use an USB extension cable: yes 
- Is there any other USB or serial devices connected to the host system? If so: Which?

ebaauw, 2024-05-17:
Not seeing that on Raspberry Pi.

I think this would be a bug in the OTAU plugin and/or the core; not in the REST API plugin.

Zehir, 2024-05-17:
> Not seeing that on Raspberry Pi.
> 
> I think this would be a bug in the OTAU plugin and/or the core; not in the REST API plugin.

there is an other repo for the OTAU plugin ?

ebaauw, 2024-05-17:
https://github.com/dresden-elektronik/deconz-ota-plugin
See also (deCONZ for Dummies)[https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-for-Dummies]

Zehir, 2024-05-18:
Thanks @ebaauw , I made an issue on the other repo;
https://github.com/dresden-elektronik/deconz-ota-plugin/issues/51

closed 2024-05-18 by Zehir