dresden-elektronik/deconz-rest-plugin #7780: DDF Policy `raw_json` still display an `ddf_hash`

Zehir, 2024-05-18:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

When setting the `ddf_policy` to `raw_json` it's look like a bundle is still used.

[image]
```json
{
    "ddf_hash": "a1001854b9854ad3c6fbaa4b5c5028b3cff97dbf3dbd53f357221b9c06c6cd38",
    "ddf_policy": "raw_json",
    "lastannounced": null,
    "lastseen": "2024-05-18T08:52Z",
    "manufacturername": "IKEA of Sweden",
    "modelid": "TRADFRI on/off switch",
    "name": "TRADFRI on/off switch (2)"
}
```

### Steps to reproduce the behavior

Set a device policy to `raw_json`

### Expected behavior

I expect the device to have the `DDF` tag on the GUI and no `ddf_hash` on the `/api/:apiKey/devices/:deviceUniqueID` endpoint

### Environment

- Host system: PC
- Running method: Windows
- Firmware version: 26720700
- deCONZ version (not Home assistant Addon version!): 2.27.0
- Device: ConBee II
- Do you use an USB extension cable: yes
- Is there any other USB or serial devices connected to the host system? If so: Which?

Zehir, 2024-05-18:
May be related to #7776

Zehir, 2024-06-16:
@manup is this one fixed in latest beta ?

manup, 2024-07-09:
Just tested with v2.27.4-beta  here it works fine.
Tested with Ikea signal Repeater and Tradfri On/Off Switch.

-----

[image]

`GET http://skull:8090/api/12345/devices/cc:86:ec:ff:fe:1b:2e:6b`

```
{
	"ddf_hash": "ba9f2028165e5db43a4d9831134cf895198b198df7831a5a33c7b5332b3f1fa4",
	"ddf_policy": "latest_prefer_stable",
	"lastannounced": "2024-07-09T08:41:42Z",
	"lastseen": "2024-07-09T08:41Z",
	"manufacturername": "IKEA of Sweden",
	"modelid": "TRADFRI Signal Repeater",
	"name": "Range extender 44",
	"productid": "TRADFRI Signal Repeater",
	"subdevices": [
		{
			"productid": "E1746",
			"state": {
				"alert": {
					"lastupdated": "2023-11-06T13:33:52Z",
					"value": "none"
				},
				"reachable": {
					"lastupdated": "2024-07-09T10:41:42Z",
					"value": true
				}
			},
			"type": "Range extender",
			"uniqueid": "cc:86:ec:ff:fe:1b:2e:6b-01"
		}
	],
	"swversion": "2.3.086",
	"uniqueid": "cc:86:ec:ff:fe:1b:2e:6b"
}
```

`PUT http://skull:8090/api/12345/devices/cc:86:ec:ff:fe:1b:2e:6b/ddf/policy`

```
{"policy": "raw_json"}
```

```
[{
  "success": {
    "/devices/cc:86:ec:ff:fe:1b:2e:6b/ddf/policy": "raw_json"
  }
}]
```

[image]

```
{
	"ddf_policy": "raw_json",
	"lastannounced": "2024-07-09T08:41:42Z",
	"lastseen": "2024-07-09T08:50Z",
	"manufacturername": "IKEA of Sweden",
	"modelid": "TRADFRI Signal Repeater",
	"name": "Range extender 44",
	"productid": "TRADFRI Signal Repeater",
	"subdevices": [
		{
			"productid": "E1746",
			"state": {
				"alert": {
					"lastupdated": "2023-11-06T13:33:52Z",
					"value": "none"
				},
				"reachable": {
					"lastupdated": "2024-07-09T10:41:42Z",
					"value": true
				}
			},
			"type": "Range extender",
			"uniqueid": "cc:86:ec:ff:fe:1b:2e:6b-01"
		}
	],
	"swversion": "2.3.086",
	"uniqueid": "cc:86:ec:ff:fe:1b:2e:6b"
}
```

-----

Note the hash is not shown when `raw_json` is set. Not shown here but I also made some tests switching back an forth with various modes `raw_json`, `latest_prefer_stable` and `pin`.

Zehir, 2024-07-09:
Ok thanks

manup, 2024-07-09:
But please check with your setup too, sometimes stuff only works in mine :smile:

Zehir, 2024-07-09:
No problem I add a reminder for this week-end

Zehir, 2024-07-14:
@manup I confirm it's working fine