dresden-elektronik/deconz-rest-plugin #7785: Philips Hue Wall Switch does not work since update to v2.27.0

Quatalspropella, 2024-05-22:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

After Update Deconz Gateway Version to 2.27.0 (use Raspberry Pi OS) the Hue Wall Switch does not work. I see the devices as activ but can not switch the light. The light working in the web gui. So i went back to 2.26.3 and the switches  working again. 

Hersteller: Signify Netherlands B.V.
Produkt: Wandschaltermodul
Version: 1.0.5
Batterie: 100
Konfiguration: Taster / eine Wippe

### Steps to reproduce the behavior

Update to 2.27.0

### Expected behavior

Update to 2.27.0

### Environment

- Host system: (Raspberry Pi on Proxmox VM)
- Running method: (Raspbian / Home Assistent deCONZ Add-on / Virtual Machine)
- Firmware version: (26780700)
- deCONZ version (not Home assistant Addon version!): (2.27.00)
- Device: ( ConBee II)
- Is there any other USB or serial devices connected to the host system? If so: Which?

manup, 2024-05-23:
Hi there seems to be a problem in some setups loading DDFs, which I can't yet reproduce in my setup.
Can you please start deCONZ on the command line with following parameters:

```
deCONZ --dbg-ddf=1 --dbg-info=1
```

Then copy the debug log from **deCONZ &rarr; Help &rarr; Debug view** and paste it here.

olemr, 2024-05-23:
I too spent some hours thinking my HUE dimmer switch was broken .. :-)
Here is the startup log:

```
22:56:47:266 sub-device: 00:15:8d:00:01:02:50:dc-03-000c, has item: attr/id
22:56:47:266 sub-device: 00:15:8d:00:01:02:50:dc-03-000c, has item: attr/lastannounced
22:56:47:266 sub-device: 00:15:8d:00:01:02:50:dc-03-000c, has item: attr/lastseen
22:56:47:267 sub-device: 00:15:8d:00:01:02:50:dc-03-000c, has item: attr/manufacturername
22:56:47:267 sub-device: 00:15:8d:00:01:02:50:dc-03-000c, has item: attr/modelid
22:56:47:267 sub-device: 00:15:8d:00:01:02:50:dc-03-000c, has item: attr/name
22:56:47:267 sub-device: 00:15:8d:00:01:02:50:dc-03-000c, has item: attr/swversion
22:56:47:267 sub-device: 00:15:8d:00:01:02:50:dc-03-000c, has item: attr/type
22:56:47:268 sub-device: 00:15:8d:00:01:02:50:dc-03-000c, has item: attr/uniqueid
22:56:47:268 sub-device: 00:15:8d:00:01:02:50:dc-03-000c, create item: config/battery
22:56:47:268 sub-device: 00:15:8d:00:01:02:50:dc-03-000c, has item: config/on
22:56:47:268 sub-device: 00:15:8d:00:01:02:50:dc-03-000c, has item: config/reachable
22:56:47:269 sub-device: 00:15:8d:00:01:02:50:dc-03-000c, create item: state/gesture
22:56:47:269 sub-device: 00:15:8d:00:01:02:50:dc-03-000c, create item: state/buttonevent
22:56:47:270 sub-device: 00:15:8d:00:01:02:50:dc-03-000c, has item: state/lastupdated
22:56:47:272 sub-device: 00:15:8d:00:01:02:50:dc-02-0012, has item: attr/id
22:56:47:272 sub-device: 00:15:8d:00:01:02:50:dc-02-0012, has item: attr/lastannounced
22:56:47:273 sub-device: 00:15:8d:00:01:02:50:dc-02-0012, has item: attr/lastseen
22:56:47:273 sub-device: 00:15:8d:00:01:02:50:dc-02-0012, has item: attr/manufacturername
22:56:47:273 sub-device: 00:15:8d:00:01:02:50:dc-02-0012, has item: attr/modelid
22:56:47:273 sub-device: 00:15:8d:00:01:02:50:dc-02-0012, has item: attr/name
22:56:47:274 sub-device: 00:15:8d:00:01:02:50:dc-02-0012, has item: attr/swversion
22:56:47:274 sub-device: 00:15:8d:00:01:02:50:dc-02-0012, has item: attr/type
22:56:47:274 sub-device: 00:15:8d:00:01:02:50:dc-02-0012, has item: attr/uniqueid
22:56:47:274 sub-device: 00:15:8d:00:01:02:50:dc-02-0012, create item: config/battery
22:56:47:274 sub-device: 00:15:8d:00:01:02:50:dc-02-0012, has item: config/on
22:56:47:275 sub-device: 00:15:8d:00:01:02:50:dc-02-0012, has item: config/reachable
22:56:47:275 sub-device: 00:15:8d:00:01:02:50:dc-02-0012, create item: state/gesture
22:56:47:275 sub-device: 00:15:8d:00:01:02:50:dc-02-0012, create item: state/buttonevent
22:56:47:275 sub-device: 00:15:8d:00:01:02:50:dc-02-0012, has item: state/lastupdated
22:56:47:276 DEV found DDF for 0x00158D00010250DC, path: /usr/share/deCONZ/devices/xiaomi/xiaomi_mfkzq01lm_cube.json
22:56:47:285 dlg action: Read binding table
22:56:47:382 sub-device: , has item: attr/id
22:56:47:383 sub-device: , has item: attr/lastannounced
22:56:47:383 sub-device: , has item: attr/lastseen
22:56:47:383 sub-device: , has item: attr/manufacturername
22:56:47:383 sub-device: , has item: attr/modelid
22:56:47:383 sub-device: , has item: attr/name
22:56:47:384 sub-device: , create item: attr/productid
22:56:47:384 sub-device: , has item: attr/swversion
22:56:47:384 sub-device: , has item: attr/type
22:56:47:384 sub-device: , has item: attr/uniqueid
22:56:47:385 sub-device: , has item: config/alert
22:56:47:385 sub-device: , has item: config/battery
22:56:47:385 sub-device: , has item: config/on
22:56:47:385 sub-device: , has item: config/reachable
22:56:47:385 sub-device: , has item: state/buttonevent
22:56:47:386 sub-device: , has item: state/lastupdated
22:56:47:386 DEV found DDF for 0x086BD7FFFE5FA50F, path: /usr/share/deCONZ/devices/ikea/tradfri_open_close_remote.json
22:56:47:386 DEV no DDF for 0x847127FFFE407E32, modelId: 4512703
22:56:47:386 DEV create on-the-fly DDF for 0x847127FFFE407E32
22:56:47:387 DEV no DDF for 0x847127FFFE40E26D, modelId: 4512703
22:56:47:387 DEV create on-the-fly DDF for 0x847127FFFE40E26D
22:56:47:387 DEV no DDF for 0x0017880106440113, modelId: SML002
22:56:47:388 DEV create on-the-fly DDF for 0x0017880106440113
22:56:47:388 DEV no DDF for 0x0017880104B529AA, modelId: SML001
22:56:47:388 DEV create on-the-fly DDF for 0x0017880104B529AA
22:56:47:389 DEV no DDF for 0x0017880104F1A39E, modelId: RWL021
22:56:47:389 DEV create on-the-fly DDF for 0x0017880104F1A39E
22:56:47:392 sub-device: 00:0b:57:ff:fe:e8:a3:07-01, has item: attr/lastannounced
22:56:47:393 sub-device: 00:0b:57:ff:fe:e8:a3:07-01, has item: attr/lastseen
22:56:47:393 sub-device: 00:0b:57:ff:fe:e8:a3:07-01, has item: attr/manufacturername
22:56:47:393 sub-device: 00:0b:57:ff:fe:e8:a3:07-01, has item: attr/modelid
22:56:47:393 sub-device: 00:0b:57:ff:fe:e8:a3:07-01, has item: attr/name
22:56:47:394 sub-device: 00:0b:57:ff:fe:e8:a3:07-01, has item: attr/swversion
22:56:47:394 sub-device: 00:0b:57:ff:fe:e8:a3:07-01, has item: attr/type
[... 2156 lines cut ...]
22:57:28:162 SC state change failed: 90:fd:9f:ff:fe:9b:29:6a-01
22:57:28:162 SC state change failed: 90:fd:9f:ff:fe:9b:29:6a-01
22:57:28:163 SC state change failed: 90:fd:9f:ff:fe:9b:29:6a-01
22:57:28:258 90:fd:9f:ff:fe:08:fe:24-01/config/on/startup expression: Item.val = Attr.val --> 255
22:57:28:418 90:fd:9f:ff:fe:9b:29:6a-01/config/color/execute_if_off expression: Item.val = (Attr.val & 0x01) !== 0 --> false
22:57:28:419 SC tick --> StateRead config/on/startup, 90:fd:9f:ff:fe:9b:29:6a-01
22:57:28:420 SC config/color/execute_if_off: synced
22:57:28:420 SC --> StateFinished
[... cut ...]
22:57:28:708 SC tick --> StateRead config/color/execute_if_off, 00:0b:57:ff:fe:8d:51:ef-01
22:57:28:708 SC state change failed: 00:0b:57:ff:fe:8d:51:ef-01
22:57:28:709 SC state change failed: 00:0b:57:ff:fe:8d:51:ef-01
22:57:28:709 SC state change failed: 00:0b:57:ff:fe:8d:51:ef-01
22:57:28:709 SC state change failed: 00:0b:57:ff:fe:8d:51:ef-01
22:57:28:709 SC state change failed: 00:0b:57:ff:fe:8d:51:ef-01
22:57:28:755 90:fd:9f:ff:fe:9b:29:6a-01/cap/color/ct/min expression: Item.val = Attr.val --> 250
```

manup, 2024-05-23:
Thanks a lot 👍 

> 22:56:47:389 DEV no DDF for 0x0017880104F1A39E, modelId: RWL021

This is suspicious, since there exists a DDF for it. Could you please send me the `zll.db`file, or an backup created in the Phoscon App to support@dresden-elektronik.de I have a rough idea what might be the problem but need to verify that in the debugger.

olemr, 2024-05-24:
Well try to find time to do that this evening..

olemr, 2024-05-25:
> Thanks a lot 👍
> 
> > 22:56:47:389 DEV no DDF for 0x0017880104F1A39E, modelId: RWL021
> 
> This is suspicious, since there exists a DDF for it. Could you please send me the `zll.db`file, or an backup created in the Phoscon App to [support@dresden-elektronik.de](mailto:support@dresden-elektronik.de) I have a rough idea what might be the problem but need to verify that in the debugger.

Sent mail Yday. Check spam if not seen.

manup, 2024-05-25:
@olemr thanks a lot, I have now yours and another zll.db file for testing. Hope to find the cause quickly.

What @Quatalspropella has seen indicates what I think there where a few attributes missing in the database to match the DDF. This can eventually resolve automatically but may take time especially for sleeping end devices. In the zll.db files I should see if that is the case and perhaps find a workaround to speed things up.

olemr, 2024-05-25:
FYI. I did the upgrade to 2.27.0 just before going to bed, and noticed that switch didn't work the following morning going to work. Followed up in the afternoon with several button events and even changed the battery in the switch. (was 3.0V, new one 3.3V).
Just saying that quite amount of time went by, without it ever started working.

olemr, 2024-05-27:
Tested 2.27.1 but no cigar ..
RWL021 still in the net does not trigger any activity in the Switch section in Phoscon. 
But I do see a blinking blue light in the GUI.
The RWL021 I deleted, I'm not able to include back in. Only a hex ID in the GUI, but the basic cluster seems OK.
Log attached is from several inclusion attempts and button presses on both Switches.
[RWL log.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/15462342/RWL.log.zip)

GUI (note, no DDF icon on the RWL021s)
[image]
[image]

manup, 2024-05-28:
The data shown in the Basic Cluster should be enough to assign the DDF :thinking: 
For a test can you please:
- Right click on the switch
- Select *"Edit DDF"* in the context menu
- In the opening dialog select *"File &rarr; Hot Reload"*

This should trigger the switch to be assigned again to a DDF after a few seconds (which normally should happen automatically).

olemr, 2024-05-28:
Hot reloading DDF worked!
But had to delete the pending Add.
Adding worked find with DDF loaded, and I now get a GUI grey line feedback and green light on the physical switch when pressing keys.

[image]

olemr, 2024-05-28:
Another clue.
Prior to reloading DDF, the heading in the editor said somehting like '*unnamed file ...' (can't remember exact wording)
After reload it reads:
[image]

SwoopX, 2024-05-28:
I've taken a quick look at the shared debug log. Seeing 2 paring attempts where the switch apparently reported manufacturer and model ID correctly. However, the respective DDF wasn't picked up (see below)? @manup maybe you can share your thoughts?

[image]

manup, 2024-05-28:
The empty path indicates that this is just the auto generated "Draft" DDF but not the actually existing one from disk.

I'm currently making some test with fresh repairing of devices to see if I can catch similar case. Also extend logging on a few places to see better what is going on.

manup, 2024-05-28:
>01:02:32:647 DEV found DDF for 0x000B57FFFE9DEA59, path: /usr/share/deCONZ/devices/ikea/tradfri_bulb_e27_cws_opal_600lm.json
>01:02:32:648 DEV no DDF for 0x000B57FFFE103E81, modelId: TRADFRI bulb E27 WS�opal 980lm

Not related to this issue, but this one is also interesting, our old friend the UTF-8 vs. Latin-1 encoding mix :)

ebaauw, 2024-05-28:
> Not related to this issue, but this one is also interesting, our old friend the UTF-8 vs. Latin-1 encoding mix :)

Light firmware still contains the weird character (0xA0), but _Cluster Info_ panel in the GUI no longer shows it.  Also, in my v2.27.1-beta log:
```
May 28 19:07:04 pi6 deCONZ[3351047]: 19:07:03:177 DEV no DDF for 0x000B57FFFE39357F, modelId: TRADFRI bulb E27 WS opal 980lm
```

Another platform-specific feature?

olemr, 2024-05-28:
I can confirm that 2.27.2 made the HUE Dimmer Switch RWL021 great again. ☺

sch115, 2024-05-31:
Hello there! I believe there is pretty much same issue with Hue Wall Modules as I am having non-responding situation with all of my RDM001 since moving to deCONZ 2.27.2 All visible in GUI but not working. Deleting and re-adding not helping either as pairing cannot be done.

Ive also tried Hot Reload method but without success.

Current versions:
Raspbian GNU/Linux Bullseye (11)
Homebridge 1.8.2
deCONZ version 2.27.2
ConbeeII 26780700

[31/05/2024, 07:15:36] [deCONZ] Phoscon-GW: Entryway and Hall Enrave: add accessory

[31/05/2024, 07:15:36] [deCONZ] Entryway and Hall Enrave: Signify Netherlands B.V. RDM001 v1.0.5 (1 resources)

[31/05/2024, 07:15:36] [deCONZ] Entryway and Hall Enrave: warning: /sensors/29: unknown ZHASwitch: {“config”:{“battery”:100,“on”:true,“reachable”:true},“etag”:“0947922d0a21cb7f5ac2143cacf36296”,“lastannounced”:“2024-05-30T09:30:29Z”,“lastseen”:“2024-05-31T05:11Z”,“manufacturername”:“Signify Netherlands B.V.”,“mode”:1,“modelid”:“RDM001”,“name”:“Entryway and Hall Enrave”,“state”:{“buttonevent”:1002,“lastupdated”:“2024-04-08T04:39:20.932”},“swversion”:“1.0.5”,“type”:“ZHASwitch”,“uniqueid”:“00:17:88:01:0b:02:2e:29-01-fc00”}

[31/05/2024, 07:15:36] [deCONZ] Entryway and Hall Enrave Battery: Name: set to “Entryway and Hall Enrave Battery”

[31/05/2024, 07:15:36] [deCONZ] Entryway and Hall Enrave Battery: Configured Name: set to “Entryway and Hall Enrave Battery”

[31/05/2024, 07:15:36] [deCONZ] Entryway and Hall Enrave Battery: Battery Level: set to 100%

[31/05/2024, 07:15:36] [deCONZ] Entryway and Hall Enrave Battery: Status Low Battery: set to 0

[31/05/2024, 07:15:36] [deCONZ] Entryway and Hall Enrave Battery: lowBatteryThreshold: set to 20%

manup, 2024-05-31:
@sch115 can you please provide a log with starting deCONZ from command line as follows.

```
deCONZ --dbg-ddf=1 --dbg-info=1 | tee deconz.log
```

Just startup and running for 10 seconds should be enough to give us some details for debugging.

manup, 2024-05-31:
On a first look of your screenshot in the forum this might be the same OpenSSL issue seen in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7783 so could be already fixed.

Can you please also post a screenshot of **deCONZ Help > About deCONZ** here we can see if OpenSSL is loaded correctly.

sch115, 2024-05-31:
> @sch115 can you please provide a log with starting deCONZ from command line as follows.
> 
> ```
> deCONZ --dbg-ddf=1 --dbg-info=1 | tee deconz.log
> ```
> 
> Just startup and running for 10 seconds should be enough to give us some details for debugging.

Here you go [debug.txt](https://pastebin.com/UJ6Bes36)

sch115, 2024-05-31:
> On a first look of your screenshot in the forum this might be the same OpenSSL issue seen in #7783 so could be already fixed.
> 
> Can you please also post a screenshot of **deCONZ Help > About deCONZ** here we can see if OpenSSL is loaded correctly.

No SSL :/      
[image]

manup, 2024-05-31:
Thank, indeed looks like the same problem. I can make test version with above fix to check if that works, are you using 32-bit or 64-bit Raspbian?

`uname -a`

should show this.

sch115, 2024-05-31:
> Thank, indeed looks like the same problem. I can make test version with above fix to check if that works, are you using 32-bit or 64-bit Raspbian?
> 
> `uname -a`
> 
> should show this.

64-bit. Thanks a lot!

manup, 2024-05-31:
Here is the version for 64-bit ARM Raspbian:

[deconz_2.27.2-debian_arm64_5f8d1d.zip](https://github.com/user-attachments/files/15514402/deconz_2.27.2-debian_arm64_5f8d1d.zip)

To install use:

```
unzip deconz_2.27.2-debian_arm64_5f8d1d.zip
sudo dpkg -i deconz_2.27.2-debian_arm64_5f8d1d.deb
```

If it works the DDF labels should show up again.

sch115, 2024-05-31:
> Here is the version for 64-bit ARM Raspbian:
>
> [deconz_2.27.2-debian_arm64_5f8d1d.zip](https://github.com/user-attachments/files/15514402/deconz_2.27.2-debian_arm64_5f8d1d.zip)
>
> To install use:
>
> ``` [...]

Sorry to bother you, Im having issues with the version. I believed I am running 64-bit (Linux homebridge 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux) but the installation process says "package architecture (arm64) does not match system (armhf)". I probably installed 32-bit system back in the days. Is there a work-around or only 32 version is usable?

manup, 2024-05-31:
[deconz_2.27.2-debian_armhf_5f8d1d.zip](https://github.com/user-attachments/files/15517331/deconz_2.27.2-debian_armhf_5f8d1d.zip)

Right armhf is the 32-bit version, the attached version should do.

sch115, 2024-05-31:
> [deconz_2.27.2-debian_armhf_5f8d1d.zip](https://github.com/user-attachments/files/15517331/deconz_2.27.2-debian_armhf_5f8d1d.zip)
> 
> Right armhf is the 32-bit version, the attached version should do.

Hat down Manu! All modules are working again, even the temperature sensores are working. DDFs are shown already. Only thing is SSL info is missing - see attached files.
Regards from Prague!
[image]
[image]

manup, 2024-06-01:
Cool glad that this works, at least to get the DDFs loaded :+1: 

Can you also provide the output of following command to check if/which OpenSSL library is installed:

```
find /usr -name 'libcrypto*'
```

It may take a few seconds until results show up.

In my test VM with Debian Bullseye `libssl1.1` package is installed and does load successfully.

sch115, 2024-06-02:
> Cool glad that this works, at least to get the DDFs loaded 👍
> 
> Can you also provide the output of following command to check if/which OpenSSL library is installed:
> 
> ```
> find /usr -name 'libcrypto*'
> ```
> 
> It may take a few seconds until results show up.
> 
> In my test VM with Debian Bullseye `libssl1.1` package is installed and does load successfully.

It shows "/usr/lib/arm-linux-gnueabihf/libcrypto.so.1.1"

manup, 2024-06-02:
Interesting then there seems to be a missing symlink of `libcrypto.so` to this version.
I've extended the code to deal with this and find the two most common OpenSSL major versions even without the symlink in https://github.com/dresden-elektronik/deconz-lib/commit/5c424b6e3120c0148350302fc2f848df7760a5bd

sch115, 2024-06-02:
> Interesting then there seems to be a missing symlink of `libcrypto.so` to this version. I've extended the code to deal with this and find the two most common OpenSSL major versions even without the symlink in [dresden-elektronik/deconz-lib@5c424b6](https://github.com/dresden-elektronik/deconz-lib/commit/5c424b6e3120c0148350302fc2f848df7760a5bd)

This is actually where my knowledge ends so I'll wait for the upcoming version)) If you want me to test it somehow, just let me know with few steps and i will do it. Anyway, thank you very much for helping me out, I really appreciate the work you do on this!

DEhlertLBA, 2024-06-10:
hello I am also having the same issue with 2.27.2, see attached log. 
[deconz.log](https://github.com/user-attachments/files/15757528/deconz.log)

DEhlertLBA, 2024-06-11:
when i try bind a button

Machine ev: BTN_NEXT_PRESSED
TypeError: Cannot read properties of null (reading 'split')
    at constants.js?656748:3:6934
    at new Promise (<anonymous>)
    at Object.createResource (constants.js?656748:3:5673)
    at Object.BTN_NEXT_PRESSED (wizard-actions.js?656748:3:4694)
    at wizard-states.js?656748:3:2326
    at o.exec (xstate.interpreter_4-18-0.js?289616656748:15:16231)
    at o.execute (xstate.interpreter_4-18-0.js?289616656748:15:9610)
    at o.update (xstate.interpreter_4-18-0.js?289616656748:15:9854)
    at xstate.interpreter_4-18-0.js?289616656748:15:8510
    at t.process (xstate.interpreter_4-18-0.js?289616656748:15:6910)

DEhlertLBA, 2024-06-12:
{
    "20:42:33:679": {
        "attr": {
            "id": "32",
            "lastannounced": null,
            "lastseen": "2024-06-12T10:42Z",
            "manufacturername": "Signify Netherlands B.V.",
            "mode": 1,
            "modelid": "RWL022",
            "name": "switch_bedroom_2",
            "productname": "Hue dimmer switch",
            "swversion": "2.44.0_hBB3C188",
            "type": "ZHASwitch",
            "uniqueid": "00:17:88:01:0b:7a:b6:53-01-fc00"
        },
        "e": "changed",
        "id": "32",
        "r": "sensors",
        "t": "event",
        "uniqueid": "00:17:88:01:0b:7a:b6:53-01-fc00"
    },
    "20:42:33:732": {
        "e": "changed",
        "id": "32",
        "r": "sensors",
        "state": {
            "buttonevent": 1000,
            "eventduration": 0,
            "lastupdated": "2024-06-12T10:42:34.256"
        },
        "t": "event",
        "uniqueid": "00:17:88:01:0b:7a:b6:53-01-fc00"
    },
    "20:42:33:780": {
        "e": "changed",
        "id": "32",
        "r": "sensors",
        "state": {
            "buttonevent": 1002,
            "eventduration": 1,
            "lastupdated": "2024-06-12T10:42:34.345"
        },
        "t": "event",
        "uniqueid": "00:17:88:01:0b:7a:b6:53-01-fc00"
    },
    "20:42:36:93": {
        "e": "changed",
        "id": "32",
        "r": "sensors",
        "state": {
            "buttonevent": 1000,
            "eventduration": 0,
            "lastupdated": "2024-06-12T10:42:36.670"
        },
        "t": "event",
        "uniqueid": "00:17:88:01:0b:7a:b6:53-01-fc00"
    },
    "20:42:36:219": {
        "e": "changed",
        "id": "32",
        "r": "sensors",
        "state": {
            "buttonevent": 1002,
            "eventduration": 1,
            "lastupdated": "2024-06-12T10:42:36.797"
        },
        "t": "event",
        "uniqueid": "00:17:88:01:0b:7a:b6:53-01-fc00"
    },
    "20:42:42:210": {
        "e": "changed",
        "id": "32",
        "r": "sensors",
        "state": {
            "buttonevent": 2000,
            "eventduration": 0,
            "lastupdated": "2024-06-12T10:42:42.788"
        },
        "t": "event",
        "uniqueid": "00:17:88:01:0b:7a:b6:53-01-fc00"
    },
    "20:42:42:276": {
        "e": "changed",
        "id": "32",
        "r": "sensors",
        "state": {
            "buttonevent": 2002,
            "eventduration": 0,
            "lastupdated": "2024-06-12T10:42:42.854"
        },
        "t": "event",
        "uniqueid": "00:17:88:01:0b:7a:b6:53-01-fc00"
    }
}
   
Yes, I cannot configure the switch, i can get all the way to the apply button, then it doesnt do anything.

ChrisHae, 2024-06-12:
This is the new Philips Remote with the "hue" button, am I right? Yes there was a bug in the latest release that the button in the switch editor is not clickable. I have fixed it and it will be in the next deCONZ release.
I think @manup can also build a preview version to make the fix available faster.

Zehir, 2024-06-12:
I think I am correct here, it's was released in [version 2_27_03](https://deconz.dresden-elektronik.de/win/deCONZ_Setup_Win32_V2_27_03.exe)

ebaauw, 2024-06-12:
I think this issue went off-topic.  The original post is about the Hue wall switch module, not about the 2nd gen Hue dimmer switch.

manup, 2024-06-19:
Could anybody test with v2.27.4-beta? There were fixes for the Wall- and Dimmer switches.

sch115, 2024-06-19:
> Could anybody test with v2.27.4-beta? There were fixes for the Wall- and Dimmer switches.

I can confirm Hue Wall Module and Hue Tap Dial in working condition. Dunno about Dimmer switch as I do not use any. See attached screens. Thanks again!
[image]
[image]

manup, 2024-06-19:
Cool thanks for testing :+1:  I think we can close this issue for now since it is only related to the Wall switch.

closed 2024-06-19 by manup

NCC1701G, 2024-07-05:
I think the error still exists.
Since the update, two of my RWL021s lose connection after a short time.
When I restart deconz the light switches work again. But again only for a short time.

Gateway Version: 2.27.4
Firmware: 26780700

`{
    "config": {
        "battery": 57,
        "group": "20003",
        "on": true,
        "reachable": true
    },
    "etag": "20ee4acbb032dd2e4e9019a8cb6e013c",
    "lastannounced": "2024-07-05T11:16:57Z",
    "lastseen": "2024-07-05T11:16Z",
    "manufacturername": "Philips",
    "mode": 1,
    "modelid": "RWL021",
    "name": "Lichtschalter Schlafzimmer",
    "productname": "Hue dimmer switch",
    "state": {
        "buttonevent": 4002,
        "eventduration": 1,
        "lastupdated": "2024-07-05T11:17:03.192"
    },
    "swversion": "6.1.1.28573",
    "type": "ZHASwitch",
    "uniqueid": "00:17:88:01:06:73:43:a2-02-fc00"
}`

NCC1701G, 2024-07-05:
I'm now back on Version 2.26.3 and I have no problems with dimmer switch. 
So the error still exists in Version 2.27.4. 
Please reopen this issue.

gibsganich, 2024-11-11:
Hi all,

Not sure if this is the right place, to talk about.
Issue was seen first some time ago after upgrade to 2.27 (PI2b/RaspBee2), and I found this article.
I did not test/use the fix in beta, waiting for the next stable release including the fix.
Due to the fact PI2 is very old old slow, I decided to upgrade to PI4/raspbee2, and changed to Homebridge running deconz/hue pluggin. 
Linux PI-4b 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux

I seems issue is not fixed in 2.28.1., I am still facing this issue with HUE Wall Switch RDM001.
All wall switches works fine initially, but after a time they stops working. A reboot fix this temporary.
Same on the old PI running 2.28.1.
Refer with your comment, please.

Thanks

DEhlertLBA, 2024-11-11:
I have had to keep an old version on as its still not fixed for me either. I am seriously considering swapping to zigbee2mqtt

NCC1701G, 2024-11-13:
I installed the version 2.28.1 and the error no longer occurs, since 3 weeks.
My System:
Proxmox 8.2.7
Debian Container

NCC1701G, 2024-12-04:
Correction, the problem still exists.

I added a new device yesterday and after that, two of my three light switches stopped working.

This light switch works:
`{
    "config": {
        "battery": 2,
        "group": "18",
        "on": true,
        "reachable": true
    },
    "etag": "680a3a9a56ccc90b7a6554f931c72be0",
    "lastannounced": "2024-12-04T08:47:00Z",
    "lastseen": "2024-12-04T09:12Z",
    "manufacturername": "Philips",
    "mode": 1,
    "modelid": "RWL021",
    "name": "Lichtschalter Kinderbad",
    "productname": "Hue dimmer switch",
    "state": {
        "buttonevent": 1002,
        "eventduration": 1,
        "lastupdated": "2024-12-04T08:46:59.201"
    },
    "swversion": "6.1.1.28573",
    "type": "ZHASwitch",
    "uniqueid": "00:17:88:01:06:73:3b:1a-02-fc00"
}`

This light switch don't work:
`{
    "config": {
        "battery": 5,
        "group": "20003",
        "on": true,
        "reachable": true
    },
    "etag": "76d1c86caacbdf2d10c350b0adb54f1b",
    "lastannounced": null,
    "lastseen": "2024-12-04T09:11Z",
    "manufacturername": "Philips",
    "mode": 1,
    "modelid": "RWL021",
    "name": "Lichtschalter Schlafzimmer",
    "productname": "Hue dimmer switch",
    "state": {
        "buttonevent": 4002,
        "eventduration": 1,
        "lastupdated": "2024-12-03T06:50:06.016"
    },
    "swversion": "6.1.1.28573",
    "type": "ZHASwitch",
    "uniqueid": "00:17:88:01:06:73:43:a2-02-fc00"
}`

Is any developer actually still reading our problems since the post is closed?

ebaauw, 2024-12-04:
I think this is another symptom of #7935: after searching for new devices, the API no longer updates some existing DDF-exposed devices.  Presumably the issue is with DDF hot reload.  Symptoms include that updates from the device are still received by deCONZ (as witnessed by the GUI updating the values), but those updates aren't processed by the REST API plugin.  After restarting deCONZ, the API resources are updated again.

NCC1701G, 2024-12-04:
> I think this is another symptom of #7935: after searching for new devices, the API no longer updates some existing DDF-exposed devices. Presumably the issue is with DDF hot reload. Symptoms include [...]

Yes, I can confirm this behavior.
After restarting the deconz service, the light switches work again.

snozzlebert, 2024-12-20:
> I have had to keep an old version on as its still not fixed for me either. I am seriously considering swapping to zigbee2mqtt

I'm considering the same. deCONZ user since 2017. There were always a lot of improvements, (critical) bugs were solved quickly. There are no commits or releases for while. Is 2.26.3 the last known version without this bug?