dresden-elektronik/deconz-rest-plugin #7787: Zigbee Smart Remote Controller -Y

tengomol, 2024-05-24:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Zigbee Smart Remote Controller -Y

### Manufacturer

_TYZB01_qm6djpta

### Model identifier

TS0215

### Device type to add

Switch

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration
[image]

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Smanar, 2024-05-25:
Hello, what is your OS ?
Because for test we need to edit the button_maps.json file.
From the picture it look like a clone of

`"modelids": ["_TZ3000_0zrccfgx", "_TZ3000_fsiepnrh", "_TZ3000_p6ju8myv"],`

tengomol, 2024-05-25:
It is:

[image]

I'm running in raspberry pi 3 model b.

BR,
Oliver.

Smanar, 2024-05-26:
Ok so hard for you to test DDF, and more to edit the file ....

tengomol, 2024-06-02:
How Can I do it? Is there a guide? 

I edited "button_maps.json" directly in Home Assistant using Portainer Addon, I added:

"modelids": ["_TZ3000_0zrccfgx", "_TZ3000_fsiepnrh", "_TZ3000_p6ju8myv","TS0215"],

But It doesn't work, is is need to modify in the code, compile and restart? 

BR,
Oliver.

Smanar, 2024-06-03:
There is 2 things:

The easier, you can create this DDF in the persistent folder `/data/.local/share/dresden-elektronik/deCONZ/devices/` (hidden)

```
{
   "schema":"devcap1.schema.json",
   "manufacturername":["_TZ3000_0zrccfgx", "_TZ3000_p6ju8myv", "_TYZB01_qm6djpta"],
   "modelid":["TS0215A", "TS0215A", "TS0215"],
   "vendor":"Woox",
   "product":"R7054 4 gang",
   "sleeper":true,
   "status":"Gold",
   "subdevices":[
      {
         "type":"$TYPE_SWITCH",
         "restapi":"/sensors",
         "uuid":[
            "$address.ext",
            "0x01",
            "0x0500"
         ],
         "fingerprint":{
            "profile":"0x0104",
            "device":"0x0401",
            "endpoint":"0x01",
            "in":[
               "0x0000",
               "0x0500",
               "0x0501"
            ]
         },
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"config/enrolled",
               "public":false
            },
            {
               "name":"config/on"
            },
            {
               "name":"config/pending"
            },
            {
               "name":"config/reachable"
            },
            {
            	"name": "state/buttonevent"
            },
            {
               "name":"state/lastupdated"
            },
            {
               "name": "state/lowbattery",
               "default": false
            }
         ]
      }
   ],
   "bindings":[
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0500"
      },
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0501"
      }
   ]
}
```

But ATM to work you need too  edit the button_maps.json file to add this model, problem HA use docker, so you realy need lot of skill to use this kind of OS, changes are not persistent, if you restart, you loose your change.

tengomol, 2024-06-05:
I did all the changes, DDF and also changed button_maps.json. DDF is working because it is detected as switch (instead of sensor), but there is no button. :-( 
I changed buttons_maps.json, delete device and added again, but still not showing any button. I restarted and change again button-maps.json (because it is not persistent) add device again and nothing change, no button was added.
I don't know  if the "hot" changes in button-maps.json are applied or there is an error.

Smanar, 2024-06-05:
It will depend on how you have changed the button_maps.json.
Change need to be persistent, as you need to restart the deconz docker, it's possible, but not easy to do, and I can't explain how to do.
You don't need to re-include the device, the DDF is a thing, the buttonmap another. Without the buttonmap file, you will have error on logs about a missing buttonmap every time you press a button.

tengomol, 2024-06-05:
I think so, thanks again for your support @Smanar .
I'll investigate if I can create a new docker image from existing ones to include new button_maps.json, change the script to build home-assistant plugin to use my new docker image, compile and packed home assistant addon, uninstall official addon and install my compiled. But not sure if I have time to do this changes, a lot of time needed.
If I finally do it and it works, I will let you know the result, thank you!

Smanar, 2024-06-06:
There is "docker request" to make possible a file edition, but not easy for me.
Something easier, just install deconz on a PC for 10mn, time for test, clone your zigbee network and make HA use the second deconz install if needed, but don't need HA to test the file.

tengomol, 2024-06-10:
Finally I installed using raspbian OS and connect using deconz integration to Home Assistant.

I configured DDF and button_maps.json, and I can see the events in home assistant, for example I configure this automation and it is working when I click one of the buttons:

platform: event
event_type: deconz_event
event_data:
  id: remote_control
  unique_id: XXXXXXXXXXX
  event: 4002
  device_id: XXXXXXXXXXXX

I think it is for button_maps.json, without add DDF I think also works (not sure)

Smanar, 2024-06-11:
It's fine for me, 4002 mean the button 4, release after press (ARM button), and it's the only even the device is able to send, so for me no problem.
Do you want to make the PR tu submit the device or want I make it ?

closed 2024-07-10 by github-actions[bot]