dresden-elektronik/deconz-rest-plugin #7795: Schneider Elektric Room Thermostat

Feldi1606, 2024-05-28:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Schneider Electric CCTFR6400

### Manufacturer

Schneider Electric

### Model identifier

Thermostat

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]
[image]

### Further relevant clusters

[image]

[image]

[image]

[image]

[image]
[image]

Feldi1606, 2024-05-28:
Hello can anyone issue a DDF for this device or tell me which DDF might work?
And is it possible to peer this room thermostat with the heating thermostat so that the two can communicate directly? And I would also like to set up weekly programs. Unfortunately I haven't found any instructions or syntax on how to do this yet. I assume that the Cluster 0201 thermostat should be used for this.

SwoopX, 2024-05-29:
Hi there.

Unfortunately, the screenshots indicate that the clusters have not been read properly. [This page](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to...#read-device-clusters) explains how to read clusters. Please take new ones and update the issue accordingly. Thanks a lot!

SwoopX, 2024-05-29:
If you update to the latest deconz version released yesterday, you should be able to get a draft DDF already by right clicking on the node and selecting the DDF editor entry. That is always a good starting point.

The device seems to be ale to communicate directly with thermostats, as it holds the grey thermostat client cluster, presumably allowing to set temperatures remotely. Schedules, however, cannot be set on this device.

Feldi1606, 2024-05-29:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Feldi1606, 2024-05-29:
So I hope all the screenshots are correct now. Unfortunately, the instructions do not say what you have to pay attention to with the clusters. 
What can I do with the DDF draft? Should I upload it somewhere?

Smanar, 2024-05-29:
If you can copy/paste the "Draft", yes it can be usefull, but now too late ^^
Can try this DDF, based on another Schneider thermostat

```
{
  "schema": "devcap1.schema.json",
  "uuid": "e4ee14c0-c6a8-431c-8bff-4b4aa32d5de3",
  "manufacturername": "Schneider Electric",
  "modelid": "Thermostat",
  "product": "CCTFR6400",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_THERMOSTAT",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0201"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0301",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0201"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/heatsetpoint"
        },
        {
          "name": "config/unoccupiedheatsetpoint"
        },
        {
          "name": "config/mode"
        },
        {
          "name": "config/offset"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/on",
          "parse": {
            "at": "0x0008",
            "cl": "0x0201",
            "ep": 1,
            "eval": "Item.val = Attr.val > 3;",
            "fn": "zcl:attr"
          },
          "read": {
            "fn": "none"
          }
        },
        {
          "name": "config/locked",
          "refresh.interval": 3660
        },
        {
          "name": "state/temperature"
        }
      ]
    },
      {
         "type":"$TYPE_TEMPERATURE_SENSOR",
         "restapi":"/sensors",
         "uuid":[
            "$address.ext",
            "0x01",
            "0x0402"
         ],
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"config/offset"
            },
            {
               "name":"config/on"
            },
            {
               "name":"config/reachable"
            },
            {
               "name":"state/lastupdated"
            },
            {
               "name":"state/temperature"
            }
         ]
      },
      {
         "type":"$TYPE_TEMPERATURE_SENSOR",
         "restapi":"/sensors",
         "uuid":[
            "$address.ext",
            "0x01",
            "0x0402"
         ],
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
               "name":"attr/swversion"
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"config/offset"
            },
            {
               "name":"config/on"
            },
            {
               "name":"config/reachable"
            },
            {
               "name":"state/lastupdated"
            },
            {
               "name":"state/temperature"
            }
         ]
      }
  ],
  "bindings": [
      {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0201",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 60,
          "max": 300,
          "change": "0x00000032"
        },
        {
          "at": "0x0012",
          "dt": "0x29",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0008",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0204",
      "report": [
        {
          "at": "0x0001",
          "dt": "0x30",
          "min": 1,
          "max": 3600
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 60,
          "max": 300,
          "change": "0x00000032"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 60,
          "max": 300,
          "change": "0x00000032"
        }
      ]
    }
  ]
}
```

Feldi1606, 2024-05-29:
The DDF works so far. But what I've noticed is that I can't see any battery or humidity values. And the heatsetpoint is not displayed analogously to the device. How can I help you fix this Problems.

SwoopX, 2024-05-29:
@Feldi1606 Can you please share the initially auto generated DDF here to have a fresh and good start? The suggested DDF above creates more issues than it actually solves, exposing lots of functionality the device does not offer. I assume Smanar had a bad day or posted any wrong version.

Smanar, 2024-05-30:
>I assume Smanar had a bad day or posted any wrong version.

Not realy, have made a copy/paste of this one https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/elko/thermostat_16_a.json but you are right, this DDf is not compatible at all, no cluster 0x0201 ....

Feldi1606, 2024-05-30:
Hello Yes, exactly the device is basically an external temperature and humidity measuring instrument and a remote control to control the temperature of one or I think even several ITRVs and also to change the mode. I'm attaching a link to the device.
[https://merten.de/produkt/cctfr6400-wiser-raumthermostat-mit-display.html](url)

The device is powered by 2 AA batteries with 1.5V so 3V should be suitable. 
But I can already see the attribute 0x0021 in deconz and it currently says 100, see screenshot
[image]

SwoopX, 2024-05-30:
> I think you don't need the automatically generated DDF or have I done something wrong?

Hm, I would have expected some more but agreed, that is not exactly helpful 🙂

Smanar, 2024-05-31:
The value was "0" on your previous capture ....
Have added the battery support on the previous DDF, but still need the thermostat part.

Feldi1606, 2024-05-31:
are you okay with that?

[image]
[image]

Feldi1606, 2024-05-31:
Is it correct that I see the room thermostat once under thermostats and once under sensors? 
The battery status is now shown to me as 50% even though Deconz says 100%.

Smanar, 2024-06-01:
>The battery status is now shown to me as 50% even though Deconz says 100%

My bad 
`"eval": "Item.val = Attr.val / 2",`

Just need to remove the /2

>And the humidity value could be output without any decimal places. It is currently sending a lot of data with very small changes of 0.01%

Can set min/max/chnage here 

```
      "cl": "0x0405",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 60,
          "max": 300,
          "change": "0x00000032"
        }
      
```

>Is it correct that I see the room thermostat once under thermostats and once under sensors?

On phoscon ?

But the problem is for the ZHAThermostat. SwoopX have removed it but I m seing you can send request to it like for classic thermostat

> 
> Action (enum)
> 
> Triggered action (e.g. a button click). Value can be found in the published state on the action property. It's not possible to read (/get) or write (/set) this value. The possible values are: screen_sleep, screen_wake, button_press_plus_down, button_press_center_down, button_press_minus_down.
> Climate
> 
> This climate device supports the following features: occupied_heating_setpoint, local_temperature, pi_heating_demand.
> 
>     occupied_heating_setpoint: Temperature setpoint. To control publish a message to topic zigbee2mqtt/FRIENDLY_NAME/set with payload {"occupied_heating_setpoint": VALUE} where VALUE is the °C between 4 and 30. Reading (/get) this attribute is not possible.
>     pi_heating_demand: Position of the valve (= demanded heat) where 0% is fully closed and 100% is fully open. To control publish a message to topic zigbee2mqtt/FRIENDLY_NAME/set with payload {"pi_heating_demand": VALUE} where VALUE is the % between 0 and 100. Reading (/get) this attribute is not possible.

But not sure it's something usefull, better for automation to control direclty the TRV for exemple than the remote.

PR #7798 (2024-06-02): Add DDF for Wiser room thermostat CCTFR6400
Closes #7795

Unless the manufacturer specific cluster holds this functionality, it is not possible to remotely set any temperatures on the device itself. However, if you physically interact with it, you can apparently steer a thermostat remotely if client and server thermostat clusters are directly bound to each other.
closed 2024-06-02 by manup