dresden-elektronik/deconz-rest-plugin #7799: Tuya Soil Sensor (diff. manufacturer)

kopierschnitte, 2024-06-01:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Tuya Soil Sensor

### Manufacturer

_TZE204_myd45weu

### Model identifier

TS0601

### Device type to add

Other

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

The device is already supported by https://github.com/dresden-elektronik/deconz-rest-plugin/issues/6731 but some units are sold with a different manufacturer-id. Mine reads _TZE**204**_myd45weu instead of _TZE**200**_myd45weu.

After cloning the current DDF and changing the manufacturer-id, the device works as expected. Therefore I guess, it's just an alias for the other device.

Smanar, 2024-06-07:
So adding your device to this DFF https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZE200_myd45weu_soil_sensor.json Make it working ?

So there is nothing to do ? (except make the PR)

kopierschnitte, 2024-06-07:
Yes, indeed ;-)

Smanar, 2024-06-07:
Thx, was fast ^^, PR done here > https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7808

kopierschnitte, 2024-06-07:
Great. Thanks!

Mimiix, 2024-06-07:
Please keep it open until its merged.

Smanar, 2024-06-08:
The DDF have worked ?
Do you want to make the PR or want I make it ?

kopierschnitte, 2024-06-08:
Yes, perfectly. I'm not so skilled using Github. Would be great if you could do the PR.

Smanar, 2024-06-08:
Oups, sorry, haven't see the PR was already done, and by me ....

closed 2024-06-08 by manup