dresden-elektronik/deconz-rest-plugin #7805: Tuya Water Leak Sensor - another manufacturer id

telfax, 2024-06-03:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Water Leak Sensor ZW06

### Manufacturer

_TZ3000_wuep9zng

### Model identifier

TS0207

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

Alarm 1 active when water is detected.
[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration
[image]

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

telfax, 2024-06-03:
Device details:
http://www.zz-power.com/product-5703-5274-109575.html

Smanar, 2024-06-07:
Hello, can just try to add it in this looonnnggg list https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_TS0207_water_leak_sensor.json

telfax, 2024-06-29:
Unfortunately this did not work.
After adding, the device still does not show up in Phoscon ☹️

Smanar, 2024-06-29:
@how you have done that ?
What is your OS ?
You need to edit 2 lines, Manufacture name AND model ID, then restart deconz and re include the device.

telfax, 2024-06-30:
Added the two lines
[image]

Stuff is running on a PI3, I uploaded the modified file restarted the phoscon gateway and tried to pair the sensor again.

Sensor indicated pairing succes by LED by it does not show up in Phoscon sensors.

Smanar, 2024-06-30:
Strange, I don't see typo, all is fine for me.
Can you check if you can see it direclty in the API ? using Phoscon/HELP/API Information/sensors ?

telfax, 2024-07-07:
Device is not listed in sensors API response :-/

closed 2024-08-06 by github-actions[bot]

cosmo84, 2024-08-13:
Hi,
I have the same device but (unfortunately) I run Deconz in docker and I have some issues in adding/testing new DDF in Docker. I open the updated DDF and do a hot reload but nothing happens. When I close the window and select "edit DDF" again, the old one is shown again.
Is the updated DDF above working? If so, it could be added and I might get a solution that way :-)

Thanks!!

Smanar, 2024-08-13:
Thx for return.
Added https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7886