dresden-elektronik/deconz-rest-plugin #7811: iCASA Zigbee Dimmer Pro ICZB-IW21D

ebaauw, 2024-06-09:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

iCASA Zigbee Dimmer Pro

### Manufacturer

icasa

### Model identifier

ICZB-IW21D

### Device type to add

Switch

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### On/Off
[image]

### Level Control
[image]

### Simple Metering
[image]

### Electrical Measurement
[image]

### OTAU
Manufacturer: 0x1224
Image Type: 0x012C
Version: 0x0000003A

ebaauw, 2024-06-09:
Look like your regular Sunricher-OEM in-wall dimmer , with light blue housing.  I'm told it's the successor of the [ICZB-IW11D](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7697), now with power and consumption measurement.

Preto, 2024-06-15:
Hi @ebaauw, might this just be the same device as #7190 from [Robbshop](https://www.robbshop.nl/robb-smarrt-pro-zigbee-dimmer-400w) (rebranded Sunricher)? These also have consumption and power measurement.

I couldn't get a DDF to work myself, without lights flickering or other weirdness :smile:. Screenshots are in the mentioned issue and look almost the same to me. I have a few different SW Build's installed of the mentioned device.

If its the same device, can you make it compatible so I can use your freshly generated DDF as well?

ebaauw, 2024-06-15:
Could very well be.  Note however the differenced in firmware version (r54 vs r58), so there might be subtle changes between the various OEM clones.

You can always copy the DDF, change the manufacturer and model, and try it on the Robbshop dimmer.

ebaauw, 2024-06-29:
Just like the ICZB-IW11D, the device supports commands _Trigger Effect_, _Off with Effect_, _Move to Level (with On/Off)_ and attributes _Execute If Off_, _On/Off Transition Time_, _On Level_, and _PowerOn OnOff_ as well as .PowerOn Level_.   _OnOff_ and _Current Level_ are reportable.

The consumption multiplier/divisor values are reported correctly:
- Current Summation Delivered is in Ws (1/3600000 kWh), so needs to de divided by 3600 for `state/consumption` in Wh;
- RMS Voltage is in 0.1 V, so needs to be divided by 10 for `state/voltage` in V;
- RMS Current is in mA (0.001 A), so can be used directly for `state/current` in mA;
- Active Power is in 0.1 W, so needs to be divided by 10 for `state/power` in W.
All these attributes are reportable.

Oddly, when the device is off, it reports 1.3V and 17mA.  For now, I'll be passing those values to the API, but I'm not sure if that's the best choice.

This should lead to the following resources:
```json
{
  "capabilities": {
    "alerts": [
      "none",
      "select",
      "lselect",
      "blink",
      "breathe",
      "okay",
      "channelchange",
      "finish",
      "stop"
    ]
  },
  "config": {
    "bri": {
      "execute_if_off": true,
      "on_level": "previous",
      "onoff_transitiontime": 4,
      "startup": "previous"
    },
    "groups": [
      "0"
    ],
    "on": {
      "startup": "previous"
    }
  },
  "etag": "1b6de7f5941db7c2aa1ca6d4f91774df",
  "hascolor": false,
  "lastannounced": null,
  "lastseen": "2024-06-29T18:28Z",
  "manufacturername": "icasa",
  "modelid": "ICZB-IW21D",
  "name": "Dimmable light 3",
  "state": {
    "alert": "none",
    "bri": 50,
    "on": true,
    "reachable": true
  },
  "swversion": "2.9.2_r58",
  "type": "Dimmable light",
  "uniqueid": "18:7a:3e:ff:fe:f7:41:40-01"
}
```
```
{
  "config": {
    "on": true,
    "reachable": true
  },
  "etag": "9def077abcdacd5251579ee30c3bcadc",
  "lastannounced": null,
  "lastseen": "2024-06-29T18:33Z",
  "manufacturername": "icasa",
  "modelid": "ICZB-IW21D",
  "name": "Consumption 308",
  "state": {
    "consumption": 478,
    "lastupdated": "2024-06-29T18:29:57.550"
  },
  "swversion": "2.9.2_r58",
  "type": "ZHAConsumption",
  "uniqueid": "18:7a:3e:ff:fe:f7:41:40-01-0702"
}
```
```
{
  "config": {
    "on": true,
    "reachable": true
  },
  "etag": "b68dedc7d01c3fe82a7d59c5413e8702",
  "lastannounced": null,
  "lastseen": "2024-06-29T18:34Z",
  "manufacturername": "icasa",
  "modelid": "ICZB-IW21D",
  "name": "Power 310",
  "state": {
    "current": 89,
    "lastupdated": "2024-06-29T18:34:58.193",
    "power": 4,
    "voltage": 53
  },
  "swversion": "2.9.2_r58",
  "type": "ZHAPower",
  "uniqueid": "18:7a:3e:ff:fe:f7:41:40-01-0b04"
}
```

closed 2024-08-20 by github-actions[bot]

Mimiix, 2024-08-20:
Do we need to keep this open @ebaauw  or is this merged?

ebaauw, 2024-08-20:
Has been merged, see above.