dresden-elektronik/deconz-rest-plugin #7812: nedis zigbee smoke detector zbds10wt

ltspicer, 2024-06-12:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Nedis zigbee smoke detector zbds10wt

### Manufacturer

Nedis

### Model identifier

zbds10wt 

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]
[image]

Mimiix, 2024-06-12:
Please make sure you clicked "read" on the clusters. Also, add all screenshots.

ltspicer, 2024-06-13:
> Please make sure you clicked "read" on the clusters. Also, add all screenshots.

Ist das so ok?

Smanar, 2024-06-13:
Hello, you can try this part for the detection 

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TYZB01_wqcac7lo",
  "modelid": "TS0205",
  "vendor": "Tuya",
  "product": "Smoke Sensor",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_FIRE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2",
            "fn": "zcl:attr"
          }
        },
        {
          "name": "config/enrolled"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/fire"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/test"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 600,
          "max": 43200,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500"
    }
  ]
}
```

But your device have too a siren ?

ltspicer, 2024-06-13:
> Hello, you can try this part for the detection
>
> ```
> {
>   "schema": "devcap1.schema.json",
>   "manufacturername": "_TYZB01_wqcac7lo",
>   "modelid": "TS0205",
>   "vendor": "Tuya",
>   "product": "Smoke [...]

How should I do this part?

Device has a siren.

Smanar, 2024-06-13:
>How should I do this part?

Depend of your OS, on classic OS, just need to create a text file called what_you_want.json with this contain in a "devices" folder, then restart deconz.

> All DDF provided directly with deCONZ typically reside in `/usr/share/deCONZ/devices/` on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. `/home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices`) will override the pre-packaged files to allow users to amend and keep their own files if desired.

ltspicer, 2024-06-13:
I have the deconz adapter in ioBroker.
Just restart this adapter after creation of the json?

ltspicer, 2024-06-13:
$ sudo systemctl status deconz
● deconz.service - deCONZ: ZigBee gateway -- REST API
     Loaded: loaded (/lib/systemd/system/deconz.service; disabled; vendor preset: enabled)
     Active: inactive (dead)

But deCONZ-RaspBee is running...

Smanar, 2024-06-13:
I don't remember for ioBroker, but from my memory they don't use dockers ?*
Wich one "devices" folder have you used ? it was empty ?

You can restart using the console I think
`sudo systemctl restart deconz` For the headless
`sudo systemctl restart deconz-gui` For the desktop

ltspicer, 2024-06-13:
I inserted your code in this file:

[image]

The GUI shows the device:

[image]

But the DDF looks like this:

[image]

ioBroker and the Zigbee gateway (RaspBee) is on the same Raspberry.

ltspicer, 2024-06-13:
Now it is online ^^
I push the test button.

[image]

[image]

Smanar, 2024-06-13:
Nice, so new version with the siren
```

{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TYZB01_wqcac7lo",
  "modelid": "TS0205",
  "vendor": "Tuya",
  "product": "Smoke Sensor",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_FIRE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2",
            "fn": "zcl:attr"
          }
        },
        {
          "name": "config/enrolled"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/fire"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery"
        },
        {
          "name": "state/test"
        }
      ]
    },
    {
      "type": "$TYPE_WARNING_DEVICE",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert"
        },
        {
          "name": "state/reachable"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 600,
          "max": 43200,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500"
    }
  ]
}
```

Do you know how to use the siren ? By security you can't use state/on, but need to use state/alert

ltspicer, 2024-06-13:
If works!!
Thank you.

What's the difference between your 1st and 2nd version?

ltspicer, 2024-06-13:
"Or you can clone the tuya_swversion.js file too."
What/what content to clone to where?
... sorry, my english is terrible.

Smanar, 2024-06-14:
Can just copy the file `tuya_swversion.js` in the same folder than your DDF (the "nedis" folder).
The difference with the previous DDF,now the DDF create 2 entries, a "sensor" for return and a "light" for the siren.

ltspicer, 2024-06-15:
I use the version 1. It is easier and give the only right status.
Thx for support ;)

ltspicer, 2024-06-15:
OK?

[image]

ltspicer, 2024-06-15:
Info for dev:
Can you pls integrate this DDL?
Thx

Smanar, 2024-06-15:
Strange you haven't more file in your "nedis" folder, you need to have at least 2 files https://github.com/dresden-elektronik/deconz-rest-plugin/tree/master/devices/nedis

I can make the PR to suport this device as it, but you don't want to test the siren stuff ?

ltspicer, 2024-06-15:
I had 2 more files. I deleted them myself.
I didn't understand the siren stuff.
The smoke detector works great with the v1.
It already has an integrated 83dB buzzer (sirene).

ltspicer, 2024-06-15:
If there's anything I can test for you, let me know.
But please express yourself precisely (and without abbreviations)!

Smanar, 2024-06-15:
>I didn't understand the siren stuff.

No problem, it's for you, if you don't need it, we can stay like that.
I don't know the device, perhaps the siren is managed internaly and can't be used using zigbee request.

But with the second DDF, you need to see a new entry in your "light" devices, a "Warning device" that can be used to trigger the siren.
You can too make test using the GUI, it's on the cluster 0x0502 "IAS WD".

PR submitted here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7815

ltspicer, 2024-06-15:
With V2 I see only "Warning device". The real device I can not see.
I use V1 ;)

Smanar, 2024-06-15:
>With V2 I see only "Warning device"

HA ? So something is bad, you need to have 2 entries with the V2, 1 in "sensor" and 1 in "light", they are not at same pplace, IDK how your third app manage them.

ltspicer, 2024-06-15:
> Strange you haven't more file in your "nedis" folder, you need to have at least 2 files https://github.com/dresden-elektronik/deconz-rest-plugin/tree/master/devices/nedis
> 
> I can make the PR to suport this device as it, but you don't want to test the siren stuff ?

I restored the 2 files ^^

ltspicer, 2024-06-15:
Is everything ok for you at the moment?
Or do you need more data for a clean implementation?

Smanar, 2024-06-15:
It's for you, not for me ^^, I haven't the device.
For me it's enought to submit a PR for the sensor part, the V1, thx.

If you want to use the siren, can try with the V2 and search where can be the "light entry", can be visible in phoscon/help/API Information/light.
But if you are happy with the V1, no problem for me.

Smanar, 2024-06-15:
Ha nice, you are "mixing" radio and wired device ?

ltspicer, 2024-06-15:
Yes. I have a system from 1998 and new ^^

ltspicer, 2024-06-29:
Here you can see the old fire alarm control panel with my interface PCB (left).
In the meantime, I have registered a second Nedis detector.
It works perfectly.

[image]

Smanar, 2024-06-30:
>with my interface PCB (left).

It's DIY stuff ?

ltspicer, 2024-06-30:
Yes. Schematic and PCB is made by me self.

Smanar, 2024-06-30:
Lol, I don't have enought skill, realy profoessionnal, what are you using for PCB support ? the green "thing" compatible with DIN support

ltspicer, 2024-06-30:
I milled the PCB with my small CNC milling machine.
The profile for the DIN rail is from there:

https://de.aliexpress.com/item/1005005574681794.html

[image]
[image]
[image]

closed 2024-07-29 by github-actions[bot]