dresden-elektronik/deconz-rest-plugin #7819: Philips Hue Adore

GiacomoFilzi, 2024-06-20:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Philips Hue Adore 

### Manufacturer

Philips

### Model identifier

3418231P6

### Device type to add

Light

### Node info

Note sure what to put here.

### Endpoints and clusters

Not sure what to put here.

### Basic

I paired it in offline mode with a Hue dimmer switch, however doesn't get recognized by Phoscon with Conbee II stick.

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Mimiix, 2024-06-20:
Hi,

If you can't pair, please open a forum topic to get it paired first. 

Closing

closed 2024-06-20 by Mimiix

GiacomoFilzi, 2024-06-20:
Sorry, what forum?