dresden-elektronik/deconz-rest-plugin #7821: SONOFF SNZB-03P

febalci, 2024-06-24:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

SONOFF SNZB-03P Motion Sensor [Link](https://sonoff.tech/product/gateway-and-sensors/snzb-03p/)

### Manufacturer

SONOFF

### Model identifier

SNZB-03P

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration
[image]

### Identify
[image]

### Occupancy Sensing
[image]

### IAS Zone
[image]

febalci, 2024-06-24:
I used DDF fromn SNZP-02D as with some changes like below; the 'state/presence' and 'config/battery' works fine. cl:0x406, attribute 0x0020 is the delay time in seconds for occupied to unoccupied delay timeout. Good thing is you can set it down to 5 seconds and it works. I can change it from within phoscon but in this DDF i couldn't make it work via Home Assistant deconz device settings. Also, the state/dark setting is nowhere to be seen in HA deconz settings. So i need to correct 'config/duration', 'state/dark' and the 'bindings' parts somehow...

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "eWeLink",
  "modelid": "SNZB-03P",
  "vendor": "SONOFF",
  "product": "SNZB-03P",
  "sleeper": false,
  "status": "Draft",
  "path": "/devices/snzb-03p.json",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0406"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0107",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0406",
          "0x0500",
          "0xFC11"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl:attr"
          },
          "default": 0
        },
        {
          "name": "config/duration",
          "refresh.interval": 3600,
          "read": {
            "at": "0x0020",
            "cl": "0x0406",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "write": {
            "at": "0x0020",
            "cl": "0x0406",
            "dt": "0x21",
            "ep": 1,
            "eval": "Item.val",
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0020",
            "cl": "0x0406",
            "ep": 1,
            "eval": "Item.val = Attr.val",
            "fn": "zcl:attr"
          },
          "default": 60
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/dark",
          "refresh.interval": 360,
          "read": {
            "at": "0x2001",
            "cl": "0xfc11",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x2001",
            "cl": "0xfc11",
            "ep": 1,
            "eval": "if (Attr.val === 0) { Item.val = true } else { Item.val = false }",
            "fn": "zcl:attr"
          },
          "default": false
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/presence",
          "read": {
            "fn": "none"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0406",
            "ep": 1,
            "eval": "Item.val = Attr.val",
            "fn": "zcl:attr"
          },
          "default": false
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0xFC11",
      "report": [
        {
          "at": "0x2001",
          "dt": "0x20",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0406",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x18",
          "min": 0,
          "max": 1,
          "change": "0x00000001"
        }
      ]
    }
  ]
}```

Smanar, 2024-06-25:
>So i need to correct 'config/duration', 'state/dark' and the 'bindings' parts somehow...

HA don't manage state/dark ?
"config/duration" is not working using API ?
For the bindong

```
      "cl": "0x0406",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x18",
          "min": 1,
          "max": 300
        }
      ]
    },
```

Will be better.

I don't think you need more stuff for bind ?

febalci, 2024-06-25:
> HA don't manage state/dark ? 
> "config/duration" is not working using API ? 
[image]
 
All i got in HA are these. There's no state/dark here. And also config/duration can't read or write to 0x0406/attr:0x0020. 

The 47 secs shown in the image is changed by myself, but in phoscon 0x0020 is still keeps 5 secs value.

Smanar, 2024-06-26:
>All i got in HA are these. There's no state/dark here.

Strange from my memory HA have no problem with them, @Kane610 you have an idea ? a Synchronisation issue ?

>The 47 secs shown in the image is changed by myself, but in phoscon 0x0020 is still keeps 5 secs value.

Your DDF seem good for me, device on battery are boring with that, need to awake them to be sure it work.
You are saying "in phoscon", it mean in phoscon/help/API information/sensor or you mean "deconz" ?

If you try to change the value using deconz, it's updated in the API ?

Kane610, 2024-06-26:
Dark is reported as an attribute on Presence binary sensor and lightlevel sensor right now. Its not its own sensor currently

febalci, 2024-06-27:
> Dark is reported as an attribute on Presence binary sensor and lightlevel sensor right now. Its not its own sensor currently

Yep, found "dark" attribute under "presence"; even tough it always shows "false" even it is in completely dark or bright location.

> You are saying "in phoscon", it mean in phoscon/help/API information/sensor or you mean "deconz" ?

I can change the "delay" value under deConz, not phoscon, sorry for the confusion. I can set the value in deconz, but in order for it to read or write the value, the sensor should be wake up by a movement.

> If you try to change the value using deconz, it's updated in the API ?

I i change the delay value in deconz, no events come up in API, then  wake the sensor up to make it write the value, usual presence event comes up:

```

                    
{
    "20:18:06:816": {
        "e": "changed",
        "id": "136",
        "r": "sensors",
        "state": {
            "dark": false,
            "lastupdated": "2024-06-27T17:18:06.939",
            "presence": true
        },
        "t": "event",
        "uniqueid": "84:27:12:ff:fe:33:23:4b-01-0406"
    },
```
I i change the delay value in Home Assistant then this comes up in the events API:
```
"20:24:59:762": {
        "config": {
            "battery": 100,
            "delay": 56,
            "on": true,
            "reachable": true
        },
        "e": "changed",
        "id": "136",
        "r": "sensors",
        "t": "event",
        "uniqueid": "84:27:12:ff:fe:33:23:4b-01-0406"
    }
```
Even tough i wake up the sensor before changing the delay in HA, it shows in API but not in deconz.

> And your DDF have the "draft" status, you have another one on the one you are using for tests ?

No i am using this draft DDF for tests and i use DDF hybrid mode to load it. Should i change it to gold for tests?

febalci, 2024-06-27:
@Smanar and @Kane610 Thanks for all help, SZNB-03P is working perfectly fine with all its features now including timeout delay setting (min 5 secs) from HA, and dark (true/false) attribute on occupancy; with the DDF below:

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "eWeLink",
  "modelid": "SNZB-03P",
  "vendor": "SONOFF",
  "product": "SNZB-03P",
  "sleeper": false,
  "status": "Bronze",
  "path": "/devices/snzb-03p.json",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0406"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0107",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0406",
          "0x0500",
          "0xFC11"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl:attr"
          },
          "default": 0
        },
        {
          "name": "config/delay",
          "refresh.interval": 3600,
          "read": {
            "at": "0x0020",
            "cl": "0x0406",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "write": {
            "at": "0x0020",
            "cl": "0x0406",
            "dt": "0x21",
            "ep": 1,
            "eval": "Item.val",
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0020",
            "cl": "0x0406",
            "ep": 1,
            "eval": "Item.val = Attr.val",
            "fn": "zcl:attr"
          },
          "default": 60
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/dark",
          "read": {
            "fn": "none"
          },
          "parse": {
            "at": "0x2001",
            "cl": "0xfc11",
            "ep": 1,
            "eval": "if (Attr.val === 0) { Item.val = true } else { Item.val = false }",
            "fn": "zcl:attr"
          },
          "default": false
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/presence",
          "read": {
            "fn": "none"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0406",
            "ep": 1,
            "eval": "Item.val = Attr.val",
            "fn": "zcl:attr"
          },
          "default": false
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0406",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x18",
          "min": 1,
          "max": 300
        },
        {
          "at": "0x0020",
          "dt": "0x21",
          "min": 5,
          "max": 60
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0xfc11",
      "report": [
        {
          "at": "0x2001",
          "dt": "0x20",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

Smanar, 2024-06-27:
Lol, fster than me.
There is no bind for battery, some device can work without, but IDK for this one.

Kane610, 2024-06-27:
Great!

febalci, 2024-06-29:
> There is no bind for battery, some device can work without, but IDK for this one.

You are correct, this also needs the bind for battery. Here is the corrected DDF:

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "eWeLink",
  "modelid": "SNZB-03P",
  "vendor": "SONOFF",
  "product": "SNZB-03P",
  "sleeper": false,
  "status": "Bronze",
  "path": "/devices/snzb-03p.json",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0406"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0107",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0406",
          "0x0500",
          "0xFC11"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "refresh.interval": 7265,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl:attr"
          },
          "default": 0
        },
        {
          "name": "config/delay",
          "refresh.interval": 3600,
          "read": {
            "at": "0x0020",
            "cl": "0x0406",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "write": {
            "at": "0x0020",
            "cl": "0x0406",
            "dt": "0x21",
            "ep": 1,
            "eval": "Item.val",
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0020",
            "cl": "0x0406",
            "ep": 1,
            "eval": "Item.val = Attr.val",
            "fn": "zcl:attr"
          },
          "default": 60
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/dark",
          "read": {
            "fn": "none"
          },
          "parse": {
            "at": "0x2001",
            "cl": "0xfc11",
            "ep": 1,
            "eval": "if (Attr.val === 0) { Item.val = true } else { Item.val = false }",
            "fn": "zcl:attr"
          },
          "default": false
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/presence",
          "read": {
            "fn": "none"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0406",
            "ep": 1,
            "eval": "Item.val = Attr.val",
            "fn": "zcl:attr"
          },
          "default": false
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0406",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x18",
          "min": 1,
          "max": 300
        },
        {
          "at": "0x0020",
          "dt": "0x21",
          "min": 5,
          "max": 60
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0xFC11",
      "report": [
        {
          "at": "0x2001",
          "dt": "0x20",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 0,
          "max": 200,
          "change": "0x00000002"
        }
      ]
    }
  ]
}
```

Smanar, 2024-06-29:
Thx, BTW do you want to make a PR to submit the DDF or want someone make it ?

febalci, 2024-06-30:
Thanks @Smanar, I would appreciate if someone else can handle the PR, I have no idea how to do it...

Smanar, 2024-06-30:
NP, PR done here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7830

closed 2024-07-29 by github-actions[bot]