dresden-elektronik/deconz-rest-plugin #7823: Tuya Watervalve _TZ3000_wpueorev

Mimiix, 2024-06-26:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Tuya

### Manufacturer

_TZ3000_wpueorev

### Model identifier

TS0001

### Device type to add

Other

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off
[image]
[image]

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest
[image]

Device works as i'm on an older deCONZ (v2.22.0) and uses legacy support. However, there';s no DDF for it.

Smanar, 2024-06-26:
So much clone for this device

```
        fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_rk2yzt0u'},
            {modelID: 'TS0001', manufacturerName: '_TZ3000_o4cjetlm'}, {manufacturerName: '_TZ3000_o4cjetlm'},
            {modelID: 'TS0001', manufacturerName: '_TZ3000_iedbgyxt'}, {modelID: 'TS0001', manufacturerName: '_TZ3000_h3noz0a5'},
            {modelID: 'TS0001', manufacturerName: '_TYZB01_4tlksk8a'}, {modelID: 'TS0011', manufacturerName: '_TYZB01_rifa0wlb'},
            {modelID: 'TS0001', manufacturerName: '_TZ3000_5ucujjts'}, {modelID: 'TS0001', manufacturerName: '_TZ3000_h8ngtlxy'},
            {modelID: 'TS0001', manufacturerName: '_TZ3000_w0ypwa1f'}, {modelID: 'TS0001', manufacturerName: '_TZ3000_wpueorev'},
        ],
```

And strange it's working natively, from Z2M it need the tuya unclock sequence, but they probably use it by defaut now.

Smanar, 2024-06-27:
I think something like

```
{
    "schema": "devcap1.schema.json",
    "manufacturername": [
        "_TZ3000_rk2yzt0u",
        "_TZ3000_o4cjetlm",
        "_TZ3000_iedbgyxt",
        "_TYZB01_4tlksk8a",
        "_TZ3000_5ucujjts",
        "_TZ3000_w0ypwa1f",
        "_TZ3000_h3noz0a5",
        "_TZ3000_h8ngtlxy",
        "_TZ3000_wpueorev"
    ],
    "modelid": [
        "TS011F",
        "TS0001",
        "TS0001",
        "TS0001",
        "TS0001",
        "TS0001",
        "TS0001",
        "TS0001",
        "TS0001"
    ],
    "vendor": "Tuya",
    "product": "Smart water-gas Valve",
    "sleeper": false,
    "status": "Gold",
    "subdevices": [
        {
            "type": "$TYPE_ON_OFF_OUTPUT",
            "restapi": "/lights",
            "uuid": [
                "$address.ext",
                "0x01"
            ],
            "items": [
                {
                    "name": "attr/id"
                },
                {
                    "name": "attr/lastannounced"
                },
                {
                    "name": "attr/lastseen"
                },
                {
                    "name": "attr/manufacturername"
                },
                {
                    "name": "attr/modelid"
                },
                {
                    "name": "attr/name"
                },
                {
                    "name": "attr/swversion",
                    "parse": {
                        "fn": "zcl:attr",
                        "ep": 1,
                        "cl": "0x0000",
                        "at": "0x0001",
                        "script": "tuya_swversion.js"
                    },
                    "read": {
                        "fn": "zcl:attr",
                        "ep": 1,
                        "cl": "0x0000",
                        "at": "0x0001"
                    }
                },
                {
                    "name": "attr/type"
                },
                {
                    "name": "attr/uniqueid"
                },
                {
                    "name": "state/on"
                },
                {
                    "name": "config/tuya_unlock"
                },
                {
                    "name": "state/reachable"
                }
            ]
        }
    ],
    "bindings": [
        {
            "bind": "unicast",
            "src.ep": 1,
            "dst.ep": 1,
            "cl": "0x0006",
            "report": [
                {
                    "at": "0x0000",
                    "dt": "0x10",
                    "min": 1,
                    "max": 300
                }
            ]
        }
    ]
}
```

As it work natively, I think it use cluster 0x0006 and not the tuya one, have added the tuya unlock sequence.
We will probably need to use the tuya cluster for special settings.

closed 2024-07-26 by github-actions[bot]