dresden-elektronik/deconz-rest-plugin #7833: Adeo 16A plug power doesn't work

Idaho947, 2024-07-03:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

I just buy adeo 16A plug. The plug is in my network. On\Off work very well. But I've no power. The section is here but nothing change when my PS5 work or not.

### Steps to reproduce the behavior

Add a plug in the network

### Expected behavior

The mesurement work.

### Screenshots

[image]

[image]

[image]

[image]

### Environment

- Host system: Nuc
- Running method: VM
- Firmware version: (26510900)
- deCONZ version : 2.26.3
- Device:  ConBee III
- Do you use an USB extension cable: yes
- Is there any other USB or serial devices connected to the host system? non

Mimiix, 2024-07-03:
Hi :) looks like this device is not supported yet:

https://github.com/dresden-elektronik/deconz-rest-plugin/issues?q=LDSENK02F

I recommend you to create a device request. However: Do the Clusters in deCONZ show any values on the Power measurement?

If that is not the case, they might not support power metering at all.

Idaho947, 2024-07-03:
Yes I just see that. Don't understand why it show up in ha without ddf...
Yes it support power mesurement.

Create the issue : https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7834

closed 2024-07-03 by Idaho947

Smanar, 2024-07-03:
>Yes I just see that. Don't understand why it show up in ha without ddf...

It's because "light" device are whitelisted by defaut but "sensor" device are Blacklisted.
So the  on/off is native but not the measurement.