dresden-elektronik/deconz-rest-plugin #7834: ADEO LDSENK02F 16A plug

Idaho947, 2024-07-03:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Adeo plug 16 A

### Manufacturer

Adeo

### Model identifier

LDSENK02F

### Device type to add

Other

### Node info

[image]

### Endpoints and clusters

[image]

[image]

### Basic

[image]

### Further relevant clusters

### On/Off
[image]

[image]

###Simple metering
[image]

[image]

[image]

### Electrical Measurement
[image]

[image]

[image]

[image]

[image]

[image]

[image]

Smanar, 2024-07-03:
Hello, DDF to try

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "ADEO",
  "modelid": "LDSENK02F",
  "vendor": "Adeo",
  "product": "Plug 16A",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_SMART_PLUG",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on",
          "refresh.interval": 360
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "refresh.interval": 360
        },
        {
          "name": "state/voltage",
          "refresh.interval": 360
        },
        {
          "name": "state/current",
          "refresh.interval": 360
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0702",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x25",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000064"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 1,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    }
  ]
}
```

Like usualy, some vaue can need to be "corrected" (*10 or / 100 for exemple) just give me the value and the needed convertion.

Idaho947, 2024-07-03:
Power is good. I look at the consumption after a couple of hour.

Smanar, 2024-07-04:
50 watt during 30 mn give 0.025 KW/h.
On your capture we have divisor 100, multiplicator 1.

Honnestly not enought consumption to be sure for me ^^

Can replace 

```
        {
          "name": "state/consumption"
        },
```

by

```
{
  "name": "state/consumption",
  "read": {
    "at": "0x0000",
    "cl": "0x0702",
    "ep": 1,
    "fn": "zcl:attr"
  },
  "parse": {
    "at": "0x0000",
    "cl": "0x0702",
    "ep": 1,
    "eval": "Item.val = Attr.val *XX"
  },
  "refresh.interval": 300,
  "default": 0
}
```

With a better convertion than `"eval": "Item.val = Attr.val *XX"`

Idaho947, 2024-07-04:
What do you mean whith :
With a better convertion than "eval": "Item.val = Attr.val *XX" ?

Smanar, 2024-07-04:
For exemple 
`"eval": "Item.val = Attr.val * 10"`

or

`"eval": "Item.val = Attr.val / 10"`

Idaho947, 2024-07-04:
Ok I let it like that one week and see if it's relevant.

Idaho947, 2024-07-25:
Ok everythinks seems OK. You can valid the ddf.

Smanar, 2024-07-25:
So without changes ? the same than here > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7834#issuecomment-2206784102

Idaho947, 2024-07-25:
Yes no change needed

Smanar, 2024-07-25:
DDF created https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7855

closed 2024-08-07 by manup