dresden-elektronik/deconz-rest-plugin #7836: Tuya - PIR 24G Radar Sensor - ZG-204ZM

howiet2011, 2024-07-07:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

ZG-204ZM

### Manufacturer

Tuya

### Model identifier

unknown

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]
[image]

### Further relevant clusters

[image]
[image]
[image]

derdon999, 2024-07-14:
Hi. If you need anybody to test the DDF and device functionality, just tell me as I have 3 of the mmW sensors at home.

wvuyk, 2024-07-18:
Here also, have several of these sensors available, would likt to test also...

Smanar, 2024-07-19:
Can try 

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_kb5noeto",
  "modelid": "TS0601",
  "product": "Tuya PIR 24G Radar Sensor",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_LIGHT_LEVEL_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0400"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "read": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl:attr",
            "script": "tuya_swversion.js"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/dark"
        },
        {
          "name": "config/tholddark"
        },
[... 219 lines cut ...]
          "name": "state/presence",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 1,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": false
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0xEF00"
    }
  ]
}
```

Based on https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/ZY-M100_human_breathing_presence.json

howiet2011, 2024-07-19:
Hey, the sensor is connecting but the data I get unfortunately make no sense.

wvuyk, 2024-07-19:
@Smanar 

A presence entry and a lightlevel are created by the rest plugin. The ZHAPresence seems to update, but the trigger to presence=true is delayed big time, like a minute or so. The ZHALightlevel part is not updating at all it seems, not even after wating quite some time. The battery level is not present at all in the REST API output. This is a battery device, so it should be there right?

I will add the REST API output and also the trace level output that is generated after pressing the button for triggering all the reports on the Tuya branch, I hope this will help you?

`ZHALightLevel

**************************** Object info *****************************
{
"config": {
"on": true,
"reachable": true,
"tholddark": 12000,
"tholdoffset": 7000
},
"etag": "ff39e11bb319e0fc05210039a97819b6",
"lastannounced": null,
"lastseen": "2024-07-19T19:01Z",
"manufacturername": "_TZE200_kb5noeto",
"modelid": "TS0601",
"name": "Beweging mmw 1 Light level",
"state": {
"dark": false,
"daylight": false,
"lastupdated": "none",
"lightlevel": 0,
"lux": 0
},
"type": "ZHALightLevel",
"uniqueid": "a4:c1:38:dd:14:17:fc:34-01-0400"
}
************************* End Object info ***************************

ZHAPresence

**************************** Object info *****************************
{
"config": {
"fadingtime": 30,
"on": true,
"reachable": true,
"sensitivity": 6,
"triggerdistance": "6"
},
"etag": "ff39e11bb319e0fc05210039a97819b6",
"lastannounced": null,
"lastseen": "2024-07-19T19:01Z",
"manufacturername": "_TZE200_kb5noeto",
"modelid": "TS0601",
"name": "Beweging mmw 1 Motion",
"state": {
"errorcode": "Unknown",
"lastupdated": "2024-07-19T18:58:51.718",
"presence": true,
"targetdistance": 0
},
"type": "ZHAPresence",
"uniqueid": "a4:c1:38:dd:14:17:fc:34-01-ef00"
}
************************* End Object info ***************************`

Output for triggering all Tuya reports:

`21:11:38:129 TY_DATA_REPORT: seq 233, dpid: 0x01, type: 0x04, length: 1, val: 1
21:11:38:130 TY_DATA_REPORT: seq 233, dpid: 0x01, type: 0x04, length: 1, val: 1
21:11:38:130 TY_DATA_REPORT: seq 233, dpid: 0x01, type: 0x04, length: 1, val: 1
21:11:38:131 TY_DATA_REPORT: seq 233, dpid: 0x01, type: 0x04, length: 1, val: 1
21:11:38:131 TY_DATA_REPORT: seq 233, dpid: 0x01, type: 0x04, length: 1, val: 1
21:11:38:132 TY_DATA_REPORT: seq 233, dpid: 0x01, type: 0x04, length: 1, val: 1
[... 45 lines cut ...]
21:11:46:738 TY_DATA_REPORT: seq 244, dpid: 0x65, type: 0x04, length: 1, val: 1
21:11:46:739 TY_DATA_REPORT: seq 244, dpid: 0x65, type: 0x04, length: 1, val: 1
21:11:46:740 TY_DATA_REPORT: seq 244, dpid: 0x65, type: 0x04, length: 1, val: 1
21:11:46:741 TY_DATA_REPORT: seq 244, dpid: 0x65, type: 0x04, length: 1, val: 1
21:11:46:741 TY_DATA_REPORT: seq 244, dpid: 0x65, type: 0x04, length: 1, val: 1
21:11:47:881 TY_DATA_REPORT: seq 245, dpid: 0x6A, type: 0x02, length: 4, val: 1680
21:11:47:881 TY_DATA_REPORT: seq 245, dpid: 0x6A, type: 0x02, length: 4, val: 1680
21:11:47:881 TY_DATA_REPORT: seq 245, dpid: 0x6A, type: 0x02, length: 4, val: 1680
21:11:47:882 TY_DATA_REPORT: seq 245, dpid: 0x6A, type: 0x02, length: 4, val: 1680
21:11:47:882 TY_DATA_REPORT: seq 245, dpid: 0x6A, type: 0x02, length: 4, val: 1680
21:11:47:882 TY_DATA_REPORT: seq 245, dpid: 0x6A, type: 0x02, length: 4, val: 1680
21:11:48:698 ZCL attribute report 0xA4C138DD1417FC34 for cluster: 0x0400, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
21:11:48:948 TY_DATA_REPORT: seq 247, dpid: 0x6A, type: 0x02, length: 4, val: 1690
21:11:48:948 TY_DATA_REPORT: seq 247, dpid: 0x6A, type: 0x02, length: 4, val: 1690
21:11:48:948 TY_DATA_REPORT: seq 247, dpid: 0x6A, type: 0x02, length: 4, val: 1690
21:11:48:949 TY_DATA_REPORT: seq 247, dpid: 0x6A, type: 0x02, length: 4, val: 1690
21:11:48:949 TY_DATA_REPORT: seq 247, dpid: 0x6A, type: 0x02, length: 4, val: 1690
21:11:48:949 TY_DATA_REPORT: seq 247, dpid: 0x6A, type: 0x02, length: 4, val: 1690
21:11:50:015 ZCL attribute report 0xA4C138DD1417FC34 for cluster: 0x0400, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
21:11:52:370 TY_DATA_REPORT: seq 249, dpid: 0x65, type: 0x04, length: 1, val: 3
21:11:52:372 TY_DATA_REPORT: seq 249, dpid: 0x65, type: 0x04, length: 1, val: 3
21:11:52:373 TY_DATA_REPORT: seq 249, dpid: 0x65, type: 0x04, length: 1, val: 3
21:11:52:374 TY_DATA_REPORT: seq 249, dpid: 0x65, type: 0x04, length: 1, val: 3
21:11:52:374 TY_DATA_REPORT: seq 249, dpid: 0x65, type: 0x04, length: 1, val: 3
21:11:52:375 TY_DATA_REPORT: seq 249, dpid: 0x65, type: 0x04, length: 1, val: 3:`

Smanar, 2024-07-20:
>This is a battery device, so it should be there right?

Right, have (tried o) add it on previous DDF

>The ZHAPresence seems to update, but the trigger to presence=true is delayed big time, like a minute or so

It's not reversed ? I don't see a setting o the device that can delay the detection, so can be the "no detection" after a delay ?

>The ZHALightlevel part is not updating at all it seems

Yeah, defaut value, the code use a common JS file 
```
          "parse": {
            "dpid": 106,
            "script": "../generic/illuminance_cluster/lux_to_lightlevel.js",
            "fn": "tuya"
          },
```
Perhaps a folder path issue ? where have you put the DDF ?
Else with deconz on help/debug view with flag "DDF' you will see usefull debug (whern the DDF core try to make the magic)

wvuyk, 2024-07-20:
Battery is still not reported at this moment, do I need to change something for that?

I had the file placed on the  .local\share\dresden-elektronik\deCONZ\devices area, where DDF's are placed when you edit the DDF for the first time... I will move the DDF to the installed area for the correction?

I will try to recheck on the motion. I indeed would expect a delay before nomotion occurs, motion should occur immediately... will monitor this a bit more.

> 21:11:39:632 TY_DATA_REPORT: seq 233, dpid: 0x01, type: 0x04, length: 1, val: 1
> 
> This is the dpid for detection, need to have this line at detection.

That is the trigger for motion? That will help to check indeed.. that way I can exclude an delay to my plugin. Thanks!

Smanar, 2024-07-20:
>Battery is still not reported at this moment, do I need to change something for that?

Just need to replace the DDF. For me it need to work, the code
```
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 121, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
```
Your log
`21:11:41:414 TY_DATA_REPORT: seq 235, dpid: 0x79, type: 0x02, length: 4, val: 100`

>I had the file placed on the .local\share\dresden-elektronik\deCONZ\devices area, where DDF's are placed when you edit the DDF for the first time... I will move the DDF to the installed area for the correction?

Pls try with the file in the "tuya" folder, don't remeber to remove the previous one.

>I will try to recheck on the motion. I indeed would expect a delay before nomotion occurs, motion should occur immediately... will monitor this a bit more.

Ha ok, I have understant it's the motion that is delayed.
So you can increase this delay with 

```
          "name": "config/fadingtime",
          "description": "Fading time : target disapperance delay time in second (from 0 to 15000). Default 90s",
```

wvuyk, 2024-07-20:
I have added the battery entry, this is now reported correctly. 
After moving the file to the installation area, it reported the ZHALightlevel fine - this one is real fast with reporting! Makes me wonder about battery life here. We will have to see on that part.

And ZHAPresence after a reset of the device is reprting immediately now as expectd... So I do think it is complete?

Below is the DDF I have now. I think it should be good now?

`{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZE200_kb5noeto",
  "modelid": "TS0601",
  "product": "Tuya PIR 24G Radar Sensor",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_LIGHT_LEVEL_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0400"
      ],
      "items": [
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
 	{
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 121, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "read": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0001",
            "cl": "0x0000",
            "ep": 1,
            "fn": "zcl:attr",
            "script": "tuya_swversion.js"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
[... 232 lines cut ...]
          "name": "state/lastupdated"
        },
        {
          "name": "state/presence",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 1,
            "eval": "Item.val = Attr.val;",
            "fn": "tuya"
          },
          "default": false
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0xEF00"
    }
  ]
}`

Smanar, 2024-07-21:
>this one is real fast with reporting! Makes me wonder about battery life here. We will have to see on that part.

Yes and it's tuya cluster, you have no settings on them, all is native, no binds, no report settings.

>And ZHAPresence after a reset of the device is reprting immediately now as expectd... So I do think it is complete?

>Below is the DDF I have now. I think it should be good now?

As you want  ^^.
If it's ok for you , you can make a PR or I can make it if you want ?

wvuyk, 2024-07-23:
I never did a PR yet, so if you can do this, it would be helpful for me. 
I do have one question though... I have several lightsensors in my house, included in Philips Motion sensors, LUMI devices (als a dedicated light sensor) I notice this Tuya sensor is very much off compared to the LUMI sensor (lumi.sen_ill.mgl01).
The tuya device shows a lux value of 2116 Lux, the LUMI sensor, laying next to it, shows 37 Lux. I am not expecting they should match, but at least be close to its value.

Also, looking at the comments in the DDF for lux, it suggests its max value is 2000, but the value from the Tuya device smees to neve drop below 2000? Is it possible the javascript being used is returning a wrong value?

wst01, 2024-07-24:
I see the same issue. The lx value makes no sense.
I have an (usually very precise) Homematic sensor at exactly same position and it shows 92,791lx (which makes sense as it wasn't much light in the room. The Tuya shows 2216,1lx. 
Another minor issue is that the Product in Phoscon App is shown as "Thermostat".

Smanar, 2024-07-24:
>Another minor issue is that the Product in Phoscon App is shown as "Thermostat".

This depend of phoscon, if I m right the DDF create a ZHALightLevel and a ZHAPresence ?

About the convertion, IDK, the device return

`21:11:47:881 TY_DATA_REPORT: seq 245, dpid: 0x6A, type: 0x02, length: 4, val: 1680`

So the DDF take the value 1680 and use this value for state/lux (without change) and this script to compute state/lightlevel https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/generic/illuminance_cluster/lux_to_lightlevel.js

10 .000 lux is outside with sun
1 000 lux indoor with lot of light.
 300 lux  indoor, with "standard" light.

I can use a convertion if needed, but need to make try to know mini/maxi.

Not a problem with mistake beetween lightlevel and lux ? Can you compare the lux value (raw value without convertion) with the light level on the  LUMI sensor ?

wvuyk, 2024-07-24:
I am checking both sensors I mentioned in my earlier post here, they are still laying next to each other.
The lumi.sen_ill.mgl01 is reporting 15911 in the JSON lightlevel attribute and 38 as lux.
The "_TZE200_kb5noeto is reporting 32908 in the JSON lightlevel attribute and 1953 as lux.

I did a trigger report on the tuya cluster and think I have the line you were posting

`21:21:14:596 TY_DATA_REPORT: seq 131, dpid: 0x6A, type: 0x02, length: 4, val: 1885
`
This was a minute r two later, so there migh be a small difference in the results. But this looks like the value is not converted at all? The DDF is on the production area still so I guess the js file is still executed?

Does this help in the search?

Smanar, 2024-07-25:
>But this looks like the value is not converted at all?

Yes exactly the state/lux  for the _TZE200_kb5noeto  is a raw value.

Can try with a /50 (38 * 50 = 1900), but not sure it will be proportionnal.

Try with replacing

```
        {
          "name": "state/lux",
          "read": {
            "fn": "none"
          },
          "parse": {
            "dpid": 106,
            "script": "../generic/illuminance_cluster/lux_to_lightlevel2.js",
            "fn": "tuya"
          },
          "default": 0
        }
```

and create the file lux_to_lightlevel2.js in the same folder with

```
const tholddark = R.item('config/tholddark').val;
const tholdoffset = R.item('config/tholdoffset').val;
const lux = Attr.val/50;
var ll = 0;

if (lux > 0 && lux < 0xffff) {
    ll = Math.round(10000 * Math.log10(lux) + 1);
}

R.item('state/lightlevel').val = ll;
R.item('state/dark').val = ll <= tholddark;
R.item('state/daylight').val = ll >= tholddark + tholdoffset;

Item.val = lux;
```

wvuyk, 2024-07-25:
I did the change and now I get values 26 on the lumi.sen_ill.mgl01  and 34 on the _TZE200_kb5noeto. This is a lot closer than before... I will do a few more checks during daylight to see if the values stay close. I guess we will never make this a 100% match.

GitHubStephan, 2024-07-30:
Tried the DDF mentioned in this post but it will not work with it. Saved it as json in share and .local folder and tried to load DDF in deconz directly (failed to load message). What can i do? Used copy&paste to create my json.

Smanar, 2024-07-30:
This one > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7836#issuecomment-2239723513 ?

I don't see typo on my side.

wvuyk, 2024-07-30:
I have the same issue after upgrading to 2.27.6. because of the upgrade I lost my files on the production devices. I tried to copy the file from my windows system to the raspberry pi. Got the error trying to hot reload. Then I opened a sudo nano command in terminal and pasted the lines and saved it (I am a nitwit on linux, but this at least I know). Then I tried a hot reload with the same result, error.

Then I tried to upload the file with the phoscon function to update a DDF, offering the windows file I mentioned earlier.. smae result. Not sure, but I think this might be something with this latest deCONZ version?

taragor, 2024-07-30:
Same issue here. However I am on version 2.27.5. after trying some other custom files, it seems none of the DDFs I try to load are working. Also tried different file encodings and line endings, however I do still get the error

howiet2011, 2024-07-31:
I copied the data from here [https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7836#issuecomment-2241097763](url) , created a ddf / json and have the same problem. Working with version 2.27.5.

wst01, 2024-07-31:
I had the same issue, in the "values" section is no escaping which causes troubles.
I've copied the section from an previous post by @Smanar where it was correctly escaped.

This one should work:

```
{
"schema": "devcap1.schema.json",
"manufacturername": "_TZE200_kb5noeto",
"modelid": "TS0601",
"product": "Tuya PIR 24G Radar Sensor",
"sleeper": false,
"status": "Gold",
"subdevices": [
{
"type": "$TYPE_LIGHT_LEVEL_SENSOR",
"restapi": "/sensors",
"uuid": [
"$address.ext",
"0x01",
"0x0400"
],
"items": [
{
"name": "attr/lastseen"
},
{
"name": "attr/manufacturername"
},
{
"name": "attr/modelid"
},
{
"name": "attr/name"
},
{
"name": "config/battery",
"parse": {"fn": "tuya", "dpid": 121, "eval": "Item.val = Attr.val;" },
"read": {"fn": "none"},
"default": 0
},
{
"name": "attr/swversion",
"refresh.interval": 86400,
"read": {
"at": "0x0001",
"cl": "0x0000",
"ep": 1,
"fn": "zcl:attr"
},
"parse": {
"at": "0x0001",
"cl": "0x0000",
"ep": 1,
"fn": "zcl:attr",
"script": "tuya_swversion.js"
}
},
{
"name": "attr/type"
},
{
"name": "attr/uniqueid"
},
{
"name": "config/on"
},
{
"name": "config/reachable"
},
{
"name": "state/lastupdated"
},
{
"name": "state/dark"
},
{
"name": "config/tholddark"
},
{
"name": "config/tholdoffset"
},
{
"name": "state/daylight"
},
{
"name": "state/lightlevel"
},
{
"name": "state/lux",
"description": "The current light intensity in Lux (max is 2000)",
"read": {
"fn": "none"
},
"parse": {
"dpid": 106,
"script": "../generic/illuminance_cluster/lux_to_lightlevel.js",
"fn": "tuya"
},
"default": 0
}
]
},
{
"type": "$TYPE_PRESENCE_SENSOR",
"restapi": "/sensors",
"uuid": [
"$address.ext",
"0x01",
"0xef00"
],
"items": [
{
"name": "attr/id"
},
{
"name": "attr/lastannounced"
},
{
"name": "attr/lastseen"
},
{
"name": "attr/manufacturername"
},
{
"name": "attr/modelid"
},
{
"name": "attr/name"
},
{
"name": "config/battery",
"parse": {"fn": "tuya", "dpid": 121, "eval": "Item.val = Attr.val;" },
"read": {"fn": "none"},
"default": 0
},
{
"name": "attr/swversion",
"refresh.interval": 86400,
"read": {
"at": "0x0001",
"cl": "0x0000",
"ep": 1,
"fn": "zcl:attr"
},
"parse": {
"at": "0x0001",
"cl": "0x0000",
"ep": 1,
"fn": "zcl:attr",
"script": "tuya_swversion.js"
}
},
{
"name": "attr/type"
},
{
"name": "attr/uniqueid"
},
{
"name": "config/fadingtime",
"range": [
0,
15000
],
"read": {
"fn": "none"
},
"write": {
"dpid": 102,
"dt": "0x2b",
"eval": "Item.val;",
"fn": "tuya"
},
"parse": {
"dpid": 102,
"eval": "Item.val = Attr.val;",
"fn": "tuya"
},
"default": 90
},
{
"name": "config/on"
},
{
"name": "config/reachable"
},
{
"name": "config/sensitivity",
"range": [
1,
9
],
"read": {
"fn": "none"
},
"write": {
"dpid": 2,
"dt": "0x2b",
"eval": "Item.val;",
"fn": "tuya"
},
"parse": {
"dpid": 2,
"eval": "Item.val = Attr.val;",
"fn": "tuya"
},
"default": 7
},
{
"name": "config/triggerdistance",
"description": "Far detection sensitivity",
"read": {
"fn": "none"
},
"write": {
"dpid": 4,
"dt": "0x2b",
"eval": "Far = ['0','1','2','3','4','5','6','7','8','9','10']; Attr.val = Math.max(0,Far.indexOf(Item.val)*100);",
"fn": "tuya"
},
"parse": {
"dpid": 4,
"eval": "Item.val = String(Attr.val/100);",
"fn": "tuya"
},
"values": [
            [
              "\"1\"",
              "1m"
            ],
            [
              "\"2\"",
              "2m"
            ],
            [
              "\"3\"",
              "3m"
            ],
            [
              "\"4\"",
              "4m"
            ],
            [
              "\"5\"",
              "5m"
            ],
            [
              "\"6\"",
              "6m"
            ],
            [
              "\"7\"",
              "7m"
            ],
            [
              "\"8\"",
              "8m"
            ],
            [
              "\"9\"",
              "9m"
            ],
            [
              "\"10\"",
              "10m"
            ],
            [
              "\"0\"",
              "Off"
            ]
          ],
"default": "10"
},
{
"name": "state/targetdistance",
"description": "Distance from detected target in cm (from 0 to 1000).",
"read": {
"fn": "none"
},
"parse": {
"dpid": 9,
"eval": "Item.val = Attr.val;",
"fn": "tuya"
},
"default": 0
},
{
"name": "state/errorcode",
"description": "Self check result code. \n0 = checking, 1 = check_success, 2 = check_failure, 3 = others, 4 = comm_fault, 5 = radar_fault",
"read": {
"fn": "none"
},
"parse": {
"dpid": 6,
"eval": "Item.val = String(Attr.val);",
"fn": "tuya"
},
"default": "Unknown"
},
{
"name": "state/lastupdated"
},
{
"name": "state/presence",
"read": {
"fn": "none"
},
"parse": {
"dpid": 1,
"eval": "Item.val = Attr.val;",
"fn": "tuya"
},
"default": false
}
]
}
],
"bindings": [
{
"bind": "unicast",
"src.ep": 1,
"cl": "0xEF00"
}
]
}

```

GitHubStephan, 2024-07-31:
Thats it. Its working now! Thank you

Smanar, 2024-08-01:
And about the ratio convertion from here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7836#issuecomment-2250703791 ?
the division by 50 is fine ?

wvuyk, 2024-08-01:
It is close enough to the most of my lightsensors 30 lux on the  lumi.sen_ill.mgl01 and 26 at the same time on the _TZE200_kb5noeto. During day time they stay close too, so I guess it is ok this way?

GitHubStephan, 2024-08-02:
Ah, i don't see before that you have to create an extra script file for this. I will try but i am sure that division by 50 will be ok. Will this be automatic included in further deconz updates?

Smanar, 2024-08-02:
As you want.
If eveybody is ok, can make a PR to submit the device, else I can make it if needed.
And yes, better to make an extra script for the /50, for me too much code to add it direclty in the DDF.

wvuyk, 2024-08-02:
I see no issues with that 👍

Smanar, 2024-08-03:
PR done here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7865

closed 2024-08-07 by manup

Daniel-Ebert, 2024-11-16:
Hi there, I just ordered a Tuya _TZE200_kb5noeto version and I'm wondering if / and if yes how I can configure this presence detector when using it with deconz. In the Tuya manual, there's a QR-Code to Scan, linking to this page:
https://doc.szalarm.com/zg-205Z/

If I know click on the docs-link for the ZG-204ZM page, I get this document here: https://doc.szalarm.com/zg-205Z/doc/ZG-204ZM.pdf

There is also a screenshot of the Zigbee2MQTT UI where you can configure the settings of this presence detector:
[image]

Is it also possible to configure this via deconz? I am using deconz on my raspberry using the docker image. I already opened the VNC connection but could not find where to configure these settings for my sensor.

Best regards

Daniel

Daniel-Ebert, 2024-11-18:
Should I maybe rather post in a new thread/is it even possible with deconz to edit such parameters?

Smanar, 2024-11-18:
Just look at field presenty in the DDF.

Daniel-Ebert, 2024-12-01:
> Just look at field presenty in the DDF, you will see wich one are supported ATM.

Thank you for your quick answer! I must say, I'm not that familiar with the technical views/settings of deconz or DDFs.

But from your answer I guess the DDF defines some schema that can to some extent be edited via the deconz UI (not the webapp, but the native program), right?
[image]

So for example, if I want to edit the sensitivity I change this value here, right? Do I need to activate "Awake on receive" for instance?

Best regards

Daniel

Smanar, 2024-12-01:
Your capture is only for DDF, so no impact on deconz, only on the API.

>I guess the DDF defines some schema that can to some extent be edited via the deconz UI (not the webapp, but the native program)

More the reverse ^^, it have an impact on the API, so on all third app, like the webapp
Deconz show only the reality, not "custom cooking".

Here you have a tuya device, so you have only 1 command possible, so you need to know all the complete request if you want to do with the GUI, clic on the cluster "tuya specific", and you will have the only thing you can do with it.

The DDF permit doing custom request on the tuya cluster, with different dpid and values.