dresden-elektronik/deconz-rest-plugin #7837: trådfri on/off buttons causing other devices to report an incorrect state

bmesuere, 2024-07-08:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

I have a conbee 2 which I use in combination with Home Assistant through the deconz addon. My network consists out of ~50 devices from various manufacturers (Hue, Osram, Ikea) and has been rock solid since I started using it. Recently, however, I started having problems when using my trådfri on/off buttons. The buttons still work as expected, but they are causing problems with the rest of my devices. When I press a button, often many unrelated devices suddenly report as "on" while they are still off. They are not actually turned on, but their state is reported as such in both home assistant as well as the phoscon web application. After a while, their reported state normalizes.

### Steps to reproduce the behavior

1. Press the button
2. The expected event is triggered
3. Often (50+%) of the time, the _reported_ state (and not actual state) of one or more other unrelated devices is changed
4. After a while, their reported state returns to normal

### Expected behavior

I expect 3 not to happen.

### Screenshots

[image]

The 2 events in the middle are expected (button press + automation triggering), the others are unexpected. Below you can see the the wrongly reported states at the same timestamp of the button press. Above, you can see the state restoring after a few seconds/minutes.

### Environment

- Host system: NUC
- Running method: Home Assistent deCONZ Add-on
- Firmware version: 26720700
- deCONZ version (not Home assistant Addon version!): 2.25.3
- Device: ConBee II
- Do you use an USB extension cable: yes
- Is there any other USB or serial devices connected to the host system? a samsung T7 SSD

### deCONZ Logs

Relevant section in the logs (sorry for the screenshot)
[image]

If needed, I can reproduce this and copy other logs.

manup, 2024-07-09:
I think this is already fixed in later version [v2.26.1-beta](https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.26.1-beta)

There was a problem with group bindings causing the switch to spam the group 0 https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7643

The next stable version will be published soon and hopefully picked up by the HA AddOn.

bmesuere, 2024-07-09:
@manup thanks for the heads up!

If I understand the linked issue correctly, will updating the firmware of the switches also fix the problem?

manup, 2024-07-10:
Yeah I think that may work as well since the new firmware doesn't do much group stuff.
Note however that the new versions also lose the ability to control light groups directly even when the gateway is offline.

bmesuere, 2024-07-10:
I have updated the firmware of all switches last night and it indeed seems to have fixed the problem I had. Since this is probably also already fixed from the deconz side in a beta version, I will close this issue. Thanks for helping me, @manup!

closed 2024-07-10 by bmesuere