dresden-elektronik/deconz-rest-plugin #7842: SonOff SNZB-05P Water Leak Sensor

madisdev, 2024-07-15:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Zigbee Water Leak Sensor

### Manufacturer

SonOff

### Model identifier

SNZB-05P

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

madisdev, 2024-07-15:
I feel I could probably create the json and PR if someone could give me some pointers of what needs to be done.

Smanar, 2024-07-18:
For me better to use

  "sleeper": true,
  "status": "Gold",

```
        {
          "name": "state/water",
          "parse": {
            "fn": "ias:zonestatus",
            "mask": "alarm1,alarm2"
          },
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0500",
            "at": "0x0002"
          },
          "awake": true,
          "refresh.interval": 84600
        }
```
But only if just 
```
          {
            "name": "state/water",
            "awake": true
          }
```
not working

You miss the bind part too
```

  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 3600,
          "max": 43200,
          "change": "0x00000002"
        }
      ]
    }
  ]
```

madisdev, 2024-07-18:
How should I verify if the state/water isnt enough? I can see it updates state on contact with water. I am not sure what the refresh.interval or awake: true is accomplishing. Can you elaborate, I would like to learn. Thanks.

Smanar, 2024-07-18:
You can't test the DDF ?
Make a test with just 
```
          {
            "name": "state/water",
            "awake": true
          }
```

The code will use defaut value for the rest, if it don't work, need to use custom settings so the big one for exemple 

```
        {
          "name": "state/water",
          "parse": {
            "fn": "ias:zonestatus",
            "mask": "alarm1,alarm2"
          },
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0500",
            "at": "0x0002"
          },
          "awake": true,
          "refresh.interval": 84600
        }
```

You are using `"refresh.interval": 1` not possible for a sleeping device, the report need to be "natural" using bind/report setting.
And in the situation the report don't work, the poll can't work too, as you haven't set the "read" part

Smanar, 2024-07-18:
"parse" is used with incoming request from the device.
"read" is used when deconz ask for information itself (not realy effective on slepping device).

The natural working mode is with value set in bind.
```
          "min": 3600,
          "max": 43200,
          "change": "0x00000002"
```
You are asking the device to make report a minimum every 3600s, at maximum 43200 s without report, or if the value move with a difference of 2.

If it don't work, there is a watchdog, deconz will use the "refresh.interval" and force a "read", this is a security, never used if possible.

```
            "fn":fonction
            "ep": endpoint
            "cl": cluster
            "at": attribute
```
There are the same you can see usng deconz, if you dont set them, deconz will use defaut one, generaly it's a better method.

closed 2024-08-14 by manup