dresden-elektronik/deconz-rest-plugin #7845: failed to open json

MSL-DA, 2024-07-19:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

When I try to open a custom ddf I get this error: `faild to open /opt/deCONZ/devices/test.json`

> 

### Steps to reproduce the behavior

Open deCONZ GUI, select node, edit ddf, open /opt/deCONZ/devices/...json

### Expected behavior

open and load test.json

### Environment

```
- Host system: (Raspberry Pi)
- Running method: (deconz-docker container)
- Firmware version: (26530900)
- deCONZ version (2.27.5)
- Device: (ConBee III)
- Do you use an USB extension cable: (yes)
- Is there any other USB or serial devices connected to the host system? No
```

MSL-DA, 2024-07-21:
Custom .ddf is not preferred...no matter what I do, .ddf is selected from `/usr/share/devices/deCONZ`, not `/opt/deCONZ/devices`

Mimiix, 2024-07-21:
Can you share the DDF your trying to load?

MSL-DA, 2024-07-21:
It's not just one DDF, it applies to all of them, just as no custom DDF in /opt/deCONZ/devices is used

but here is an example

```
{
  "schema": "devcap1.schema.json",
  "uuid": "d9e2a290-e2b4-42de-bd95-4b46b4f0a142",
  "manufacturername": "$MF_LUMI",
  "modelid": "lumi.switch.n0agl1",
  "vendor": "Xiaomi",
  "product": "Aqara T1 single rocker switch (with neutral wire) SSM-U01",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_LIGHT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {
            "at": "0x00f7",
            "ep": 1,
            "fn": "xiaomi:special",
            "idx": "0x08",
            "script": "xiaomi_swversion.js"
          },
          "read": {
            "fn": "none"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert"
        },
        {
          "name": "state/on",
          "refresh.interval": 300
        },
        {
          "name": "state/reachable"
        }
      ]
    },
     {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x15",
        "0x000C"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0100",
        "endpoint": "0x15",
        "in": [
          "0x000C"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {
            "at": "0x00f7",
            "ep": 1,
            "fn": "xiaomi:special",
            "idx": "0x08",
            "script": "xiaomi_swversion.js"
          },
          "read": {
            "fn": "none"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current",
          "parse": {
            "at": "0x00F7",
            "eval": "Item.val = Math.round(Attr.val);",
            "fn": "xiaomi:special",
            "idx": "0x97"
          },
          "read": {
            "fn": "none"
          }
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "refresh.interval": 10,
          "read": {
            "at": "0x0055",
            "cl": "0x000C",
            "ep": 21,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0055",
            "cl": "0x000C",
            "ep": 21,
            "eval": "Item.val = Math.round(Attr.val);"
          }
        },
        {
          "name": "state/voltage",
          "parse": {
            "at": "0x00F7",
            "eval": "Item.val = Math.round(Attr.val / 10);",
            "fn": "xiaomi:special",
            "idx": "0x96"
          },
          "read": {
            "fn": "none"
          }
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x1F",
        "0x000C"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0100",
        "endpoint": "0x1F",
        "in": [
          "0x000C"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {
            "at": "0x00f7",
            "ep": 1,
            "fn": "xiaomi:special",
            "idx": "0x08",
            "script": "xiaomi_swversion.js"
          },
          "read": {
            "fn": "none"
          }
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "refresh.interval": 300,
          "read": {
            "at": "0x0055",
            "cl": "0x000C",
            "ep": 31,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0055",
            "cl": "0x000C",
            "ep": 31,
            "eval": "Item.val = Math.round(Attr.val * 1000);"
          }
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ]
}
```

MSL-DA, 2024-07-21:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/xiaomi/xiaomi_ssm-u01_t1_switch.json

SwoopX, 2024-07-21:
I can confirm this. Doesn't matter if it's a correct or erroneous DDF

MSL-DA, 2024-07-21:
I experienced the problem before version 2.27.5, so it's not a new bug. But I know it is after version 2.26.3

Mimiix, 2024-07-22:
> I can confirm this. Doesn't matter if it's a correct or erroneous DDF

Is this the first report ?!

manup, 2024-07-22:
Hi sorry my bad, this is a known open item for v2.28.x

Currently the loading via "Open" menu entry in DDF editor isn't implemented/active. This happened due the refactor of the whole loading process a few versions back.

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/9e46fe3de0643283f504ad6b6faa5ef756875c91/device_descriptions.cpp#L1439-L1445

As a workaround if you have a device using the DDF in your network, you can open the DDF by:

Selecting the device node, press **Ctrl+E**

or via

The node context menu &rarr; *Edit DDF*

For this the DDF needs to be in the right location and loaded on startup.

manup, 2024-07-22:
But in addition, the load order of the raw JSON needs to be checked too to prefer user directory over system location, I'm on it..

MSL-DA, 2024-07-22:
> Selecting the device node, press **Ctrl+E**
> 
> or via
> 
> The node context menu → _Edit DDF_
> 
> For this the DDF needs to be in the right location and loaded on startup.

This is exactly what I can't do after version 2.26.3. I can open the DDF editor but I get the error when I open a .ddf. I can save a custom DDF, and as I said, it is not selected on restart

Like neither custom DDF nor bundle is preferred from /opt/deCONZ/ it is not only from /opt/deCONZ/devices/ but also /opt/deCONZ/bundles

manup, 2024-07-24:
Not sure where the /opt path comes from, but by default the user location is in `$HOME/.local/share/dresden-elektronik/deCONZ/devices`. Perhaps inside the Docker container this is `/root/.local/share/dresden-elektronik/deCONZ/devices`?

MSL-DA, 2024-07-25:
> Not sure where the /opt path comes from, but by default the user location is in `$HOME/.local/share/dresden-elektronik/deCONZ/devices`. Perhaps inside the Docker container this is `/root/.local/share/dresden-elektronik/deCONZ/devices`?

Please ignore https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7845#issuecomment-2243887146

MSL-DA, 2024-07-27:
> As a workaround if you have a device using the DDF in your network, you can open the DDF by:
> 
> Selecting the device node, press **Ctrl+E**
> 
> or via
> 
> The node context menu → _Edit DDF_
> 
> For this the DDF needs to be in the right location and loaded on startup.

I might add that custom DDF is selected as intended with version 2.27.6

Although I can't open a DDF `The node context menu → _Edit DDF_` , I close as resolved

closed 2024-07-27 by MSL-DA