dresden-elektronik/deconz-rest-plugin #7848: Neo Temperature Sensor NAS-TH07BH

pakwan, 2024-07-20:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Neo Smart Temperature and Humidity Sensor NAS-TH07BH

### Manufacturer

_TZ3000_utwgoauk

### Model identifier

TS0201

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

_**Note: This device can be powered via AA battery or alternatively via micro USB cable.**_

[image]

Link to details of the device as advertised by seller:
https://www.galaxus.at/en/s1/product/neo-smart-temperature-and-humidity-sensor-homekit-nas-th07bh-zigbee-smart-home-hubs-44742350

Smanar, 2024-08-11:
Hello, probably just a new clone.
Can try to add your device to this list https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_TS0201_temp_hum_sensor.json

You need to add Model ID + Manufacture name.

pakwan, 2024-08-18:
Thanks for your answer :) I use the home assistant plugin, so as far as I can see I cannot build the code and test it myself, right? So should I just add it to the mentioned file and raise a PR? Or what is the usual way to proceed in such cases?

Smanar, 2024-08-18:
No need to build the code, just create a text file, but an easy thing like that is a pain on this kind of OS.
You can create a new DDF in the permanent "user" device folder, it will by-pass the native one.
- Short way the folder is `/data/.local/share/dresden-elektronik/deCONZ/devices/`
- long way https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

pakwan, 2024-08-18:
Thanks for the hint! In parallel I also did my research and succeeded :partying_face: 

For future reference, this is what I did:
(from https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839/18)
    Login via SSH to HA (install the "Advanced SSH & Web Terminal" addon, not the other SSH addon)
    Login deCONZ docker container: docker exec -it addon_core_deconz bash
    Install a text editor e. g. nano for editing files on the CLI: apt update; apt install nano (note: this will be undone once the container is restarted or updated)
    Edit your DDF file with e. g. nano /usr/share/deCONZ/devices/blitzwolf/bw_shp13_smart_plug.json
    Don’t forget to “hot reload” the DDF changes using deCONZ (VNC GUI)

In my case I updated the file using this:
nano /usr/share/deCONZ/devices/tuya/_TZ3000_TS0201_temp_hum_sensor.json
to contain this device manufacturer and modelid.

copied to /data/.local/share/dresden-elektronik/deCONZ/devices/tuya/_TZ3000_TS0201_temp_hum_sensor.json

Went into the DDF mode of the temperature sensor in the VNC UI. Clicked on hot reload. Nothing happened. Restarted deconz plugin. Tried again with hot reload. No visible change. Deleted node in VNC UI. Tried pairing again. Now it is showing and working perfectly fine :)

Anyways, I will create a PR now, so that it will work for others in the future as well :)

Smanar, 2024-08-18:
Perfect ^^, thx a lot.

closed 2024-09-16 by github-actions[bot]