dresden-elektronik/deconz-rest-plugin #7849: Tuya Motion Sensor and Scene Switch -  TS0202 | _TZ3210_cwamkvua

phuongpham, 2024-07-21:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Tuya Motion Sensor and Scene Switch

### Manufacturer

_TZ3210_cwamkvua

### Model identifier

TS0202

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

More info about the device:

https://zigbee.blakadder.com/Tuya_LKMSZ001.html
https://www.zigbee2mqtt.io/devices/TS0202_2.html

Example how the device is implemented in HA
https://github.com/zigpy/zha-device-handlers/issues/2163

phuongpham, 2024-08-12:
no one?

phuongpham, 2025-03-20:
Can we please repoen the issue? I can't see that this device is supported yet.

Smanar, 2025-03-20:
He is not.
What is this device ? A motion sensor+ switch 1 gang on battery ?
Can try this DDF

```
{
    "schema": "devcap1.schema.json",
    "manufacturername": "_TZ3210_cwamkvua",
    "modelid": "TS0202",
    "vendor": "Tuya",
    "product": "Motion Sensor and Scene Switch",
    "status": "Gold",
    "subdevices": [
        {
            "type": "$TYPE_PRESENCE_SENSOR",
            "restapi": "/sensors",
            "uuid": [
                "$address.ext",
                "0x01",
                "0x0500"
            ],
            "fingerprint": {
                "profile": "0x0104",
                "device": "0x0402",
                "endpoint": "0x01",
                "in": [
                    "0x0001",
                    "0x0500"
                ]
            },
            "items": [
                {
                    "name": "attr/id"
                },
                {
                    "name": "attr/manufacturername"
                },
                {
                    "name": "attr/modelid"
                },
                {
                    "name": "attr/name"
                },
                {
                    "name": "config/tuya_unlock"
                },
                {
                    "name": "attr/swversion",
                    "parse": {
                        "fn": "zcl:attr",
                        "ep": 1,
                        "cl": "0x0000",
                        "at": "0x0001",
                        "script": "tuya_swversion.js"
                    },
                    "read": {
                        "fn": "zcl:attr",
                        "ep": 1,
                        "cl": "0x0000",
                        "at": "0x0001"
                    }
                },
                {
                    "name": "attr/type"
                },
                {
                    "name": "attr/uniqueid"
                },
                {
                    "name": "attr/lastseen"
                },
                {
                    "name": "config/battery",
                    "parse": {
                        "fn": "zcl:attr",
                        "ep": 1,
                        "cl": "0x0001",
                        "at": "0x0021",
                        "eval": "Item.val = Attr.val"
                    },
                    "awake": true
                },
                {
                    "name": "config/duration",
                    "default": 120
                },
                {
                    "name": "config/enrolled"
                },
                {
                    "name": "config/on"
                },
                {
                    "name": "config/pending"
                },
                {
                    "name": "config/reachable"
                },
                {
                    "name": "state/presence",
                    "awake": true
                },
                {
                    "name": "state/lowbattery"
                },
                {
                    "name": "state/tampered"
                },
                {
                    "name": "state/lastupdated"
                }
            ]
        },
        {
            "type": "$TYPE_SWITCH",
            "restapi": "/sensors",
            "uuid": [
                "$address.ext",
                "0x01",
                "0xEF00"
            ],
            "items": [
                {
                    "name": "attr/id"
                },
                {
                    "name": "attr/lastannounced"
                },
                {
                    "name": "attr/lastseen"
                },
                {
                    "name": "attr/manufacturername"
                },
                {
                    "name": "attr/modelid"
                },
                {
                    "name": "attr/name"
                },
                {
                    "name": "attr/swversion",
                    "parse": {
                        "fn": "zcl:attr",
                        "ep": 1,
                        "cl": "0x0000",
                        "at": "0x0001",
                        "script": "tuya_swversion.js"
                    },
                    "read": {
                        "fn": "zcl:attr",
                        "ep": 1,
                        "cl": "0x0000",
                        "at": "0x0001"
                    }
                },
                {
                    "name": "attr/type"
                },
                {
                    "name": "attr/uniqueid"
                },
                {
                    "name": "config/battery",
                    "parse": {
                        "fn": "zcl:attr",
                        "ep": 1,
                        "cl": "0x0001",
                        "at": "0x0021",
                        "eval": "Item.val = Attr.val"
                    },
                    "awake": true
                },
                {
                    "name": "config/on"
                },
                {
                    "name": "config/reachable"
                },
                {
                    "name": "state/buttonevent",
                    "read": {
                        "fn": "tuya"
                    },
                    "parse": {
                        "fn": "tuya",
                        "dpid": 101,
                        "eval": "Item.val = 1000 + Attr.val"
                    }
                },
                {
                    "name": "state/lastupdated"
                }
            ]
        }
    ]
}
```

Will need to ajust the  "state/buttonevent", ATM it will return 1001 1002 and 1003

phuongpham, 2025-03-20:
@Smanar great!
Just trying the DDF file now.
The push button works fine. It return 1000 when pushed.

But the PIR sensor and battery value doesn't return.
Battery is return value: 0.
The lastseen and lastupdated wil also not update.

{
    "config": {
        "battery": 0,
        "duration": 120,
        "on": true,
        "pending": [],
        "reachable": true
    },
    "ep": 1,
    "etag": "XXXXX",
    "lastannounced": null,
    "lastseen": "2025-03-20T18:41Z",
    "manufacturername": "_TZ3210_cwamkvua",
    "modelid": "TS0202",
    "name": "PIR_touch",
    "state": {
        "lastupdated": "2025-03-20T18:41:57.717",
        "lowbattery": false,
        "presence": false,
        "tampered": false
    },
    "swversion": "1.2.9",
    "type": "ZHAPresence",
    "uniqueid": "XXXX"
}

phuongpham, 2025-03-20:
@Smanar  I just notice something strange.. When I double click on the button the PIR motion switch from OFF to ON and the lastseen and lastupdated get new timestamp. If i double click again the PIR go from ON to OFF and the dates get updated.

phuongpham, 2025-03-20:
@Smanar  ok I managed to make the PIR/Presence sensor working. Seems like after restarting deconz it takes (3-5 min) before the PIR sensor detect.

I have also done some changes in the DDF file. But still struggeling that the battery still show 0.
How can I identify/find the correct dpid?

`
{
    "schema": "devcap1.schema.json",
    "manufacturername": "_TZ3210_cwamkvua",
    "modelid": "TS0202",
    "vendor": "Tuya",
    "product": "Motion Sensor and Scene Switch",
    "status": "Gold",
    "subdevices": [
        {
            "type": "$TYPE_PRESENCE_SENSOR",
            "restapi": "/sensors",
            "uuid": [
                "$address.ext",
                "0x01",
                "0x0500"
            ],
            "fingerprint": {
                "profile": "0x0104",
                "device": "0x0402",
                "endpoint": "0x01",
                "in": [
                    "0x0001",
                    "0x0500"
                ]
            },
            "items": [
                {
                    "name": "attr/id"
                },
                {
                    "name": "attr/manufacturername"
                },
                {
                    "name": "attr/modelid"
                },
                {
                    "name": "attr/name"
                },
                {
                    "name": "config/tuya_unlock"
                },
                {
                    "name": "attr/swversion",
                    "parse": {
                        "fn": "zcl:attr",
                        "ep": 1,
                        "cl": "0x0000",
                        "at": "0x0001",
                        "script": "tuya_swversion.js"
                    },
                    "read": {
                        "fn": "zcl:attr",
                        "ep": 1,
                        "cl": "0x0000",
                        "at": "0x0001"
                    }
                },
                {
                    "name": "attr/type"
                },
                {
                    "name": "attr/uniqueid"
                },
                {
[... 136 lines cut ...]
    ],
   "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 3600,
          "max": 14400,
          "change": "0x00000001"
        }
      ]
    }
  ]
}

`

phuongpham, 2025-03-21:
[image]
Please see the info about the battery.
Where do I read the cluster information?

Smanar, 2025-03-22:
Ok so the device display "0" value (and as other values are grayed I think you have succed in procedure). So battery is not working or it use the Tuya cluster for sure.

And with searching on google I think lot of version of your device with the same Manufacture name, so for information the final DDF will never work on all device.

To get the battery, no other way than looking in logs, with the GUI, help/debug view with flag "info" and "info_l2", you will see tuya request.

If you look in cluster command in the Tuya cluster you will foud a command to ask for all dpid, can have usefull information too (the result is visible on logs only, nothing will move in the GUI)

closed 2025-04-21 by github-actions[bot]