dresden-elektronik/deconz-rest-plugin #7852: Third Reality inc - Smart Plug E2(type F, power monitoring), Model 3RSPE01044BZ

TheBisk, 2024-07-22:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Third Reality 3RSPE01044BZ

### Manufacturer

Third Reality

### Model identifier

3RSPE01044BZ

### Device type to add

Switch

### Node info

[image]

### Endpoints and clusters

nn

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

SwoopX, 2024-07-22:
Hi,

please provide the remaining and requested screenshots. Many thanks!

TheBisk, 2024-07-23:
Do need something else? 
Thanks

Smanar, 2024-07-24:
Hello, can try this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Third Reality, Inc",
  "modelid": "3RSPE01044BZ",
  "vendor": "Third Reality",
  "product": "Smart plug",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert"
        },
        {
          "name": "state/on",
          "refresh.interval": 360
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current",
          "refresh.interval": 360
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "read": {
            "at": "0x050b",
            "cl": "0x0b04",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x050b",
            "cl": "0x0b04",
            "ep": 1,
            "eval": "Item.val = Attr.val / 10"
          },
          "refresh.interval": 360
        },
        {
          "name": "state/voltage",
          "read": {
            "at": "0x0505",
            "cl": "0x0b04",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0505",
            "cl": "0x0b04",
            "ep": 1,
            "eval": "Item.val = Attr.val / 10"
          },
          "refresh.interval": 360
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0702",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x25",
          "min": 10,
          "max": 300,
          "change": "0x0000000A"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 10,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 10,
          "max": 300,
          "change": "0x00000064"
        },
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 10,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
```

Based on this one https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/third_reality/3RSP02028BZ_smart_plug.json

TheBisk, 2024-07-24:
Tested quiclky and it seems to work. 
Now i have the switch and the sensors for power and consuption. I will test it better next days.

Thank you Smanar

TheBisk, 2024-07-25:
i made some test and it seems that **the consumption value is too high**.
is there a way to reset the consumption and start from 0? the value looks read only...

Power
[image]

Consumption
[image]

Simple meetering
[image]
[image]

Smanar, 2024-07-25:
Do you have the same capture for electrical measurement ?
Or use a device with a known power to see the convertion needed ? (a 10W bulb for exemple).
what do you have as current ?

For consumption can replace
```
        {
          "name": "state/consumption"
        },
```
by
```
        {
          "name": "state/consumption",
          "read": {
            "at": "0x0000",
            "cl": "0x0702",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0702",
            "ep": 1,
            "eval": "Item.val = Attr.val / 10"
          },
          "refresh.interval": 365
        },
```
Here it's just a "/10" to show you how it work because a divisor = 360 000 (from your capture) I m not sure it's something reliable.
Better to get 2 value with 1 h delay and a constant power.

Smanar, 2024-07-25:
To reset consumption, with the GUI try on cluster 0x0000 the reset factory request.

TheBisk, 2024-07-25:
Electrical Measurement 
[image]
[image]
[image]
[image]

Smanar, 2024-07-25:
So from the device, same convertion than for voltage 1/10 for power.

TheBisk, 2024-07-25:
> To reset consumption, with the GUI try on cluster 0x0000 the reset factory request.

i dont know how to do that...

TheBisk, 2024-07-25:
> Hello, can try this DDF
>
> ```
> {
>   "schema": "devcap1.schema.json",
>   "manufacturername": "Third Reality, Inc",
>   "modelid": "3RSPE01044BZ",
>   "vendor": "Third Reality",
>   "product": "Smart plug", [...]

This is the last working DDF, thanks Smanar
 ```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Third Reality, Inc",
  "modelid": "3RSPE01044BZ",
  "vendor": "Third Reality",
  "product": "Smart plug",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_PLUGIN_UNIT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/alert"
        },
        {
          "name": "state/on",
          "refresh.interval": 360
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_POWER_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0b04"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/current",
          "refresh.interval": 360
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/power",
          "read": {
            "at": "0x050b",
            "cl": "0x0b04",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x050b",
            "cl": "0x0b04",
            "ep": 1,
            "eval": "Item.val = Attr.val / 10"
          },
          "refresh.interval": 360
        },
        {
          "name": "state/voltage",
          "read": {
            "at": "0x0505",
            "cl": "0x0b04",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0505",
            "cl": "0x0b04",
            "ep": 1,
            "eval": "Item.val = Attr.val / 10"
          },
          "refresh.interval": 360
        }
      ]
    },
    {
      "type": "$TYPE_CONSUMPTION_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0702"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/consumption",
          "read": {
            "at": "0x0000",
            "cl": "0x0702",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0000",
            "cl": "0x0702",
            "ep": 1,
            "eval": "Item.val = Attr.val / 10"
          },
          "refresh.interval": 365
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0702",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x25",
          "min": 10,
          "max": 300,
          "change": "0x0000000A"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0B04",
      "report": [
        {
          "at": "0x0505",
          "dt": "0x21",
          "min": 10,
          "max": 300,
          "change": "0x00000001"
        },
        {
          "at": "0x0508",
          "dt": "0x21",
          "min": 10,
          "max": 300,
          "change": "0x00000064"
        },
        {
          "at": "0x050B",
          "dt": "0x29",
          "min": 10,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    }
  ]
}
 ```

Smanar, 2024-07-26:
> i dont know how to do that...

Do you have the GUI, you have used it for capture. Just select the cluster 0x0000 "basic" (the blue number) and on the cluster panel you will the button.

The DDF is finished ? do you want to make a PR or want I make it ? (to have it officialy)

TheBisk, 2024-08-06:
I think that is better if you make the PR (I don't know how to do that).
Thanks

Smanar, 2024-08-07:
Done https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7875

closed 2024-08-07 by manup

TheBisk, 2024-09-23:
Hi Smanar,
I did some tests and I can confirm that the **consumption value** **is too high** compared to the watts supplied. 
My dryer worked for 2 hours and consumed on average 500watts per hour so I would expect +1kwh of consumption for 2 hours and not +250kwh. what do you think? 

Thank you.

[image]

[image]

Smanar, 2024-09-23:
Lol, I think too ^^, from your capture the divisor is 3600000
So can try with 
`"eval": "Item.val = Attr.val / 360000"`

It will make 6kw/h if I m right.

BTW you haven't issue too with curent and voltage ?

TheBisk, 2024-10-04:
> Lol, I think too ^^, from your capture the divisor is 3600000 So can try with `"eval": "Item.val = Attr.val / 3600000"`
> 
> It will make 0.6kw/h if I m right.
> 
> BTW you haven't issue too with curent and voltage ?

Smanar,
I did some more testing and it seems the correct divisor is 36000. As regards current, the value does not seem to update, when the device is turned off it shows current 66 and 1W while it should be 0. 
The voltage however seems ok

[image]

Smanar, 2024-10-05:
You mean when you turn off the device, it keep the last value for current and power ? Or it use lower one ?
For information it can be the power used by the plug itself ? If I m right a plug use at least 0.5W in stand by mode.