dresden-elektronik/deconz-rest-plugin #7857: Sonoff SNZB-04P

Wuehli, 2024-07-27:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

SNZB-04P

### Manufacturer

Sonoff 

### Model identifier

SNZB-04P

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]
[image]

### Further relevant clusters

### Power Configuration
[image]

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

[image]

[image]

[image]
[image]

1Rolf, 2024-08-20:
Any updates here? Did i miss something?

dab0u, 2024-09-08:
I can't include this device with my conbee 3 neither.  Can I do something to help ?

Mimiix, 2024-09-08:
@Smanar perhaps you can help after your holiday :)?

Smanar, 2024-09-16:
Hello can try this DDF

```
{
  "schema": "devcap1.schema.json",
  "uuid": "cd25fbd8-9b9b-4051-9c01-1c08a9d65203",
  "manufacturername": ["eWeLink", "zbeacon", "eWeLink"],
  "modelid": ["DS01", "DS01", "SNZB-04P"],
  "vendor": "eWeLink",
  "product": "Open/close sensor (DS01)",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_OPEN_CLOSE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl:attr"
          },
          "read": {
            "fn": "none"
          }
        },
        {
          "name": "config/checkin"
        },
        {
          "name": "config/enrolled"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery",
          "awake": true
        },
        {
          "name": "state/open",
          "awake": true
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 3600,
          "max": 43200,
          "change": "0x00000002"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0500"
    }
  ]
}
```

It's just an optimisation of this one https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/sonoff/snzb-04_open_close_sensor.json

JasonMask, 2024-10-27:
Hi,

as you can see from my screenshots in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7995 the "Model Identifier" of my device is "SNZB-04" (with out the P given for this device)

once I added that to the manufacturername/modelid lines the device could be found in the phoscon app:
`"manufacturername": ["eWeLink", "zbeacon", "eWeLink", "eWeLink"],`
`"modelid": ["DS01", "DS01", "SNZB-04P", "SNZB-04"],`

I hope this helps

Best regards

Smanar, 2024-10-28:
Sure it help, and thx for it
Added here > https://github.com/dresden-elektronik/deconz-rest-plugin/pull/8002

closed 2024-11-04 by manup