dresden-elektronik/deconz-rest-plugin #7859: Web access does not work in v2.27.6

molnarti, 2024-07-27:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

The container stopped working since updating to 2.27.6. Logs claim some VNC error - see below attached. After disabling VNC the container runs, but the web interface does not load - see screenshot.

### Steps to reproduce the behavior

Update to 2.27.6

### Expected behavior

VNC and web access works as before

### Screenshots

[image]

### Environment

- Host system: Debian Linux
- Running method: deconz-docker container
- Firmware version: (26xxyy00)
- deCONZ version (not Home assistant Addon version!): 2.27.6
- Device: ConBee II
- Do you use an USB extension cable: no
- Yes, UPS

### deCONZ Logs

Sat Jul 27 15:31:09 2024
 vncext:      VNC extension running!
 vncext:      Listening for VNC connections on all interface(s), port 5900
 vncext:      created VNC server for screen 0
Inconsistency detected by ld.so: ../sysdeps/x86_64/dl-machine.h: 487: elf_machine_rela_relative: Assertion `ELFW(R_TYPE) (reloc->r_info) == R_X86_64_RELATIVE' failed!
==============================================================================================
tigervncserver: /usr/bin/Xtigervnc exited with status 127, please look into '/home/deconz/.vnc/dockerhost.radirpok.cloud:0.log' to determine the reason! -2
Cleaning stale X11 lock '/tmp/.X0-lock'!
Cleaning stale X11 lock '/tmp/.X11-unix/X0'!
WebSocket server settings:
  - Listen on :6080
  - Web server. Web root: /usr/share/novnc
  - SSL/TLS support
  - Backgrounding (daemon)
[deconzcommunity/deconz] NOVNC port: 6080
qt.qpa.xcb: could not connect to display :0
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, xcb.

wvuyk, 2024-07-27:
I see the same issue here on 2.27.5 also. Both on conbee2 and rasbee2

Mimiix, 2024-07-27:
Is this vnc in docker only or phoscon?

joselito11, 2024-07-28:
This is only phoscon, not one  `external` browser is working after update. Vnc is working.
I have managed to access phoscon via vnc to rasbian and there via chromium, because I have to change light in group after each reboot or updgrade.

Mimiix, 2024-07-28:
> This is only phoscon, not one `external` browser is working after update. Vnc is working. I have managed to access phoscon via vnc to rasbian and there via chromium, because I have to change light in group after each reboot or updgrade.

If its phoscon > please report on the forums as bugs are collected there.

rikroe, 2024-07-28:
I guess it is partly a bug report here as well. Comparing the last `stable` release with 2.27.6, all the new beta features are included. Or is 2.27.6 suppose to be the new stable release?

wvuyk, 2024-07-28:
Nice find... maybe it is time to update the default page?

MSL-DA, 2024-07-28:
It's been like this since v2.26.3 https://hub.docker.com/layers/deconzcommunity/deconz/2.26.3/images/sha256-715b7a08dc9cae696376b561c75b4c0458776f1d8c6ced66d77d682f4431d540 so nothing new

rikroe, 2024-07-28:
But then I still don't unterstand why calling the base URL (e.g. `http://ip:8080/`) forwards to `pwa/login.html` instead of `pwa/login2.html` even when no cache whatsoever is used (browser incognito mode).

Mimiix, 2024-07-28:
As this is a phoscon bug and not a deconz bug, I'm closing and locking this issue.

Please use the opened phoscon topic: 

https://forum.phoscon.de/t/web-access-not-working/5023/3

closed 2024-07-28 by Mimiix