dresden-elektronik/deconz-rest-plugin #7863: Third Reality, Inc. Smart Button 3RSB22BZ

louix, 2024-07-29:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Smart Button

### Manufacturer

Third Reality, Inc.

### Model identifier

3RSB22BZ

### Device type to add

Switch

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration
[image]

### Multistate Input (Basic)
[image]

### IAS Zone
[image]

### Level Control
[image]

Mimiix, 2024-08-20:
@Smanar Can you help out :)?

Smanar, 2024-08-20:
Yep, but lot of work to do, can start with this DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Third Reality, Inc",
  "modelid": "3RSB22BZ",
  "product": "Smart Button 3RSB22BZ",
  "status": "Gold",
  "sleeper": true,
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0012"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl:attr"
          }
        },
        {
          "name": "state/buttonevent"
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 3600,
          "max": 84600,
          "change": "0x00000002"
        }
      ]
    },
  ]
}
```

Then use log in deconz/help/debug view with flag "info" and "info_l2" and press the buttons.
I need the button pressed, the action and the result on logs, at least a part, I can guess the rest.

For the moment as I don't know if the device need it, I don't make enrollement, if the device need them, can add

```
        {
          "name": "config/pending"
        },
        {
          "name": "config/enrolled",
          "public": false
        },
```

And No bind for the 2 clusters 0x0500 and 0x0008 as from that I m reading this device will use the cluster 0x0012 to make report when buttons are pressed.
And this is a good new as it mean the device can probably work without the use of the buttonmap.json file.

louix, 2024-08-21:
Hey @Smanar, thanks a lot. It shows up in Phoscon with that DDF.

There's only a single button, here's what I could see in the logs.

Short Press and release:
```
02:11:53:472 [INFO] - No button map for: 3RSB22BZ, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), command: ON (0x01), payload: None, zclSeq: 93
02:11:53:477 [INFO] - No button map for: 3RSB22BZ, unicast to: 0x0000, endpoint: 0x01, cluster: MULTISTATE_INPUT (0x0012), command: ATTRIBUTE_REPORT (0x0A), payload: 5500210100, zclSeq: 94
02:11:53:478 ZCL attribute report 0x282C02BFFFEE0881 for cluster: 0x0012, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
02:11:53:479 	payload: 5500210100
```

Long press and release:
```
02:13:00:816 [INFO] - No button map for: 3RSB22BZ, unicast to: 0x0000, endpoint: 0x01, cluster: MULTISTATE_INPUT (0x0012), command: ATTRIBUTE_REPORT (0x0A), payload: 5500210000, zclSeq: 104
02:13:00:816 ZCL attribute report 0x282C02BFFFEE0881 for cluster: 0x0012, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
02:13:00:817 	payload: 5500210000
02:13:00:936 [INFO] - No button map for: 3RSB22BZ, unicast to: 0x0000, endpoint: 0x01, cluster: LEVEL_CONTROL (0x0008), command: 0x0A, payload: 00002004, zclSeq: 105
02:13:00:936 ZCL attribute report 0x282C02BFFFEE0881 for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
02:13:00:936 	payload: 00002004
02:13:01:278 [INFO] - No button map for: 3RSB22BZ, unicast to: 0x0000, endpoint: 0x01, cluster: MULTISTATE_INPUT (0x0012), command: ATTRIBUTE_REPORT (0x0A), payload: 550021FF00, zclSeq: 106
02:13:01:279 ZCL attribute report 0x282C02BFFFEE0881 for cluster: 0x0012, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
02:13:01:279 	payload: 550021ff00
```

Smanar, 2024-08-21:
So good news, enrollment not needed, bind not needed, and it seem can use only the cluster 0x0012

`02:13:00:816 [INFO] - No button map for: 3RSB22BZ, unicast to: 0x0000, endpoint: 0x01, cluster: MULTISTATE_INPUT (0x0012), command: ATTRIBUTE_REPORT (0x0A), payload: 5500210000, zclSeq: 104`

I will try to use the fourth byte on payload:
55 00 21 00 00 : 00 > hold
55 00 21 FF 00 : FF > release
55 00 21 01 00 : 01 > short
And ect ....

This DDF is just a test

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Third Reality, Inc",
  "modelid": "3RSB22BZ",
  "product": "Smart Button 3RSB22BZ",
  "status": "Gold",
  "sleeper": true,
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0012"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl:attr"
          }
        },
        {
          "name": "state/buttonevent",
          "awake": true,
          "parse": {
            "cl": "0x0012",
            "cmd": "0x0A",
            "eval": "Item.val = 1000 + ZclFrame.at(3)"
          }
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 3600,
          "max": 84600,
          "change": "0x00000002"
        }
      ]
    },
  ]
}
```

If I m right buttonevent will return 1001 for short, 1000 for hold, and ect ....
Need to be adjusted if it work

louix, 2024-08-21:
Yep, amazing. These are the button events:

1001: short press
1002: double press
1000: long press (start)
1255: long press (release)

Smanar, 2024-08-21:
Deconz rules are https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices#wireless-switches
So I think this DDF will work, or you can easily correct it.

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "Third Reality, Inc",
  "modelid": "3RSB22BZ",
  "product": "Smart Button 3RSB22BZ",
  "status": "Gold",
  "sleeper": true,
  "subdevices": [
    {
      "type": "$TYPE_SWITCH",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0012"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl:attr"
          }
        },
        {
          "name": "state/buttonevent",
          "awake": true,
          "parse": {
            "cl": "0x0012",
            "cmd": "0x0A",
            "eval": "const t={'00':1000,'01':1002,'02':1004,'03':1005,'04':1006,'255':1003};if(ZclFrame.at(3) in t){Item.val=t[ZclFrame.at(3)]}"
          }
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 3600,
          "max": 84600,
          "change": "0x00000002"
        }
      ]
    },
  ]
}
```

The magic is here

`"eval": "const t={'00':1000,'01':1002,'02':1004,'03':1005,'04':1006,'255':1003};if(ZclFrame.at(3) in t){Item.val=t[ZclFrame.at(3)]}"`

louix, 2024-08-21:
Fantastic, 255 is correct. 

01, 02 etc. did not work (probably 0 padding, converting to number fixes it). Here is the updated magic:
```js
Item.val={1: 1002, 2: 1004, 0: 1001, 255: 1003}[Number(ZclFrame.at(3))]
```
or closer to the original:
```js
n=Number(ZclFrame.at(3));t={'1': 1002, '2': 1004, '0': 1001, '255': 1003};if(n in t){Item.val=t[n]}
```

Thanks again for taking the time, your replies have been very informative.

Smanar, 2024-08-22:
From Z2M this device have too triple and quadruple clic, It's for that I have added too
`'3':1005,'4':1006,`

If all is working for you, do you want to make a PR or want I make it, to add it officialy ?

Not a problem to help users when they are able to do all the work themself ^^, and BTW thx a lot for your gift, really generous.

louix, 2024-08-22:
No worries at all, it's been a good learning experience.

Testing it (without the eval mapping) it doesn't seem like it has triple & quadruple events (I just get multiple double click events), the manual only seem to mention double too.

Opened the PR #7897.

closed 2024-09-09 by manup