dresden-elektronik/deconz-rest-plugin #7866: Hue Motion Sensor (SML001) no longer pairing in latest STABLE v2.27.6

mtressl, 2024-08-03:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Lost two of my Hue Motion Sensors due to a prolonged power outage. These two (SML001) are no longer reachable, and after deleting one of them also no longer pairable. Another Hue Outdoor Sensor (SML002) is working like a charm.

### Steps to reproduce the behavior

1. install latest STABLE v2.27.6
2. delete Hue Motion Sensor (SML001)
3. try to pair Hue Motion Sensor again (even after a device reset)

### Expected behavior

Latest STABLE v2.27.6 is able to pair Hue Motion Sensors (SML001)

### Environment

- Host system: Raspberry Pi
- Running method: deconz-docker container
- Firmware version: (26780700)
- deCONZ version: 2.27.6
- Device: RaspBee II
- Do you use an USB extension cable: no
- Is there any other USB or serial devices connected to the host system? no

mtressl, 2024-08-08:
A downgrade to v2.25.3 solves the pairing issues w/ SML001 and SML002.

SwoopX, 2024-08-11:
Will have a look once we've hunted down a more severe bug, hindering me to follow up here 🙂

alfaskop, 2024-08-12:
I have similar issue with RDM001, wall switch button. It's possible to reconnect but buttons aren't working.

Also all my different Aqara devices stopped working, and it wasn't possible to pair them until I downgraded to 2.26.3. I have water leak sensors, temperature/humidity sensors and motion sensors.

SwoopX, 2024-08-14:
@alfaskop Please do not hijack this issue. It is unfortunate that you seem to have some issues, but this is not related to a Hue motion sensor.

@mtressl Got my environment back up and running. Please try to delete the motion sensor(s) in deconz GUI via a right click. I also wasn't able to re-pair it unless I took that action.

mtressl, 2024-09-08:
hey @SwoopX 
little update from my side: it seems the only way to solve the issue with 'lost' sensors is really to delete them and re-pair afterwards. This is valid not only for the hue sensors, but also for Xiaomi (mi) ones... (/cc @alfaskop).

my problem(s) are herewith solved, but I guess this is not the state deconz should work like.

closed 2024-09-08 by mtressl