dresden-elektronik/deconz-rest-plugin #7867: Add support for MOES 1 Gang Switch

svenvoss, 2024-08-03:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

MOES ZigBee Funkschalter, Smartschalter ohne Installation, 1 Gang

### Manufacturer

_TZ3000_axpdxqgu

### Model identifier

TS0041

### Device type to add

Switch

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Smanar, 2024-08-03:
Hello, can try adding your device to this list https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_TS0041_1gang_remote.json
And in the button_maps.json file to test.

svenvoss, 2024-08-04:
Hi,
thank you for the quick reply. I am sorry, but it is not fully clear to me what I should do. Could you please give me some more details what I should do.

The device is currently recognized as light but it is light/scene switch.

Smanar, 2024-08-04:
>The device is currently recognized as light but it is light/scene switch.

Ha yes, this device is already supported, sorry.
But from the DDF this device need to be reconised as a ZHAswitch.

If I m right you need to be able to see websocket notification every time you press it, visible in Phoscon/help/API information/event

svenvoss, 2024-08-05:
Deconz 6.23.0 

Installation via Homeassistant Addon on Raspberry Pi. The requested log:

[image]

For any reason copying from the log doesn't working.

Smanar, 2024-08-06:
>For any reason copying from the log doesn't working.

This is normal on HA, but I don't understand On my side I can see it in my file.

`[1, "0x01", "ONOFF", "0xfd", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "B1 short"],`

>Deconz 6.23.0

This is the plugin version, not the deconz version (you can see it in phoscon)
This device was added here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7718 so need a deconz version > v2.27.0

svenvoss, 2024-08-12:
HI, sorry for the delay. Is this what you meant?: 
[image]

I have tried to update the conbee FW as described  here: https://community.home-assistant.io/t/how-to-update-conbee-ii-on-hassio-with-deconz-phoscon-integration/402826/12 but he didn't find any new software. Do I need to update something else and if yes how.

Smanar, 2024-08-12:
Ok so the problem is here, you are on older version 2.25.3 (I mean the deconz version, not the device firmware), and this device have it's support on version v2.27.0

svenvoss, 2024-08-12:
Is there any manual how to update the deconz version?

Smanar, 2024-08-12:
Not on HA if you are using the "integrated deconz version".
They are using only Stable version, so they can use the version 2.27.6 if I m right.

svenvoss, 2024-08-12:
Thank you very much for your time and support.

closed 2024-08-12 by svenvoss

Zehir, 2024-08-12:
> Not on HA if you are using the "integrated deconz version".
> They are using only Stable version, so they can use the version 2.27.6 if I m right.

Latest version of the HA plugin in hassio (6.23.0) use deconz version 2.25.3

Smanar, 2024-08-13:
Yep but as the version 2.27.6 is stable, they will use it nope ,

svenvoss, 2024-08-13:
because of your hints I have checked the plugin page/repo. They have already a closed pull request for the actual stable version. Hopefully it is included in the next release. 
Thank you all for your support!