dresden-elektronik/deconz-rest-plugin #7872: Woox water leak sensor R7050

raderack3948, 2024-08-06:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Woox R7050

### Manufacturer

Woox

### Model identifier

R7050

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

[image]

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

[image]

Smanar, 2024-08-06:
Hello, probably just one more clone, can you try adding this model in this DDF https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_TS0207_water_leak_sensor.json

Need to add the modelID AND the manufacture name.

raderack3948, 2024-08-06:
Added it, I can pair it and see it in phoscon - but w/o any values?
[image]

Smanar, 2024-08-07:
Perhaps Phoscon issue, can you check direclty in the API ? Using Phoscon/help/API Information/sensor ?

raderack3948, 2024-08-07:
Hello, sure - attached:

```
{
    "config": {
        "battery": 0,
        "on": true,
        "pending": [],
        "reachable": true
    },
    "ep": 1,
    "etag": "XXX",
    "lastannounced": null,
    "lastseen": "2024-08-06T16:25Z",
    "manufacturername": "_TZ3000_arw23zcs",
    "modelid": "TS0207",
    "name": "Water",
    "state": {
        "lastupdated": "none",
        "lowbattery": null,
        "tampered": null,
        "water": null
    },
    "type": "ZHAWater",
    "uniqueid": "XXX"
}
```

Smanar, 2024-08-07:
Strange no "swversion" and nothing updated (even battery)
And same after 24h ...

It probably a new device, I can't found information about it on google, but Woox have ever locked their tuya device. Can you try to re-include the device, but without deleting it (just to trigger again an enrollement for the IAS stuff) ?

raderack3948, 2024-08-07:
Sorry to ask - how to I reinclude the device? :-)

Smanar, 2024-08-08:
Just set the network in permit join, using phoscon, add new sensor, and make again the reset procedure on the device, all settings will be keep.

raderack3948, 2024-08-09:
It worked! Sensor works after repairing. On the first one I've got the following message, but the sensor is in the system and working:
[image]

Smanar, 2024-08-09:
Ok, so was because of a bad inclusion/enrollement.
Do you want to make a PR to add this device or want I make it ?

raderack3948, 2024-08-09:
It would be great if you could do that (I don't know how) :-). Thanks a lot for your help!!

Smanar, 2024-08-10:
Depend of OS, what is your ?
You just need to create a text file with this contain in a "devices" folder.

Smanar, 2024-08-10:
Done https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7878

closed 2024-08-14 by manup