dresden-elektronik/deconz-rest-plugin #7874: Mumibiz TYZGTH4CH with temperature

enboer, 2024-08-07:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Mumibiz TYZGTH4CH

### Manufacturer

Mumibiz

### Model identifier

TYZGTH4CH

### Device type to add

Switch

### Node info

[image]

### Endpoints and clusters

DEvice has 4 switches and a temp sensor. 4 switches are working fine. But I cannot get the temperature to work. Added the ZHA temperature as a new item (DDF edit). But I do not get that to work.

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Mimiix, 2024-08-07:
Please add all screenshots.

enboer, 2024-08-09:
additional info
[image]

[image]

[image]

[image]

[image]

enboer, 2024-08-18:
Maybe this helps? https://www.zigbee2mqtt.io/devices/TYZGTH4CH-D1RF.html

Smanar, 2024-08-27:
The temperature sensor need to be like this one

```
   {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 4, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = 10 * Attr.val + R.item('config/offset').val;" },
          "read": {"fn": "none"},
          "default": 0
        }
      ]
    }
```

Not sure for the dpid, it's from https://github.com/Koenkk/zigbee2mqtt/issues/23041

Smanar, 2024-08-27:
The temperature sensor need to be like 

```
   {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "parse": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/battery",
          "parse": {"fn": "tuya", "dpid": 4, "eval": "Item.val = Attr.val;" },
          "read": {"fn": "none"},
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 1, "eval": "Item.val = 10 * Attr.val + R.item('config/offset').val;" },
          "read": {"fn": "none"},
          "default": 0
        }
      ]
    }
```

Just add it to your previous DDF, or if you share it I can complete it.

closed 2024-09-25 by github-actions[bot]

enboer, 2024-09-27:
Sorry for replying this late... Thanks for your help. I added it to the json file I get the devices, but not the temp sensor. Also in Deconz I do not see the Temp section or the temp value.

[TZ3218.json](https://github.com/user-attachments/files/17167760/TZ3218.json)

In Deconz I see the updated file is used (changed the device description) but no temp updates

enboer, 2024-09-27:
[image]

Smanar, 2024-09-28:
It will be not visible on deconz but in the API, using for exemple Phoscon/help/API Information/sensor

enboer, 2024-10-02:
Added device again. In Phoscon I see the temp sensor. Only temp value stays at 0 degree

Smanar, 2024-10-02:
Have you aded the tuya unlock sequence ?
```
	{
	 "name": "config/tuya_unlock"
	},
```

You need to re-include the device after or wait 24h.

enboer, 2024-10-02:
I have added this. unpaired, stopped deconz. Started deconz and paired again. But in the .json file I added the text you have suggested. But in the DDF in Deconz Gui I see some more info. See picture.
[image]

Smanar, 2024-10-03:
Yeah, the editor make too much "magic".
On my side I use only text editor, but all codes are good, one is the 'light" code that use the defaut code, and the second one is the defaut code (complete).

And still no report for temperature ? lot of tuya device are locked and need this request.
You can too take a look on logs, deconz/help/debug view, with flag "DDF", to see DDF output, if you have nothing on them can use "info" and "info_l2" you will see "tuya output", perhaps I m using a bad dpid.

enboer, 2025-03-11:
I am again debugging this  device as you suggest. This is what I see in the info/infl_l2 view. Tried with dpid 1 and 2, but same behavior. Does this info give you any clue?

[image]

)

Smanar, 2025-03-11:
Nope this device use tuya cluster, so need to catch tuya debug.
But I have found the dpid on another project

```
        {
          "name": "state/temperature",
          "parse": {"fn": "tuya", "dpid": 102, "eval": "Item.val = 10 * Attr.val + R.item('config/offset').val;" },
          "read": {"fn": "none"},
          "default": 0
        }
```

and 103 for humidity.

enboer, 2025-03-12:
yes this is working thanks!!

Smanar, 2025-03-14:
Sure, do you want to make the PR ?
Or want I make it ?

enboer, 2025-03-20:
Can you do that, I do not exactly know the process. Thanks

Smanar, 2025-03-20:
Done > https://github.com/dresden-elektronik/deconz-rest-plugin/pull/8142
Have edited the json, but for me it's fine, have just removed useless stuff.