dresden-elektronik/deconz-rest-plugin #7876: Third Reality Zigbee Smart Vibration Sensor 3RVS01031Z

MaStahl84, 2024-08-07:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Third Reality Zigbee Smart Vibration Sensor

### Manufacturer

Third Reality, Inc

### Model identifier

3RVS01031Z

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration
[image]

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest
[image]
[image]

### OTAU
[image]

Smanar, 2024-08-09:
Here there is a long story.
Vibration sensor don't have "off" state, but make a notification at every detection, so for tuya device there is a DDF that use the ZHAPresence type.

Can  make a try with this DDF
```

{
  "schema": "devcap1.schema.json",
  "manufacturername": "Third Reality, Inc",
  "modelid": "3RVS01031Z",
  "vendor": "Third Reality, Inc",
  "product": "Vibration sensor",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_PRESENCE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0500"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0000",
          "0x0001",
          "0x0500"
        ]
      },
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          },
          "default": 0
        },
        {
          "name": "config/duration",
          "default": 1
        },
        {
          "name": "config/enrolled",
          "public": false
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/pending"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/lowbattery",
          "awake": true
        },
        {
          "name": "state/presence",
          "awake": true,
          "default": false
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0500"
    }
  ]
}
```

Smanar, 2024-08-12:
The DDF is not used, the name is still on 0xXXXX form, have you re-included the device ?

MaStahl84, 2024-09-03:
I deleted and resetted the device and joined it again to the network, but it only show up as 0xXXXX. I tried several times, but i did not find a way to join it correctly. The new DDF is found when i open the DDF-Editor.

Smanar, 2024-09-04:
>The new DDF is found when i open the DDF-Editor.

It mean when you make, right clic + edit DDF , you are seen the DDF from here ? https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7876#issuecomment-2278123767 with gold status ?

You can see the path/file on the titlebar of the editor too.

closed 2024-10-03 by github-actions[bot]