dresden-elektronik/deconz-rest-plugin #7881: SNZB-02P temperature and humidity sensor - Sonoff

danieldemus, 2024-08-11:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

SNZB-02P Zigbee Temperature and Humidity Sensor

### Manufacturer

Sonoff/eWeLink

### Model identifier

SNZB-02P

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration
[image]

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest
#### Poll control
[image]

#### Temperature measurement
[image]

#### Relative humidity measurement
[image]

#### OTAU
[image]
[image]

#### Sonoff specific
[image]

Mimiix, 2024-08-11:
@SwoopX looks like they use a new manufacturer name? This was added before afaik in 2.25.0

Smanar, 2024-08-11:
Ha yes right, there is a DDF but can be optimised > https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/sonoff/snzb-02-multisensor.json

Can you try editing the file to have

```
  "manufacturername": ["eWeLink","SONOFF","SONOFF", "eWeLink"],
  "modelid": ["TH01","SNZB-02D","SNZB-02P", "SNZB-02P"],
```

SwoopX, 2024-08-11:
> @SwoopX looks like they use a new manufacturer name? This was added before afaik in 2.25.0

Yeah. What I dislike looking at Smanar's referenced DDF is that the old and new model of the sensor are getting mixed together, although content-wise, they might be the same. Would prefer to have that seperated in the future, but it can easily be detangled.

danieldemus, 2024-08-12:
The manufacturername string is the same as the old SNZB-02: eWeLink. I don't know if that's mapped from some enum.
The main differences between the old and new SNZB-02 (aka TH01) and the SNZB-02P are the OTAU, Poll control and Sonoff specific clusters, which is quite a lot, so mashing it together with the old sensors DDF seems optimistic.
I am not aware of any light sensor in a temperature and humidity sensor, but YMMV...

closed 2024-09-10 by github-actions[bot]