dresden-elektronik/deconz-rest-plugin #7884: Request for Tuya Smart Zigbee Temp and Humidity Sensor manu:"_TZ3000_bjawzodf" / model: "TS0201"

GitHubStephan, 2024-08-12:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

 Tuya Smart Zigbee Temp and Humidity Sensor

### Manufacturer

_TZ3000_bjawzodf

### Model identifier

TS0201

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

GitHubStephan, 2024-08-12:
there is already a "quirk" for this device in ZHA but i dont know if its possible to convert a quirk in a DDF:
https://github.com/zigpy/zha-device-handlers/issues/2862

GitHubStephan, 2024-08-17:
Hab jetzt die json von einem ähnlichen Gerät genommen und Manufacturer angepasst. Einlesen in Phoson war erfolgreich und Daten werden gelesen und aktualisiert. Ich werde mal bis heute Abend beobachten, ob die Daten konsequent weiter im Systen ankommen.

`{
  "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3000_bjawzodf",
  "modelid": "TY0201",
  "vendor": "Tuya",
  "product": "Temperature and humidity locked sensor",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",                   
          "parse": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl:attr"
          }
        },
        {
          "name": "config/offset"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/tuya_unlock"
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "awake": true
        }
      ]
    },
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",                   
          "parse": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "awake": true,
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2;",
            "fn": "zcl:attr"
          }
        },
        {
          "name": "config/offset"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "awake": true
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 3600,
          "max": 84600,
          "change": "0x00000002"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0405",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    }
  ]
}`

Smanar, 2024-08-17:
Hello, there is the model _TZE200_bjawzodf in this DDF https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZE200_bjawzodf_humidity_temp.json
Can you try to add your model to this list ? probably one more clone ?

GitHubStephan, 2024-08-17:
Thanks, the last one you found is the same i used at the end. Since this morning its working without issues. For me topic is solved. Perhaps someone can start a PR with my modified code. I dont know how to do.

Smanar, 2024-08-17:
Sure https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7889
As you have used the same DDF, have just added your device to the list, thx for report and tests.

GitHubStephan, 2024-08-17:
one fault on my side. i started the thread with Model identifier : TS0201. That was not correct.
it is TY0201 (was a typo). can you pls correct it in the PR? sorry for the extra work

Smanar, 2024-08-18:
Np, PR updated

RneeJ, 2024-08-30:
using deCONZ v2.28.1 with ConbeeII.

I have  "manufacturername": "_TZ3210_ncw88jfq",  "modelid": "TS0201", simple . 
I used the json from Stefan. 

Had to place it in usr/../devices.... 
$home/.local... doesn't get read at startup/reboot etc. 

Works well, is recognized in deCONZ and Phoscon App as a sensor (battery/temperature/humidity) . 

Humidity is reported wrong 6,3 % instead of 63 %.
Can i change that in the .json ? Or any other idea ?

Smanar, 2024-08-30:
Hello better to try to add your Modelid/Manufacture Name in this DDF https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZE200_bjawzodf_humidity_temp.json

It already have a *10 corrective.

```

        {
          "name": "state/humidity",
          "parse": {"fn": "tuya", "dpid": 2, "eval": "Item.val = 10 * Attr.val + R.item('config/offset').val;" },
          "read": {"fn": "tuya"},
          "default": 0
        },
```

Smanar, 2024-08-30:
I don't don't found another DDF as sample, but for you better to create a new DDF instead of using an old one and change this part 
```

        {
          "name": "state/humidity",
          "awake": true
        },
```

by
```

{
  "name": "state/humidity",
  "parse": {
    "at": "0x0000",
    "cl": "0x0405",
    "ep": 1,
    "eval": "Item.val = Attr.val * 10 + R.item('config/offset').val",
    "fn": "zcl:attr"
  },
  "default": 0
}
```

RneeJ, 2024-08-30:
Thank you Smanar ! , i figured that out but changed to : 
`{
          "name": "state/humidity",
          "awake": true,
          "parse": {
            "at": "0x0000",
            "cl": "0x0405",
            "ep": 0,
            "eval": "Item.val = Attr.val * 10 + R.item('config/offset').val",
            "fn": "zcl:attr"
          },
          "default": 0
        },`

I also added battry reading 
`  {
          "name": "config/battery",
          "refresh.interval": 7265,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          },
          "default": 0
`

[image]

Smanar, 2024-08-31:
Ha ? the previous battery reading was not working ?
If all is fine can make a PR to submit the device ? Or want I make it ? (if you share the complete working DDF)

RneeJ, 2024-09-01:
@Smanar thank you. you can add the device to the mentioned json. I can’t make a PR.
I just started last week with DeCONZ, Zigbee etc. I still have to learn about them. Neither do i have experience with Github. 
i am also working on some other sensors i bought.

Smanar, 2024-09-01:
Can you copy/paste the full working DDF ?
It prevent typo if I make the PR

RneeJ, 2024-09-01:
Directly from DeCONZ

[_TZ3210_ncw88jfq_temp-hum-sensor.json](https://github.com/user-attachments/files/16828949/_TZ3210_ncw88jfq_temp-hum-sensor.json)

Smanar, 2024-09-01:
Min and Max are not displayed in phoscon, but decimal need to bi visible can you check direclty in the API ? Using Phoscon/help/API Information/sensors

PR created https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7908

RneeJ, 2024-09-02:
Phoscon/help/API Information/sensors :
[image]

But in `http://192.168.2.32:8080/pwa/devices-sensors2.html?_v=8af8975ffa0c93c20a8c5db60e99f4a70b44c8c5,8,1,19&gwid=00212EFFFF09D402#sensor/3`

[image]

Also 25 and not 25,7

Smanar, 2024-09-03:
Ok so the issue is from Phoscon, not the API. IDK why they have made this choice.
Here I can do nothing ....

closed 2024-09-05 by manup