dresden-elektronik/deconz-rest-plugin #7885: Xiaomi Aqara FP1E

thomasgolda, 2024-08-13:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Aqara Millimeter Wave Presence Sensor FP1E

### Manufacturer

Xiaomi

### Model identifier

FP1E

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

SwoopX, 2024-08-14:
@thomasgolda could you please change the model identifier and manufacturer as per our template? Thanks!

thomasgolda, 2024-08-15:
@SwoopX I updated both as given in the basic cluster. Hope that's correct.

SwoopX, 2024-08-15:
Looking good, thanks a lot! Helps finding the issue again 👍

ebaauw, 2024-08-20:
Note that it uses _Manufacturer Code_ 0x1234, and consequently, that version of the 0xFCC0 cluster (with generic attributes).

Sniffing the device while connected to the Aqara Hub M1 reveals the following attributes:

Attr | DT | R | W | Description
-- | -- | -- | -- | --
0x00E8 | bool | - | W | Restart device.
0x010C| u8 | - | - | Motion Sensitivity (got a report, but no corresponding setting in the Aqara app).
0x0142 | u8 | R | - | Presence.
0x0157 | u8 | - | W | Reset Presence (app sets this on AI Learning).
0x0159 | u8 | R | W | Unknown (value 1).
0x015A | u32 | R | W | Unknown (value 600).
0x015B | u32 | R | W | Detection Range (in cm, Aqara app sets from 0 .. 600, in increments of 30).
0x015C | u8 | R | W | Unknown (value 1).
0x015D | u8 | R | W | Unknown (value 1).
0x015E | u8 | R | W | Unknown (value 1).
0x015F | u32 | R | - | Position (distance of detected human, in cm).
0x0160 | u8 | R | - | Unknown (seeing values 2 (when no presence), and 3, 4 (when presence)).

0x0142, 0x015F and 0x0160 are reported frequently, as soon as changes occur.  The other attributes where reported only on device restart.

Also got a Xiaomi special attribute 0x00F7:
```
                     1 2  3  4 5  6 7  8 9  0 1  2 3  4 5  6 7  8  9 0  1  2 3  4  5 6  7  8 9  0
	 attr tp  l    temp   RSSI dB  firmware    parent       ?       ?       ?       ?       ?
         ---- -- -- ------- --------- --------- --------- ------- ------- ------- ------- -------
fp1e:    00f7 41 1e 0328 22 0521 0400 0821 1c01 0a21 0000 0c20 14 1020 01 1220 00 6520 00 6620 01
                 30                  0.0.0_0028
```
This seems to be sent only on restart, not after occupancy changes.

Note that I got a firmware update to 0.0.0_0028.

Also note that the device reacts to _Identify_.

ebaauw, 2024-08-23:
The DDF results in the following resource:
```json
{
  "config": {
    "detectionrange": 600,
    "on": true,
    "reachable": true,
    "resetpresence": false
  },
  "etag": "87b6a6502dc209729465632492469ebf",
  "lastannounced": null,
  "lastseen": "2024-08-23T20:18Z",
  "manufacturername": "aqara",
  "modelid": "lumi.sensor_occupy.agl1",
  "name": "Presence 155",
  "state": {
    "distance": 30,
    "lastupdated": "2024-08-23T20:18:45.196",
    "presence": true
  },
  "swversion": "0.0.0_0028",
  "type": "ZHAPresence",
  "uniqueid": "54:ef:44:10:00:db:19:b1-01-0406"
}
```
Note that the cluster in the `uniqueid` is a blatant lie, just as for the FP1.

closed 2024-09-05 by manup