dresden-elektronik/deconz-rest-plugin #7887: Moes ZSS-S01-GWM-C - Smart Door/Window Sensor

alexborro, 2024-08-15:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Moes ZSS-S01-GWM-C - Smart Door/Window Sensor

### Manufacturer

_TZ3000_8yhypbo7

### Model identifier

TS0203

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration
[image]

### IAS Zone
[image]

Smanar, 2024-08-15:
Hello can you  try to adding your device to this list ?

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_TS0203_door_sensor.json

alexborro, 2024-08-15:
Very good, it shows up like the pictures below.
The Open/Close state is working, but it didnt display any other information, like the battery level (it seems to have it according to the Power Configuration cluster).

Any idea how I can try to add it ?

When finished, I can make a PR for these changes.

Cheers.
Alex.

[image]

[image]

Smanar, 2024-08-16:
The DDF (try to) catch the battery level too, perhpas need some time or the settings have failed at inclusion.
But this device haven't a "force read" every 24h.

If after 24h you still have nothing, try to re-include the device, without deleting it.

If after 24h again try adding on config/battery part

```
          "refresh.interval": 86400,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl:attr"
          },
```

closed 2024-09-14 by github-actions[bot]