dresden-elektronik/deconz-rest-plugin #7888: TZ3000 BSEED reports wrong power

enboer, 2024-08-16:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

_TZ3000_4ux0ondb

### Manufacturer

BSEED

### Model identifier

https://www.bseed.com/products/bseed-zigbee-eu-wall-sockets-power-outlets-with-energy-monitoring-kids-protection

### Device type to add

Switch

### Node info

https://www.bseed.com/products/bseed-zigbee-eu-wall-sockets-power-outlets-with-energy-monitoring-kids-protection

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off
[image]

### Level Control
[image]

### Color Control

### Thermostat

### Simple Metering
[image]

### Electrical Measurement

### Any other cluster of relevance/interest

enboer, 2024-08-16:
Device is working but reports wrong RMS current and Active Power. In this measurement picture the RMS current should be around 186, and the Active power should be around 42Watt

Smanar, 2024-08-16:
>but reports wrong RMS current and Active Power

Youi have made a DDF ? I don't found this device on my side

enboer, 2024-08-16:
Device was added to Deconz. When I do a added DDF I see this:
[image]

The metering is then already working (then already with wrong values) but I do not see the Power/Voltage device in Domoticz.

I then dragged these metering to the binding section of the DDF Editor. But that does not work yet.

Smanar, 2024-08-16:
Your DDF is not used, you have the "draft" status.
You can try with editing this one https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3000_cehuw1lw_smartplug_EU.json

Can just replace the manufacture name by yours.

enboer, 2024-08-16:
I did change that to Gold later. But you json file is working, however I still have incorrect power value. It reports 1000Watt, but this should be 40-45watt. I see these values already in the DeconzGUI.

Smanar, 2024-08-17:
On your capture, I m seing U = 233 and I = 2.3A so the power is more than 500W, why 45 W, what is the device connected to the plug ?

enboer, 2024-08-18:
Here are the correct screenshots. I have put 2 devices in serie. You see the difference in domoticz. And also 2 screenshots of the value in Deconz. The one is I 186 and Power of 42 is correct. The TZ3000 reports 2357 and 1130Watt for the same load.

[image]
[image]
[image]

enboer, 2024-08-18:
That strange as for other zigbee devices these values can be used directly. Do I need to do that in the DDF? or in eg Domoticz via some automation?

Smanar, 2024-08-18:
>Do I need to do that in the DDF

You mean a "convertion" like the /12 needed for the current ? So yes, but if you have another device with the same manufacture name / Model ID, I almost sure it will not need it, for me the problem is from this device.

If you want to use a "corrective", can create a DDF (or clone an existing one and change the ManufactureName/ModelID) and use for exemple

```
        {
          "name": "state/current",
          "read": {
            "at": "0x0508",
            "cl": "0x0b04",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0508",
            "cl": "0x0b04",
            "ep": 1,
            "eval": "Item.val = Attr.val / 12"
          },
          "refresh.interval": 365
        },
```

This code will /12 the raw value for the current, and Domoticz will use the corrected one.

enboer, 2024-08-18:
Thanks this works, seems it should be / 24. Is there a way to reset the total consumed power. I see the value in Deconz, but that is readonly.

Smanar, 2024-08-20:
Yes If I remember, on the GUI on the cluster 0x0000 the "Basic", you have a "reset" command.

closed 2024-09-18 by github-actions[bot]