dresden-elektronik/deconz-rest-plugin #7891: Sensor status stops working after a short time

Darkem1, 2024-08-19:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Since the latest update, all of my sensors stop working after a while. After a restart everything is still reported correctly (example: door open/closed), but after about 30 minutes it no longer works. The status of all sensors essentially freezes and is no longer updated. Everyone is online. When you manually click buttons on the sensors, they also show a transmission, but the status no longer changes. After restarting deconz it works again for about 30 minutes.
ConBee 3, latest firmware, deConz is now beta channel, but the bug was also present in the last stable version.
Sensors are predominantly Aqara window sensors, but also temperature sensors.

### Steps to reproduce the behavior

Restart deConz and wait

### Expected behavior

Updating the correct status

### Environment

- Host system: Raspberry Pi
- Running method: Raspbian
- Firmware version: 26530900
- deCONZ version (not Home assistant Addon version!): 2.28.0
- Device: ConBee III
- Do you use an USB extension cable: no
- Is there any other USB or serial devices connected to the host system? no

Mimiix, 2024-08-19:
Hi,

As this is not a bug as described in #5113 , i am closing this issue.

Please ask this on the forums and provide a screenshot of the deCONZ gui aswell.

Kind regards,

closed 2024-08-19 by Mimiix

Darkem1, 2024-08-19:
Why is this not a bug?