dresden-elektronik/deconz-rest-plugin #7892: Zigbee Solar Rain Sensor RB-SRAIN01

Idaho947, 2024-08-20:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Zigbee Solar Rain Sensor

### Manufacturer

Tuya

### Model identifier

RB-SRAIN01

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

[image]

### Further relevant clusters

### Power Configuration
[image]

### Scene
[image]

### IAS Zone
[image]

### Groups
[image]

More usefull information here :
https://smarthomescene.com/reviews/tuya-zigbee-solar-rain-sensor-rb-srain01-review/

Smanar, 2024-08-21:
Lol, seriously you are specialised in fucking exotic device .....

What to do with his one ....

`[101, 'illuminance', tuya.valueConverter.raw]`
ok can use `state/lightlevel`

But for others

```
            [102, 'illuminance_average_20min', tuya.valueConverter.raw],
            [103, 'illuminance_maximum_today', tuya.valueConverter.raw],
            [104, 'cleaning_reminder', tuya.valueConverter.trueFalse],
            [105, 'rain_intensity', tuya.valueConverter.raw],
```

?????

Idaho947, 2024-08-21:
yes, I love testing things! Most of the time I resell them but this one, a zigbee peripheral with solar power, I couldn't pass it up ^^.

I've put a link in topic 1 that explains how it works in English much better than I could. 
https://smarthomescene.com/reviews/tuya-zigbee-solar-rain-sensor-rb-srain01-review

it displays intensities that must then be converted in the home automation software into something that corresponds to reality.

Shikaban, 2024-09-01:
I have the same device, and while reading the information in raw format is possible, having it integrated into Deconz would be awesome. Is there any way I can contribute to help with this?

Smanar, 2024-09-01:
But my problem is how to show the device in the API.
I can't use state/voltage for a ZHAHumidity, the API is realy locked.

 And not sure create 3 ZHAPower is a good thing, hard for you to reconise what is what ....

Idaho947, 2024-09-10:
> Everything seams good, but I can't integrate the ddf file, it keeps failing

What do you mean ? It doesn't recongnize you sensor ?

Shikaban, 2024-09-10:
No the sensor is inside the deconz ( I can only see him from the gui and not from phoscon )
But I've tried to load the ddf even restarting the raspberry or the service but nothing changed sensor is there but no information available.

Idaho947, 2024-09-15:
I try this week end and my deconz server can't had the sensor...

Idaho947, 2024-09-15:
OK done so got that in phoscon :
[image]

But nothing in HA. The entitie is not created.

Smanar, 2024-09-16:
For me it's normal.
ZHAHumidity for exemple have generaly a state/humidity but not a state/voltage, so HA ignore them.

Idaho947, 2024-09-16:
Yes logic.

So I can t use it in home assistant ?

Shikaban, 2024-09-21:
How you managed to import the ddf in deconz? I always get "failed to import"

Idaho947, 2024-09-21:
I put it in the tuya folder in device folder.

RneeJ, 2024-09-30:
I want to buy it to, would be great if it could be used. 

I found this recent info : [https://github.com/zigpy/zha-device-handlers/issues/3249](url)

closed 2024-10-29 by github-actions[bot]

Beersteak, 2025-04-08:
Hi,

i dont know if its the right way but i like to reopen this device request. There has been some development in the meantime.
The sensor has been added to ZHA, before that you had to use a custom quirk.
I got it running in deCONZ, thanks to @Smanar 
I just changed the state/voltage to state/humidity for the dpid 105.

Here's my .json
`{
    "schema": "devcap1.schema.json",
    "manufacturername": "_TZ3210_tgvtvdoc",
    "modelid": "TS0207",
    "vendor": "Tuya",
    "product": "Solar Rain Sensor",
    "sleeper": true,
    "status": "Gold",
    "ddfvalidate": false,
    "subdevices": [
        {
            "type": "$TYPE_HUMIDITY_SENSOR",
            "restapi": "/sensors",
            "uuid": [
                "$address.ext",
                "0x01",
                "0x0405"
            ],
            "items": [
                {
                    "name": "attr/id"
                },
                {
                    "name": "attr/lastannounced"
                },
                {
                    "name": "attr/lastseen"
                },
                {
                    "name": "attr/manufacturername"
                },
                {
                    "name": "attr/modelid"
                },
                {
                    "name": "attr/name"
                },
                {
                    "name": "attr/swversion",
                    "parse": {
                        "fn": "zcl:attr",
                        "ep": 1,
                        "cl": "0x0000",
                        "at": "0x0001",
                        "script": "../tuya/tuya_swversion.js"
                    },
                    "read": {
                        "fn": "zcl:attr",
                        "ep": 1,
                        "cl": "0x0000",
                        "at": "0x0001"
                    }
                },
                {
                    "name": "attr/type"
                },
                {
                    "name": "attr/uniqueid"
                },
                {
                    "name": "config/battery",
                    "parse": {
[... 107 lines cut ...]
                {
                    "name": "state/lastupdated"
                },
                {
                    "name": "state/voltage",
                    "parse": {
                        "fn": "tuya",
                        "dpid": 101,
                        "eval": "Item.val = Attr.val;"
                    },
                    "read": {
                        "fn": "none"
                    },
                    "default": 0
                }
            ]
        }
    ]
}`

it now looks like this:

[image]

BUT it is still not exposed to Home Assistant. Can someone help me and maybe then we can add the sensor to a future release?

Smanar, 2025-04-08:
Thx, do you want to make a Pr to correct the DDF or want I make it ?

And looking DDF you need to have at least the ZHAhumidity on HA.
But I m agree the ZHALightLevel is "exotic" with the state/voltage (dpid 101), It's not possible to convert the voltage to lightlvel ?

Beersteak, 2025-04-09:
Thanks for your answer, i tried a few new things and i am a bit further; if we fix the problem i can make a Pr myself

The new Version is
```
{
   "schema": "devcap1.schema.json",
   "manufacturername": "_TZ3210_tgvtvdoc",
   "modelid": "TS0207",
   "vendor": "Tuya",
   "product": "Solar Rain Sensor",
   "sleeper": true,
   "status": "Gold",
   "ddfvalidate": false,
   "subdevices": [
       {
           "type": "$TYPE_LIGHT_LEVEL_SENSOR",
           "restapi": "/sensors",
           "uuid": [
               "$address.ext",
               "0x01",
               "0x0400"
           ],
           "items": [
               {
                   "name": "attr/id"
               },
               {
                   "name": "attr/lastannounced"
               },
               {
                   "name": "attr/lastseen"
               },
               {
                   "name": "attr/manufacturername"
               },
               {
                   "name": "attr/modelid"
               },
               {
                   "name": "attr/name"
               },
               {
                   "name": "attr/swversion",
                   "parse": {
                       "fn": "zcl:attr",
                       "ep": 1,
                       "cl": "0x0000",
                       "at": "0x0001",
                       "script": "../tuya/tuya_swversion.js"
                   },
                   "read": {
                       "fn": "zcl:attr",
                       "ep": 1,
                       "cl": "0x0000",
                       "at": "0x0001"
                   }
               },
               {
                   "name": "attr/type"
               },
               {
                   "name": "attr/uniqueid"
               },
               {
                   "name": "config/offset",
                   "default": 0
               },
               {
                   "name": "config/on"
               },
               {
                   "name": "config/reachable"
               },
               {
                   "name": "state/lastupdated"
               },
               {
                   "name": "state/lightlevel",
                   "parse": {
                       "fn": "tuya",
                       "dpid": 101,
                       "eval": "Item.val = Math.round(10000 * Math.log10(Attr.val) + 1);"
                   },
                   "read": {
                       "fn": "none"
                   },
                   "default": 0
               }
           ]
       },
       {
           "type": "$TYPE_HUMIDITY_SENSOR",
           "restapi": "/sensors",
           "uuid": [
               "$address.ext",
               "0x01",
               "0x0405"
           ],
           "items": [
               {
                   "name": "attr/id"
               },
               {
                   "name": "attr/lastannounced"
                },
                {
                    "name": "attr/lastseen"
                },
                {
                    "name": "attr/manufacturername"
                },
                {
                    "name": "attr/modelid"
                },
                {
                    "name": "attr/name"
                },
                {
                    "name": "attr/swversion",
                    "parse": {
                        "fn": "zcl:attr",
                        "ep": 1,
                        "cl": "0x0000",
                        "at": "0x0001",
                        "script": "../tuya/tuya_swversion.js"
                    },
                    "read": {
                        "fn": "zcl:attr",
                        "ep": 1,
                        "cl": "0x0000",
                        "at": "0x0001"
                    }
                },
                {
                    "name": "attr/type"
                },
                {
                    "name": "attr/uniqueid"
                },
				{
				  "name": "config/battery",
				  "awake": true,
				  "refresh.interval": 7265,
				  "read": {
					"at": "0x0021",
					"cl": "0x0001",
					"ep": 1,
					"fn": "zcl:attr"
				  },
				  "parse": {
					"at": "0x0021",
					"cl": "0x0001",
					"ep": 1,
					"eval": "Item.val = Attr.val / 2"
				  },
				  "default": 0
				},
                {
                   "name": "state/humidity",
                   "parse": {
                       "fn": "tuya",
                       "dpid": 105,
                       "eval": "Item.val = Attr.val;"
                   },
                   "read": {
                       "fn": "none"
                   },
                   "default": 0
               }
           ]
       }
   ]
}
```
in deCONZ it now looks like this:

[image]

But in Home Assistant the Lightsensor is missing:

[image]

i can see the Lightlevel in the Diagnostics:

[image]

do you have an idea?

Smanar, 2025-04-09:
Yep for me, the DDF is fine.
When the HA plugin detect a new device it make his magic, check the fields and add stuff or not to HA.
If the DDF is edited, the json in the API change, but if I m right HA don't remake his magic again.

You don't have a way to remove the device from HA for it detect it again ?

Beersteak, 2025-04-14:
ok so, i tried adding the device from scratch with my DDF, works almost fine.
the state/voltage and state/lux are gone, but the battery is not there, but in deConz.

I tried adding it as TYPE_BATTERY_SENSOR
'''
{
   "schema": "devcap1.schema.json",
   "manufacturername": "_TZ3210_tgvtvdoc",
   "modelid": "TS0207",
   "vendor": "Tuya",
   "product": "Solar Rain Sensor",
   "sleeper": true,
   "status": "Gold",
   "ddfvalidate": false,
   "subdevices": [
       {
           "type": "$TYPE_LIGHT_LEVEL_SENSOR",
           "restapi": "/sensors",
           "uuid": [
               "$address.ext",
               "0x01",
               "0x0400"
           ],
           "items": [
               {
                   "name": "attr/id"
               },
               {
                   "name": "attr/lastannounced"
               },
               {
                   "name": "attr/lastseen"
               },
               {
                   "name": "attr/manufacturername"
               },
               {
                   "name": "attr/modelid"
               },
               {
                   "name": "attr/name"
               },
               {
                   "name": "attr/swversion",
                   "parse": {
                       "fn": "zcl:attr",
                       "ep": 1,
                       "cl": "0x0000",
                       "at": "0x0001",
                       "script": "tuya_swversion.js"
                   },
                   "read": {
                       "fn": "zcl:attr",
                       "ep": 1,
                       "cl": "0x0000",
                       "at": "0x0001"
                   }
               },
               {
                   "name": "attr/type"
               },
               {
                   "name": "attr/uniqueid"
               },
               {
                   "name": "config/offset",
                   "default": 0
               },
               {
                   "name": "config/on"
               },
               {
                   "name": "config/reachable"
               },
               {
                   "name": "state/lastupdated"
               },
               {
                   "name": "state/lightlevel",
                   "parse": {
                       "fn": "tuya",
                       "dpid": 101,
                       "eval": "Item.val = Math.round(10000 * Math.log10(Attr.val) + 1);"
                   },
                   "read": {
                       "fn": "none"
                   },
                   "default": 0
               }
           ]
       },
	   {
		  "type": "$TYPE_BATTERY_SENSOR",
		  "restapi": "/sensors",
		  "uuid": [
			"$address.ext",
			"0x01",
			"0x0001"
		  ],
		  "fingerprint": {
			"profile": "0x0104",
			"device": "0x0402",
			"endpoint": "0x01",
			"in": [
			  "0x0001",
			  "0x0102"
			]
		  },
		  "items": [
			{
			  "name": "attr/manufacturername"
			},
			{
			  "name": "attr/modelid"
			},
			{
			  "name": "attr/name"
			},
			{
			  "name": "attr/swversion",
			  "refresh.interval": 86400,
			  "parse": {
				"fn": "zcl:attr",
				"ep": 1,
				"cl": "0x0000",
				"at": "0x0001",
				"script": "tuya_swversion.js"
			  },
			  "read": {
				"fn": "zcl:attr",
				"ep": 1,
				"cl": "0x0000",
				"at": "0x0001"
			  }
			},
			{
			  "name": "attr/type"
			},
			{
			  "name": "attr/uniqueid"
			},
			{
			  "name": "attr/lastseen"
			},
			{
			  "name": "config/on"
			},
			{
			  "name": "config/reachable"
			},
			{
			  "name": "state/battery",
			  "parse": {
				"cl": "0x0001",
				"at": "0x0021",
				"eval": "Item.val = Attr.val / 2"
			  },
			  "read": {
				"cl": "0x0001",
				"at": "0x0021"
			  },
			  "refresh.interval": 3600
			},
			{
			  "name": "state/lastupdated"
			}
		  ]
	   },
       {
           "type": "$TYPE_HUMIDITY_SENSOR",
           "restapi": "/sensors",
           "uuid": [
               "$address.ext",
               "0x01",
               "0x0405"
           ],
           "items": [
               {
                   "name": "attr/id"
               },
               {
                   "name": "attr/lastannounced"
                },
                {
                    "name": "attr/lastseen"
                },
                {
                    "name": "attr/manufacturername"
                },
                {
                    "name": "attr/modelid"
                },
                {
                    "name": "attr/name"
                },
                {
                    "name": "attr/swversion",
                    "parse": {
                        "fn": "zcl:attr",
                        "ep": 1,
                        "cl": "0x0000",
                        "at": "0x0001",
                        "script": "tuya_swversion.js"
                    },
                    "read": {
                        "fn": "zcl:attr",
                        "ep": 1,
                        "cl": "0x0000",
                        "at": "0x0001"
                    }
                },
                {
                    "name": "attr/type"
                },
                {
                    "name": "attr/uniqueid"
                },
                {
                   "name": "state/humidity",
                   "parse": {
                       "fn": "tuya",
                       "dpid": 105,
                       "eval": "Item.val = Attr.val;"
                   },
                   "read": {
                       "fn": "none"
                   },
                   "default": 0
               }
           ]
       }
   ]
}
'''

[image]

[image]

i see it in the Diagnostic Data:

[image]

i deleted it, restartet Home Assistant completely, readded the device but its still the same

Smanar, 2025-04-14:
The ZHABattery is used for "light type" device (by defaut this kind of device don't have the "config" field, so not possible to use config/battery).

But your first DDF was fine ? I can see a 93 value for config/battery ?

You can miss a bind report, but for me your first DDF was fine for battery.

```
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
```

Beersteak, 2025-04-14:
#8171 i created a pull request, it seems to work now

Smanar, 2025-04-15:
I have answered on your PR page, I think you have removed too much stuff ^^