dresden-elektronik/deconz-rest-plugin #7901: Sonoff Smart Water valve

mscodeNEO, 2024-08-25:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Smart valve

### Manufacturer

Sonoff

### Model identifier

SWV

### Device type to add

Other

### Node info

[image]

### Endpoints and clusters

[image]
[image]
[image]
[image]
[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Smanar, 2024-08-25:
Hello, again ^^

https://forum.phoscon.de/t/sonoff-smart-water-valve-not-pairing/4994

So the DDF corrected

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "SONOFF",
  "modelid": "SWV",
  "vendor": "Sonoff",
  "product": "Smart water valve",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "config/battery",
          "parse": {"at": "0x0021", "cl": "0x0001", "ep": 1, "eval": "Item.val = Attr.val"},
          "read": {"fn": "zcl:attr", "ep": 1, "cl": "0x0001", "at": "0x0021"},
          "refresh.interval": 4000,
          "awake": true
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on"
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 300
        }
      ]
    }
  ]
}
```

To test it, if you haven't HA, you just need to create a text file, called what_you_want.json in a "devices" folder, then restart deconz, and re-include the device

`All DDF provided directly with deCONZ typically reside in /usr/share/deCONZ/devices/ on a Linux system and are loaded first. However, files residing in the home directory of the user running deCONZ (e.g. /home/<DECONZUSER>/.local/share/dresden-elektronik/deCONZ/devices) will override the pre-packaged files to allow users to amend and keep their own files if desired`

mscodeNEO, 2024-08-25:
Hello Smanar,

you are very speedy, very cool.
A)  create a text file, called what_you_want.json ->> **done** (easy)
B) in a "devices" folder --> usr/share/...* ....  usr/deconzuser/.local/...*   --> both **done** (easy)
C) then restart deconz   -> done (easy, reboot raspberry), 
D) and re-include the device --> **does not work** :-( , I set the sonoff device into pairing mode and sear for new devices in lights category in phoscon web interface, no device is found (try also to delete the node in decon-gui, no success, am I doing something wrong? 
By the way I am wondering why is the device connected in decon-gui, otherwise I could not provide the screenshots, I know, but when it is connected, maybe I cannot connect it twice in phoscon? I think I do not really understand the zigbee philosophy of pairing, nodes, clusters, ....

Maybe there are some issues of read permission rights for deconz and the new jsons files, could this be the problem?

cu
matthias

mscodeNEO, 2024-08-26:
I put the  file here:
[image]
and here (as I understand you right this is the first place, deCONZ is looking for (new?) json files):
[image]

Pairing Mode and "search new lights" does not find the device :-(

cu
matthias

mscodeNEO, 2024-08-26:
Isn't it a problem, that the node info just provide "unknown" info for manufaturer and model indentifier, and just privide these info int the basic cluster infos?
[image]

Smanar, 2024-08-26:
Sorry was my fault, have made a typo, so the DDF was not loaded, here the corrected DDF

```
{
    "schema": "devcap1.schema.json",
    "manufacturername": "SONOFF",
    "modelid": "SWV",
    "vendor": "Sonoff",
    "product": "Smart water valve",
    "sleeper": true,
    "status": "Gold",
    "subdevices": [
        {
            "type": "$TYPE_ON_OFF_OUTPUT",
            "restapi": "/lights",
            "uuid": [
                "$address.ext",
                "0x01"
            ],
            "items": [
                {
                    "name": "attr/id"
                },
                {
                    "name": "attr/lastannounced"
                },
                {
                    "name": "attr/lastseen"
                },
                {
                    "name": "config/battery",
                    "parse": {
                        "at": "0x0021",
                        "cl": "0x0001",
                        "ep": 1,
                        "eval": "Item.val = Attr.val"
                    },
                    "read": {
                        "fn": "zcl:attr",
                        "ep": 1,
                        "cl": "0x0001",
                        "at": "0x0021"
                    },
                    "refresh.interval": 4000,
                    "awake": true
                },
                {
                    "name": "attr/manufacturername"
                },
                {
                    "name": "attr/modelid"
                },
                {
                    "name": "attr/name"
                },
                {
                    "name": "attr/swversion"
                },
                {
                    "name": "attr/type"
                },
                {
                    "name": "attr/uniqueid"
                },
                {
                    "name": "state/on"
                }
            ]
        }
    ],
    "bindings": [
        {
            "bind": "unicast",
            "src.ep": 1,
            "cl": "0x0001",
            "report": [
                {
                    "at": "0x0021",
                    "dt": "0x20",
                    "min": 60,
                    "max": 3600,
                    "change": "0x00000001"
                }
            ]
        },
        {
            "bind": "unicast",
            "src.ep": 1,
            "dst.ep": 1,
            "cl": "0x0006",
            "report": [
                {
                    "at": "0x0000",
                    "dt": "0x10",
                    "min": 1,
                    "max": 300
                }
            ]
        }
    ]
}
```

Deconz select DDF using only
```
    "manufacturername": "SONOFF",
    "modelid": "SWV",
```

You can too select the node, right clic, then "edit DDF" you will see an autogenerated DDF, you can compare value.

>Isn't it a problem, that the node info just provide "unknown" info for manufaturer and model indentifier, and just provide these info in the basic cluster infos

Don't worry when the device come in the network, the API will ask for informations and not all is updated on the GUI, and if you can see the value on the "Basic" cluster and not in this panel can be just a refresh issue, you have tried to close and re open the panel ?

Smanar, 2024-08-26:
For the (A) it's a deconz standard, the battery voltage is converted in battery percent, more understandable.
For the (C) there is a native deconz request with "ontime", but I don't remember how it work, not on the help page, probably an hidden request, what is your third app ? 
For the (B) you can try that https://github.com/Smanar/deconz-rest-plugin/commit/91d5eb5f4bdf4fde70e4c3bd53856a118410c004

More work to do, you have 1 more file to create, the  state_flow_item.json, for the other, you can just change the contain.

I have added a ZHAWater sensor, and inside this one you will have state/flow and state/water (for leakage)

theflyer08, 2024-08-31:
I tested the newest ddf. It functioned correctly for approximately one minute, but then the valve became unresponsive to on/off commands. Curiously, Phoscon continued to show the valve's status accurately.

Smanar, 2024-08-31:
It stop to react to on/off request ? But return still working ?
I don't see something about this issue on google ... 
Can try with

`"sleeper": false,`

mscodeNEO, 2024-08-31:
The ddf works fine to me, because on/off works quiet well. The only issue is, that this state sometimes (not reproducible) switch to on (bad for water bill😁). The workarround is a script to handle this behavior. I do not think this is because of the ddf, but I never chance any parameter to this sononoff valve. I have no third party app. Just had a look to deconz-gui to make some screenshots.

cu matthias

theflyer08, 2024-08-31:
> It stop to react to on/off request ? But return still working ?

Yes. 

> I don't see something about this issue on google ... 
> Can try with
> 
> `"sleeper": false,`

I will try that and give feedback.

> 
> (7th line) This device is not a sleeper.

Smanar, 2024-09-01:
>that this state sometimes (not reproducible) switch to on by themselve (bad for water bill😁). The workarround is a script to handle this behavior.

It seem this device have a security.
>Automatically shut down the water valve after the water shortage exceeds X minutes

I can add it in the DDF but IDK how it work on the device ?

Can try too increase the report time for the on/off, here a new DDF to test.

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": "SONOFF",
  "modelid": "SWV",
  "vendor": "Sonoff",
  "product": "Smart water valve",
  "sleeper": false,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_ON_OFF_OUTPUT",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "config/battery",
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val"
          },
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0001",
            "at": "0x0021"
          },
          "refresh.interval": 4000,
          "awake": true
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/on"
        }
      ]
    },
    {
      "type": "$TYPE_WATER_LEAK_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0404"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val"
          },
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0001",
            "at": "0x0021"
          },
          "refresh.interval": 4000,
          "awake": true
        },
        {
          "name": "config/lock",
          "parse": {
            "at": "0x5011",
            "cl": "0xfc11",
            "ep": 1,
            "eval": "Item.val = Attr.val != 0"
          },
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0xfc11",
            "at": "0x5011"
          },
          "write": {
            "at": "0x5011",
            "cl": "0xfc11",
            "dt": "0x21",
            "ep": "0x01",
            "eval": "if (Item.val) {30} else {0}",
            "fn": "zcl:attr"
          },
          "refresh.interval": 4000,
          "awake": true
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/flow"
        },
        {
          "name": "state/water",
          "parse": {
            "at": "0x500C",
            "cl": "0xFC11",
            "ep": 1,
            "eval": "Item.val = Attr.val != 0;",
            "fn": "zcl:attr"
          }
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0006",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x10",
          "min": 1,
          "max": 1800
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0404",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 10,
          "max": 3600,
          "change": "0x0000000A"
        }
      ]
    }
  ]
}
```

mscodeNEO, 2024-09-06:
I just overwrite the new code to existing json file in the device folder on the RPi4 and restart the deconz instance, is this the correct precudure to update the json or do I have to repair the device? 
I set the transition time to 10 s, and after setting on to true, 10 s later it switches to off, as expected I guess. 
I don'T know if this works also before overwriting the old code. 

What do you mean by "security", switching to "on" seems to me a very dangerous and strongly unintended security feature.

What is do do, so that the device is not handle es an light, there are a lot of paramters, which obviously are made for lights:
[image]

Smanar, 2024-09-06:
>I just overwrite the new code to existing json

You mean a random one ? Can create a new one too, less destructive (in a "devices" folder)

>and restart the deconz instance, is this the correct precudure to update the json or do I have to repair the device?

Yes, deconz load DDF only at start so need to restart it, and need to re-include the device for deconz create the entry.

>What do you mean by "security", switching to "on" seems to me a very dangerous and strongly unintended security feature.

No, it switch to "off" after a time.

>What is to do, so that the device is not handle as a light, there are a lot of paramters, which obviously are made for lights:

Yes there is a problem, "state/bri" is not in the DDF for exemple, you are sure it's the good entry ?

mscodeNEO, 2024-09-06:
I delete the object tree in iobroker and restart deconz and make a new screenshot, now it looks like this:
[image]

Smanar, 2024-09-07:
>an overwrite the existing json file on my RP4 in the device folder

Bu there is not existing file ! https://github.com/dresden-elektronik/deconz-rest-plugin/tree/master/devices/sonoff

Can you share the json from the API direclty pls ? visible in Phoscon/help/debug view/light (only the "light" one)

mscodeNEO, 2024-09-07:
> Sorry was my fault, have made a typo, so the DDF was not loaded, here the corrected DDF
>
> ```
> {
>     "schema": "devcap1.schema.json",
>     "manufacturername": "SONOFF",
>     "modelid": "SWV", [...]

I am sorry, we were talking past each other. This initial code was the first proposal by you, which enable succesfully to find the sonoff device in my case. With "overwrite" I meant, I have overwritten this (your) code by your new version from 1st Semtember.

mscodeNEO, 2024-09-07:
> > an overwrite the existing json file on my RP4 in the device folder
> 
> But there is not existing file ! https://github.com/dresden-elektronik/deconz-rest-plugin/tree/master/devices/sonoff
> 
> Can you share the json from the API direclty pls ? visible in Phoscon/help/debug view/light/3 (only the "light" one)

I guess you meant this, right?
[image]

`

                    
{
    "schema": "devcap1.schema.json",
    "manufacturername": "SONOFF",
    "modelid": "SWV",
    "vendor": "Sonoff",
    "product": "Smart water valve",
    "sleeper": true,
    "status": "Gold",
    "path": "/devices/SONOFF/sonoff_smart_water_valve.json",
    "subdevices": [
        {
            "type": "$TYPE_ON_OFF_OUTPUT",
            "restapi": "/lights",
            "uuid": [
                "$address.ext",
                "0x01"
            ],
            "items": [
                {
                    "name": "attr/id"
                },
                {
                    "name": "attr/lastannounced"
                },
                {
                    "name": "attr/lastseen"
                },
                {
                    "name": "config/battery",
                    "awake": true,
                    "refresh.interval": 4000,
                    "read": {
                        "at": "0x0021",
                        "cl": "0x0001",
                        "ep": 1,
                        "fn": "zcl:attr"
                    },
                    "parse": {
                        "at": "0x0021",
                        "cl": "0x0001",
                        "ep": 1,
                        "eval": "Item.val = Attr.val"
                    },
                    "default": 0
                },
                {
                    "name": "attr/manufacturername"
                },
                {
                    "name": "attr/modelid"
                },
                {
                    "name": "attr/name"
                },
                {
                    "name": "attr/swversion"
                },
                {
                    "name": "attr/type"
                },
                {
                    "name": "attr/uniqueid"
                },
                {
                    "name": "state/on",
                    "refresh.interval": 5
                }
            ]
        }
    ],
    "bindings": [
        {
            "bind": "unicast",
            "src.ep": 1,
            "cl": "0x0001",
            "report": [
                {
                    "at": "0x0021",
                    "dt": "0x20",
                    "min": 60,
                    "max": 3600,
                    "change": "0x00000001"
                }
            ]
        },
        {
            "bind": "unicast",
            "src.ep": 1,
            "dst.ep": 1,
            "cl": "0x0006",
            "report": [
                {
                    "at": "0x0000",
                    "dt": "0x10",
                    "min": 1,
                    "max": 300
                }
            ]
        }
    ]
}

                    
                `

Smanar, 2024-09-07:
>I guess you mean this, right?

Nope the device json, see in the api, you can see it in Phoscon/help/API Information/light and an your capture it's the ID = 3.

mscodeNEO, 2024-09-07:
This one?
[image]
```
{
    "config": {
        "groups": [
            "10"
        ]
    },
    "etag": "a160dac10178c77e11215f79ac8f85f6",
    "hascolor": false,
    "lastannounced": "2024-09-05T09:34:44Z",
    "lastseen": "2024-09-07T19:27Z",
    "manufacturername": "SONOFF",
    "modelid": "SWV",
    "name": "Sonoff_Smart_Water_Valve",
    "state": {
        "on": false,
        "reachable": true
    },
    "swversion": "1.0.2",
    "type": "On/Off output",
    "uniqueid": "84:27:12:ff:fe:17:f6:d2-01"
}
```

Smanar, 2024-09-08:
Yes, so this one is exactly like on the DDF, as you can see you have only state/on and state/reacheable, so on your capture it's perhaps old entries ? There is no "action" or "alert" for exemple.

craggyh, 2024-09-09:
I'm getting Import Failed in deCONZ when trying to import my DDF based on the code above. Do I need anything else to be included in the file?

mscodeNEO, 2024-09-12:
No I know, why my sonoff valve is somestime turning on. The valve configuration is now par of one of my lights, and the motion sensor now triggers the valve and no longer triggers the light. That is why my sonoff valve seems to have light properties. How could that happen? How can I fix this, I think unpair both devices and start new pairing or is there a more elegant way?

cu
matthias

Smanar, 2024-09-16:
@craggyh how are you doing ? what is your OS ? You generaly need to create a text file in a "devices" folder,  called what_you_want.json with the DDF contain.

>where does this information/ properties come from and why they are assigned to the sonoff device?

I realy don't see, for deconz side, you can see the contain direclty in the API using phoscon/help/API Informatio/light.
Even the type "On/Off output", it's used generaly for plug for exemple, there is no dim action possible.
It's perhpas something by defaut on IOBroker side ? you haven't a plug for exemple to compare ?

korkmazk, 2024-09-18:
Thank you so much for this ddf. I received the valve today and on/off is working in HomeAssistant (HA OS).
Hope someone will expand the integration and it will be included in Deconz

Smanar, 2024-09-19:
You are using the version with state/flow ? https://github.com/Smanar/deconz-rest-plugin/commit/91d5eb5f4bdf4fde70e4c3bd53856a118410c004

korkmazk, 2024-09-19:
I used https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7901#issuecomment-2335181822

I compared the file and its not the same as yours. Is that one have more options. I could change de ddf and re-pair the device if it helps getting the integration better. Let me know!

Smanar, 2024-09-20:
Yes the other version are more options, like alarm or flow measurement.
But you need to add 2 files to test it.

korkmazk, 2024-09-20:
I added both files to deconz and repaired but not able to turn on or off the valve anymore. Also a weird thing, I have two devices, a philips light and another unknow one thats not visible in HA. When turning on the valve manualy via the device button, the philips light turns on of but never turns off anymore

[image]

korkmazk, 2024-09-20:
It seems to be working, will test a bit:
[image]

At first i had some issues with deconz and the device was recognized as a Philips light and after deleting all entries and re-pair it worked.

craggyh, 2024-09-20:
> @craggyh how are you doing ? what is your OS ? You generaly need to create a text file in a "devices" folder, called what_you_want.json with the DDF contain.
>
>
> I realy don't see, for deconz side, [...]

I have Phoscon/deCONZ installed as a Homeassistant Add-on. I was attempting to do this by using the Device integration function in deCONZ UI. I created a text file with the DDF and tried to import but get "Import failed" error.

Smanar, 2024-09-20:
HA don't support yet the new field state/flow.
You need to take a look directly in the API, using Phoscon/help/API Information/sensor, you need to see a ZHAWater created for this device with somes options, like the state/flow.

But you have installed too the `state_flow_item.json` file ?

korkmazk, 2024-09-23:
@Smanar Yes i installed both and found it:
`
    "138": {
        "config": {
            "battery": 0,
            "on": true,
            "reachable": true
        },
        "etag": "af954508c89a4aec674axxxxxxxxxxxx",
        "lastannounced": "2024-09-22T13:43:28Z",
        "lastseen": "2024-09-23T12:26Z",
        "manufacturername": "SONOFF",
        "modelid": "SWV",
        "name": "Water 138",
        "state": {
            "flow": 0,
            "lastupdated": "2024-09-23T12:05:54.285",
            "water": true
        },
        "swversion": "1.0.3",
        "type": "ZHAWater",
        "uniqueid": "xx:xx:xx:xx:xx:xx:xx:xx-xx-xxxx"
    }
}
`

I still did not use it with water on because I need to install the drippers in the garden.

> I have Phoscon/deCONZ installed as a Homeassistant Add-on. I was attempting to do this by using the Device integration function in deCONZ UI. I created a text file with the DDF and tried to import [...]

I installed it via SSH. For SSH I used "Advanced SSH & Web Terminal" add-on (disable Protection mode)
Via docker enter de deconz container:
- docker exec -it addon_core_deconz bash
- cd /user/share/deCONZ/devices/sonoff
- I'm not that familiar with SSH within HA so I used a url-shortener to the raw files https://github.com/Smanar/deconz-rest-plugin/commit/91d5eb5f4bdf4fde70e4c3bd53856a118410c004 and used wget with the short url to download the files directly into the correct directoies
- for the general file is placed it into: /user/share/deCONZ/devices/generic/items/ en named the file state_flow_item.json

Note that this isn't a persistent solution! An update or full reboot will probably remove the files.

Smanar, 2024-09-24:
Yes, there is a persistent folder for HA, but you can use only the DDF, not the second file needed to have all other options.

`/data/.local/share/dresden-elektronik/deCONZ/devices/` hidden folder

https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839/20

PR created https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7940

felker, 2024-10-17:
Does #7940 supersede this PR / should it be closed? Is the Sonoff SWF supported in the next deCONZ release?

Smanar, 2024-10-17:
I think yes, all last changes was in the PR, so the next deconz version will use them too.

korkmazk, 2024-10-17:
Yes it is, check out the Beta version released last week: https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.29.0-beta

I did not install this beta version and waiting for the stable release.

closed 2024-11-16 by github-actions[bot]

Nazze, 2025-02-03:
Hi! I need this device as well. Does anyone know, when this is released as stable? This page says September: https://github.com/dresden-elektronik/deconz-rest-plugin .
But seriously Sept-2025?

Smanar, 2025-02-03:
I haven't read all the post, but the device already have his DDF https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/sonoff/swv_smart_water_valve.json

Nazze, 2025-02-03:
Thanks for your reply. I am running deconz inside HA (as an add-on). Even getting into the docker container to update json files is a pain. That's why I am wondering, when the stable version is available.

Smanar, 2025-03-20:
Even with some more time, the battery sensor stay at 0 % ?
The DDF seem fine, perhaps with using the attribute 0x0020 instead of the 0x0021 ?

Where are you checking value ?
Can you try with Phoscon/help/API Information, to check for state/water and state/flow

craggyh, 2025-03-21:
Yes, even after 48h the battery sensors stay at 0%. I tried with 0x0020 instead but stayed 0.

Here is the output from API information while the valve is On with water flowing through it.
{
    "capabilities": {
        "sleeper": true
    },
    "config": {
        "battery": 0,
        "offset": 0,
        "on": true,
        "reachable": true
    },
    "ddf_hash": null,
    "ddf_policy": "latest_prefer_stable",
    "etag": "8962ca470254e923d03efde8959abfee",
    "lastannounced": null,
    "lastseen": "2025-03-21T15:01Z",
    "manufacturername": "SONOFF",
    "modelid": "SWV",
    "name": "Sonoff SWV 2",
    "nwkaddress": 18482,
    "state": {
        "flow": 0,
        "lastupdated": "2025-03-21T14:59:04.044",
        "water": false
    },
    "swversion": "1.0.3",
    "type": "ZHAWater",
    "uniqueid": "0c:ef:f6:ff:fe:ae:79:96-01-0404"
}

Smanar, 2025-03-22:
>I tried with 0x0020 instead but stayed 0.

You need to make 4 changes

```
        {
          "name": "config/battery",
          "parse": {
            "at": "0x0020",
            "cl": "0x0001",
            "ep": 1,
           "eval": "const vmin = 20; const vmax = 30; var bat = Attr.val; if (bat > vmax) { bat = vmax; } else if (bat < vmin) { bat = vmin; } bat = ((bat - vmin) / (vmax - vmin)) * 100; if (bat > 100) { bat = 100; } else if (bat <= 0)  { bat = 1; } Item.val = bat;"
          },
          "read": {
            "fn": "zcl:attr",
            "ep": 1,
            "cl": "0x0001",
            "at": "0x0020"
          },
          "refresh.interval": 4000,
          "awake": true
        },
```

and

```
      "report": [
        {
          "at": "0x0020",
          "dt": "0x20",
          "min": 60,
          "max": 3600,
          "change": "0x00000001"
        }
```

for the issue on flow, can you take a look directly in the GUI ? Cluster 0x0404 / Attribute 0x0000

craggyh, 2025-03-23:
I made the changes to config/battery and now it seems to be reporting batttery state correctly. :-)

Where does the "report" section go in the file? I don't have that in my DDF and I tried to add it in a few places as a guess but it results in a broken DDF. Not sure if I still need it though.

For the flow I checked the GUI and it shows the following:

[image]

Also, the water sensor state still doesn't reflect correct state of water detected, it just stays as Dry.

Smanar, 2025-03-23:
To make bind/report > https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet#bindings-panel
But easier to use copy/paste and text editor.

So yes, you have a value in the GUI, and still 0 in the API ? (reading the value in the GUI need to force a value update in the API, if it's that, the problem is the bind/report)

The code used by defaut for flow (hidden in the DDF) is

```
{
  "schema": "resourceitem1.schema.json",
  "id": "state/flow",
  "datatype": "Int16",
  "access": "R",
  "public": true,
  "description": "The current relative flow in m3/h.",
  "parse": {
    "at": "0x0000",
    "cl": "0x0404",
    "ep": 0,
    "eval": "Item.val = Attr.val / 10 + R.item('config/offset').val",
    "fn": "zcl:attr"
  },
  "default": 0
}
```

So 3/10 = 0, it's for that you have a zero value.
From the zigbee doc it's fine
```
i16MeasuredValue is a mandatory attribute corresponding to 10 times the
measured flow, in units of cubic metres per hour (m 3/h). The range of possible
values is 0x0000 (0 m3 /h) to 0xFFFE (representing 6553.4 m 3/h). The value
0xFFFF is used to indicate that the measurement was invalid. In practice, the
stored value is limited within the range i16MinMeasuredValue to
i16MaxMeasuredValue (see below)
```

You don't have valva caracteristic ? for it's maximum fflow ?
To know if the defaut value to low is only for this device or for all device.

Smanar, 2025-03-24:
I have looked better your capture, you have a vaue "3" and a max value 27.
So it's normal value.
The problem is 27 give the value 2, but 3 give the value 0, the code divide the value by 10.

I think I will make a PR to have the value *1000.

You are not using dockers ? or HA ?
If you have found the  state_flow_item.json file try to edit this line 

`"eval": "Item.val = Attr.val / 10 + R.item('config/offset').val",`

by

`"eval": "Item.val = Attr.val *100 + R.item('config/offset').val",`

Then restart deconz, you will have the value 300.

Nazze, 2025-04-03:
> Thanks for your reply. I am running deconz inside HA (as an add-on). Even getting into the docker container to update json files is a pain. That's why I am wondering, when the stable version is available.

Hi

does anyone know when the stable version is coming out? The watering season starts and I would like to use HA for this together with the Sonoff SWV.

felker, 2025-05-06:
> I think I will make a PR to have the value *1000.

@Smanar did you ever modify that value in a separate PR? Or should I open one?

Smanar, 2025-05-06:
Nope, have done nothing.
You have the same issue ? You are talking about the too low level for flow value, to use other value than m3/h ?

But haven't asked wich one will be better , m3/s or dm3/h ...

felker, 2025-05-12:
yes I can give both a try in ~2 weeks if no one else gets around to it first!

molnarti, 2025-06-11:
How is the support for this device currently? I need a new irrigation switch as my manual one died. I already got the Lidl irrigation switch https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4958 but I never got that work in Deconz. 
So I'd like to confirm this one works reliably before getting it.