dresden-elektronik/deconz-rest-plugin #7902: _TZE284_aao3yzhs Soil Sensor

emlnpro, 2024-08-25:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

_TZE284_aao3yzhs

### Manufacturer

_TZE284_aao3yzhs

### Model identifier

TS0601

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

ZHA request

https://github.com/zigpy/zha-device-handlers/issues/3305

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Smanar, 2024-08-25:
Hello, your device is just a clone of thoses one https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZE200_myd45weu_soil_sensor.json
Try adding your DDF to this list, you need to add Manufacture Name AND model id, even they are same.

emlnpro, 2024-08-29:
> Hello, your device is just a clone of thoses one https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZE200_myd45weu_soil_sensor.json Try adding your DDF to this list, you need to add Manufacture Name AND model id, even they are same.

Not sure what I need to change, onyl these lines ?
```
  "manufacturername": ["_TZE200_myd45weu", "_TZE200_9cqcpkgb", "_TZE200_ga1maeof", "_TZE204_myd45weu"],
  "modelid": ["TS0601", "TS0601", "TS0601", "TS0601"],
```

Smanar, 2024-08-29:
```
  "manufacturername": ["_TZE200_myd45weu", "_TZE200_9cqcpkgb", "_TZE200_ga1maeof", "_TZE204_myd45weu", "_TZE284_aao3yzhs"],
  "modelid": ["TS0601", "TS0601", "TS0601", "TS0601", "TS0601"],
```

emlnpro, 2024-08-29:
Not sure I can do that since I use HA deconz container, will try :)

Smanar, 2024-08-29:
On HA container, you can create a new DDF (a complete file) in the hidden folder `/data/.local/share/dresden-elektronik/deCONZ/devices/`

Long story > https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

emlnpro, 2024-08-30:
> On HA container, you can create a new DDF (a complete file) in the hidden folder `/data/.local/share/dresden-elektronik/deCONZ/devices/` it's a persistent folder.
> 
> Long story > https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

Thank you very much for the docs.

Deconz addon was updated today on hassio (finally !) and now I have upload DDF function in Phoscon, and I discovered the community store which is nice !!
So I ended up searching on this store, finding :
https://deconz-community.github.io/ddf-tools/#/store/bundle/21b7cc5dc3768bd906f6ff3e4498129ae0d5f2c593fecc1db4bdafdfc665638c

and edited with the online editor the json file to include my new ref.

It is almost working, moisture seems accurate but temperature is not, I got 260°C, it's 22.6 °C inside my home so not even a x10 error unfortunately.

Thank you again, so interesting to learn we can access ddf file inside container !

Smanar, 2024-08-31:
Ha yes so just a /10 willl by not enought 4°C error is a lot for use a offset corrector.
But from this post your device use the dpid 5 > https://github.com/Koenkk/zigbee2mqtt/issues/23455

Can make test to see if it's not just because the device is not precise at all ?

closed 2024-09-29 by github-actions[bot]

PolestarWx, 2024-09-30:
Still need to be integrated in ddf
For now I manually import it

Smanar, 2024-09-30:
There is no DDF planed.
Can you submit a PR, or want I make it ? But I haven't the final tested DDF

PolestarWx, 2024-09-30:
To be honest I don't know how to do it, I used https://deconz-community.github.io/ddf-tools/#/store/bundle/21b7cc5dc3768bd906f6ff3e4498129ae0d5f2c593fecc1db4bdafdfc665638c and edited the file inside the web interface, then downloaded it.
Temp need /10 conversion.

Zehir, 2024-09-30:
> To be honest I don't know how to do it, I used https://deconz-community.github.io/ddf-tools/#/store/bundle/21b7cc5dc3768bd906f6ff3e4498129ae0d5f2c593fecc1db4bdafdfc665638c and edited the file inside the web interface, thank to your help, then downloaded it.
> Temp need /10 conversion.

There is the doc to create a new ddf bundle : 
https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/ddf/#create-a-ddf-bundle

But for this case the better way is to do a pull request on github

m4k2k, 2024-12-01:
Hello all,
just received the device and saw its not yet supported. Based on above information I created a DDF.
Since I didn't see any PR, I did one: https://github.com/dresden-elektronik/deconz-rest-plugin/pull/8050
@Smanar maybe you could have a review :)
Thank you in advance

Smanar, 2024-12-03:
Thx to you ^^.
Don't woprry,the DDF is valided and you have tested the device, I think the PR will be fine.

Mini1954, 2025-09-06:
Hallo,

I need support.

My hardware: Raspberry pi 3B, Conbee 2 Stack an USB-Verlängerung. Firmware 26780700, Gatway Version: 2.30.2., Openhab 2.5

To run a soilsensor _TS0601, _TZE284_aao3yzhs, I installed the DDF for _TS0601, _TZE284_aao3yzhs.
The sensor connected immidiatly, all dates are shown Phoscon App and are updated regularly.

[image]

In phoscon app api-informations both sensors (temp and humidity) appear:

`{
    "capabilities": {
        "sleeper": true
    },
    "config": {
        "battery": 80,
        "offset": 0,
        "on": true,
        "reachable": true
    },
    "etag": "5e105314015bcb6b3aad216a04656b30",
    "lastannounced": null,
    "lastseen": "2025-09-05T15:11Z",
    "manufacturername": "_TZE284_aao3yzhs",
    "modelid": "TS0601",
    "name": "168 Garten Bodenfeuchte",
    "nwkaddress": "0x8363",
    "state": {
        "lastupdated": "2025-09-05T15:11:26.030",
        "lowbattery": false,
        "temperature": 2410
    },

    "swversion": "1.0.13",
    "type": "ZHATemperature",
    "uniqueid": "a4:c1:38:c5:68:e9:ec:82-01-0402"
}`

`{
    "capabilities": {
        "sleeper": true
    },
    "config": {
        "battery": 80,
        "offset": 0,
        "on": true,
        "reachable": true
    },
    "etag": "ebd5bd319dbdf8ccff03b6db7733518e",
    "lastannounced": null,
    "lastseen": "2025-09-05T15:43Z",
    "manufacturername": "_TZE284_aao3yzhs",
    "modelid": "TS0601",
    "name": "168 Garten Bodenfeuchte",
    "nwkaddress": "0x8363",
    "state": {
        "lastupdated": "2025-09-05T15:43:28.891",
        "lowbattery": false,
        "moisture": 500
    },
    "swversion": "1.0.13",
    "type": "ZHAMoisture",
    "uniqueid": "a4:c1:38:c5:68:e9:ec:82-01-0408"
}`

Unfortunately, the humidity sensor is not listed among the available sensors for automations and is therefore not usable. The temperature sensor is included, but that is of no use to me.

The temperatursensor appears in the inbox of openhab automatically, the humidity sensor does not but can be added manually. It is shown as online but no measurements are transmitted.

Has anyone an idea whats wrong?
Thanks

What's wrong

Smanar, 2025-09-06:
Yes, there is no humidity sensor on this device (at least on the DDF).
This device device measure the ZHAMoisture ( = the ground humidity)

Mini1954, 2025-09-07:
Thanks for that. I  I phrased that incorrectly. I'm missing the values of ZHAMoisture (ground humidity).

Smanar, 2025-09-07:
But you have it on your post
```
    "state": {
        "lastupdated": "2025-09-05T15:43:28.891",
        "lowbattery": false,
        "moisture": 500
    },
```

IDK how work openHAB but it don't support yet ZHAMoisture ?

Mini1954, 2025-09-07:
yes. I can't use the ZHA-Moisture for "automation" because it is not listed.