dresden-elektronik/deconz-rest-plugin #7903: Sonoff SNZB-02D & SNZB-02P not pairable on 2.27.6 and ConBeeII 26780700

masterpwarz, 2024-08-25:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Hey,

I had massive problems adding the Sonoff SNZB-02D and SNZB-02P to my headless Raspberry Pi ConBee II system. Both adapters just entered pairing mode and stopped pairing after a few seconds, but they didn’t show up in Phoscon or deCONZ. deCONZ was on version 2.27.6 and the ConBee II firmware was at 26780700.

I had another configuration where they worked, so I downgraded to deCONZ 2.25.3 and firmware 26660700, where I was able to add both and they work fine now. Sadly, I have no logs or screenshots because it is headless.

### Steps to reproduce the behavior

Go into pairing mode in phoscon and try to add the Sensor. Its gonna blink 5 Times till it stopped and wasnt founded in phoscon.

### Expected behavior

that they would have paired

### Screenshots

sry

### Environment

- Host system: (Raspberry Pi / PC / NAS / MAC)
- Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / deconz-docker container / Windows / Virtual Machine)
- Firmware version: (26xxyy00)
- deCONZ version (not Home assistant Addon version!): (2.xx.yy)
- Device: (ConBee I / ConBee II / ConBee III / RaspBee I / RaspBee II)
- Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II/III
- Is there any other USB or serial devices connected to the host system? If so: Which?

Smanar, 2024-08-26:
Hello, perhaps something similar to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7881

What is the couple manufactureName/ModelID ?

rumtraubenuss, 2024-09-04:
Have the same issue with those sensors on deconz 2.28.1 and RaspBee II. Also wonder to get the modelID from a headless system.

mrstackit, 2024-09-21:
Hello,

I also have SNZB-02P and the same problem with conbee2 and 2.28.1. Is there a solution in sight? Unfortunately, the other issue was automatically closed due to inactivity.

Smanar, 2024-09-22:
Some have make a PR here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7934

mrstackit, 2024-09-22:
> Some have make a PR here #7934

Thanks, it works. I hadn't found it. Good thing it's linked here now.

c-hoeger, 2024-10-12:
I have the same problem, but the recent ddf doesn't solve it.

[image]

[image]

[image]

When I tried to load the recent DDF, I just see an error message:

[image]

Any idea what could cause the issue?

Smanar, 2024-10-12:
Are you sure using the last DDF ?
You have this part inside ? 

```
  "manufacturername": ["eWeLink","SONOFF","SONOFF","eWeLink"],
  "modelid": ["TH01","SNZB-02D","SNZB-02P","SNZB-02P"],
```

c-hoeger, 2024-10-13:
> Are you sure using the last DDF ? You have this part inside ?
> 
> ```
>   "manufacturername": ["eWeLink","SONOFF","SONOFF","eWeLink"],
>   "modelid": ["TH01","SNZB-02D","SNZB-02P","SNZB-02P"],
> ```

Yes, I have.

I am using https://raw.githubusercontent.com/dresden-elektronik/deconz-rest-plugin/refs/heads/master/devices/sonoff/snzb-02-multisensor.json

After some more adding and removing it showed up without showing temperature or humidity.
After some more adding and removing it now seems to work...

Smanar, 2024-10-13:
Deconz load DDF only at start, so need to restart deconz after the DDF update.

closed 2024-11-12 by github-actions[bot]