dresden-elektronik/deconz-rest-plugin #7906: Temp. Sensort Humidity Sensor Xiaomi Mi. (flashed from BT to Zigbee)

Obelixje, 2024-08-27:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Temp. humidity sensor Mi Zigbee

### Manufacturer

Xiaomi

### Model identifier

LYWSD03MMC-z

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Mimiix, 2024-08-27:
@Smanar can you help with a DDF?

Smanar, 2024-08-27:
It's not a clone of this one https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7327 ?
What is the difference with the previous one ? Same DDF can't be used ?

Obelixje, 2024-09-02:
@Smanar 

Do I need to change anything in that DDF?
Or just upload it to DeConz?

caspermet, 2024-09-20:
Is it any update on this topic? I have same issue.

Smanar, 2024-09-20:
Pls try the previous procedure, if it work I can make a PR to add it.

caspermet, 2024-09-20:
Hello Smanar, 

Sorry I am new to this topic, I have trouble modifying DDF. What is best practice? I tried this tutorial https://www.youtube.com/watch?v=rgTQpOVg8w0 but I wasn't able to edit DDF. 

And I think that LYWSD03MMC and LYWSD03MMC-z are the same models.

Smanar, 2024-09-20:
>And I think that LYWSD03MMC and LYWSD03MMC-z are the same models

For me too, but need to be sure.

What is you OS ? On normal OS you can just edit the actual file `devices\xiaomi\xiaomi_lywsd03mmc.json` if you have HA or other dockers, you need to create a new DDF in the persistent folder.

caspermet, 2024-09-22:
I am using HA in Docker, but for a quick test, I plugged the ConBee II stick into my Windows machine. I added xiaomi_lywsd03mmc.json file into devices folder, and after a reboot, it seems that the deCONZ app is now correctly reading data from the thermostat:
[image]

But I still don't see the thermostat in my Phoscon App.

Smanar, 2024-09-22:
Ha yes, the DDF is selected, gold status, all seem fine.

Just one thing, the device was included before or not ? Deconz create entry in the database only if the gateway is in "permit join", so if you have nothing you can need to re-include the device.

caspermet, 2024-09-22:
Nice! I finaly see my devices in Phoscon app. Thank you verry much! Do you have plans to create a PR to add the LYWSD03MMC-z model?

Smanar, 2024-09-22:
Done https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7939
Thx for tests.

closed 2024-09-23 by manup