dresden-elektronik/deconz-rest-plugin #7909: Extend & Improve support for BTicino/Legrand Shutter SW with level control - K4027C/L4027C/N4027C/NT4027C

CodeForLove83, 2024-09-02:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Hi,

I'm going to fix issues with cluster/attributes for NLV Legrand devices. 
For instance the attribute `0xf002` is wrongly identified as `motor reversal` while is `calibration mode`.

[image]

I'm wondering if there's a way to extract the complete template DDF for `Window Covering Device` to avoid starting from scratch. If I just click on Edit DDF on the specific device it seems just a subset and does not include the whole Cluster Info, commands, etc.

[image]

Thanks

### Steps to reproduce the behavior

Read Cluster Info

### Expected behavior

Report correct cluster/attribute

### Environment

- Running method: deconz-docker container
- Firmware version: (26530900)
- deCONZ version: (2.28.1)
- Device: (ConBee III)
- Do you use an USB extension cable: (yes)

Smanar, 2024-09-02:
> For instance the attribute 0xf002 is wrongly identified as motor reversal while is calibration mode.

This is for tuya, not Legrand, can add a mfcode in the xml file to disable this part for Legrand device, but If you ask for attribute, I think thoses ones will be grayed ?
For Legrand it's not the attribute 0x0017 ?

>I'm wondering if there's a way to extract the complete template DDF for Window Covering Device to avoid starting from scratch. If I just click on Edit DDF on the specific device it seems just a subset and does not include the whole Cluster Info, commands, etc.

You have done exactly what you need to do, you can complete it using the GUI, but better to copy/paste that you have in the "preview" tab and use a text editor.

CodeForLove83, 2024-09-02:
Hi @Smanar 

Yes, indeed I was working on the XML file. I moved today to deconz from z2m and I'm trying to get used to it. 

I added the following code to handle Legrand Window Covering (NLV devices).

```
    <cluster id="0x0102" name="Window Covering" mfcode="1021">
      <description>The window covering cluster provides an interface for controlling and adjusting automatic window coverings such as drapery motors, automatic shades, and blinds.</description>
      <server>
        <attribute-set id="0x0000" description="Window Covering Information">
          <attribute id="0x0000" name="Window Covering Type" type="enum8" default="0x00" access="r" required="m">
            <value name="Rollershade" value="0"></value>
            <value name="Rollershade - 2 Motor" value="1"></value>
            <value name="Rollershade – Exterior" value="2"></value>
            <value name="Rollershade - Exterior - 2 Motor" value="3"></value>
            <value name="Drapery" value="4"></value>
            <value name="Awning" value="5"></value>
            <value name="Shutter" value="6"></value>
            <value name="Tilt Blind - Tilt Only" value="7"></value>
            <value name="Tilt Blind - Lift and Tilt" value="8"></value>
            <value name="Projector Screen" value="9"></value>
          </attribute>
          <attribute id="0x0007" name="Config / Status" type="bmp8" default="0x03" access="r" required="o">
            <value name="Operational" value="0"></value>
            <value name="Online" value="1"></value>
            <value name="Commands Reversed" value="2"></value>
            <value name="Lift control is Closed Loop" value="3"></value>
            <value name="Tilt control is Closed Loop" value="4"></value>
            <value name="Lift: Encoder Controlled" value="5"></value>
            <value name="Tilt: Encoder Controlled" value="6"></value>
          </attribute>
          <attribute id="0x0008" name="Current Position Lift Percentage" type="u8" default="0x00" access="r" required="o" range="0,0x64"></attribute>
          <attribute id="0x0009" name="Current Position Tilt Percentage" type="u8" default="0x00" access="r" required="o" range="0,0x64"></attribute>
          <attribute id="0xf003" name="Target Position Tilt Percentage" type="u8" default="0x00" access="r" required="o" range="0,0x64"></attribute>
        </attribute-set>
        <attribute-set id="0x0010" description="Window Covering Settings">
          <attribute id="0x0010" name="Installed Open Limit – Lift" type="u16" default="0x0000" access="r" required="o"></attribute>
          <attribute id="0x0011" name="Installed Closed Limit – Lift" type="u16" default="0xffff" access="r" required="o"></attribute>
          <attribute id="0x0017" name="Mode" type="bmp8" default="0x0000" access="rw" required="o">
            <value name="Reversed" value="0"></value>
            <value name="Calibration Mode" value="1"></value>
            <value name="Maintenance Mode" value="2"></value>
            <value name="LED feedback" value="3"></value>
          </attribute>
        </attribute-set>
        <attribute-set id="0xf000" description="Legrand specific Window Covering Setting" mfcode="1021">
          <attribute id="0xf001" name="Step Position Lift" type="u16" default="0x0000" access="r" required="o"></attribute>
          <attribute id="0xf004" name="Step Position Tilt" type="u16" default="0x0000" access="r" required="o"></attribute>
          <attribute id="0xf002" name="Calibration Mode" type="enum8" default="0" required="m" access="rw">
            <value name="Classic (only nllv)" value="0"></value>
            <value name="Specific (only nllv)" value="1"></value>
            <value name="Up Down Stop" value="2"></value>
            <value name="Temporal" value="3"></value>
            <value name="Venetian" value="4"></value>
          </attribute>
        </attribute-set>
        <command id="0x00" dir="recv" name="Up / Open" required="m"></command>
        <command id="0x01" dir="recv" name="Down / Close" required="m"></command>
        <command id="0x02" dir="recv" name="Stop" required="m"></command>
        <command id="0x05" dir="recv" name="Go to Lift Percentage" required="o">
          <payload>
            <attribute id="0x0000" type="u8" name="Lift Percentage" required="m" default="0x00" range="0,0x64"></attribute>
          </payload>
        </command>
        <command id="0x08" dir="recv" name="Go to Tilt Percentage" required="o">
          <payload>
            <attribute id="0x0000" type="u8" name="Tilt Percentage" required="m" default="0x00" range="0,0x64"></attribute>
          </payload>
        </command>
      </server>
      <client>
      </client>
    </cluster>
```

With this implementation you can run different type of calibrations supported by Legrand/BTicino straight from deconz.

If you guys are interested I can write a specific XML file just for Legrand devices.

closed 2024-10-03 by github-actions[bot]