dresden-elektronik/deconz-rest-plugin #7911: eWelink Waterleakage sensor. Device found Deconz but not in Phoscon

Sjonn, 2024-09-04:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Ewelink Zigbee3.0 Watersensor Alarm Water Lek Detector 

### Manufacturer

eWelink

### Model identifier

SNZB-03

### Device type to add

Sensor

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

Basic
[image]

Identify
[image]

Power Configuration Cluster
[image]

IAS Zone
[image]

Poll Control
[image]

OTAU
[image]

### Further relevant clusters

### Power Configuration

### On/Off

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

SwoopX, 2024-09-04:
Hi there.

Unfortunately, the screenshots indicate that the clusters have not been read properly. [This page](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to...#read-device-clusters) explains how to read clusters. Please take new ones and update the issue accordingly. Thanks a lot!

Sjonn, 2024-09-04:
Hello SwoopX

Enclosed the updated screenshots.

Basic Cluster
[image]

Identify
[image]

Power Configuration
[image]

IAS Zone
[image]

Poll Control
[image]

OTAU
[image]

Hope that this time the screenshots are good.

berndthebread, 2024-12-09:
Hello Smanar, thanks for your quick reply. I tested the DDF and it works. Kind regards and have a nice day!

berndthebread, 2024-12-10:
Hello Smanar, if you could create a PR that would be great.

Smanar, 2024-12-10:
Done https://github.com/dresden-elektronik/deconz-rest-plugin/pull/8057

Sjonn, 2024-12-10:
Hello Smanar, thank you for reopening the issue and creating the DDF.

closed 2025-01-09 by github-actions[bot]

mattreim, 2025-08-06:
@Sjonn @berndthebread Could one of you please help us and upload a screenshot of your device's (eWelink/SNZB-03) basic cluster here. Press the read button and the sensor button until the list is either filled or grayed out.