dresden-elektronik/deconz-rest-plugin #7912: LIVARNO home Automatik-Verdunkelungsrollo

Smibser, 2024-09-04:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

LIVARNO home Automatik-Verdunkelungsrollo, »Zigbee Smart Home«

### Manufacturer

_TZB000_yqjaollc

### Model identifier

TS030F

### Device type to add

Other

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]
[image]

### Further relevant clusters

### Power Configuration
[image]

Smibser, 2024-09-04:
There was already a device request for this device in [this PR](https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7325)
Unfortunately they changed the chip as I found out in [this forum post](https://forum.phoscon.de/t/cant-pair-with-livarno-home-window-covering-device/5099)

Smanar, 2024-09-05:
Can try to add it on the DDF

```
{
  "schema": "devcap1.schema.json",
  "manufacturername": ["_TZB000_42ha4rsc", "_TZB000_yqjaollc"],
  "modelid": ["TS030F", "TS030F"],
  "vendor": "Livarno",
  "product": "Window blinds (TS030F)",
  "sleeper": true,
  "status": "Gold",
  "subdevices": [
    {
      "type": "$TYPE_WINDOW_COVERING_DEVICE",
      "restapi": "/lights",
      "uuid": [
        "$address.ext",
        "0x01"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "parse": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "state/lift"
        },
        {
          "name": "state/open"
        },
        {
          "name": "state/reachable"
        }
      ]
    },
    {
      "type": "$TYPE_BATTERY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0001"
      ],
      "fingerprint": {
        "profile": "0x0104",
        "device": "0x0402",
        "endpoint": "0x01",
        "in": [
          "0x0001",
          "0x0102"
        ]
      },
      "items": [
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion",
          "refresh.interval": 86400,
          "parse": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
          "read": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001"}
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/battery",
          "parse": {
            "cl": "0x0001",
            "at": "0x0021",
            "eval": "Item.val = Attr.val / 2"
          },
          "read": {
            "cl": "0x0001",
            "at": "0x0021"
          },
          "refresh.interval": 3600
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "dst.ep": 1,
      "cl": "0x0102",
      "report": [
        {
          "at": "0x0008",
          "dt": "0x20",
          "min": 1,
          "max": 300,
          "change": "0x00000001"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001",
      "report": [
        {
          "at": "0x0021",
          "dt": "0x20",
          "min": 300,
          "max": 600,
          "change": "0x01"
        }
      ]
    }
  ]
}
```

Can create a new one or edit the old one for test.

Smibser, 2024-09-07:
@Smanar sorry for the late response, I didn't get a notification.
When I try to open the DFF it says `Failed to open /home/pi/.local/share/dresden-elektronik/deCONZ/devices/ts030f.json`
Maybe I am doing something wrong. I did that:
1. Right click on the device and click 'Edit DFF'
2. File -> Save and saved it to the devices folder
3. I edit the created json file with your json. Compressed without line breaks
4. I open 'Edit DFF' again and try to File -> Open

Smanar, 2024-09-08:
Better to avoid the editor.
Just create a new file, or edit the existing one `devices\tuya\_TZB000_42ha4rsc_window_blinds.json`

Then restart deconz, deconz load DDF only at start.
Then re-include the device.

Smibser, 2024-09-08:
Thanks for the help. It works perfectly

Smanar, 2024-09-08:
Nice, thx for return, have added the device here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7919

closed 2024-09-23 by manup