dresden-elektronik/deconz-rest-plugin #7913: Aqara T2 Dual Relay Module

sch115, 2024-09-05:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

Hello to all!
I ran into a issue with energy consumption statistics. Voltage and watt numbers are OK but Total Consumption in kWh is 1000 times higher. I believe this issue bellow was based on the same problem as mentioned: https://github.com/Koenkk/zigbee2mqtt/issues/22148

The relay was updated to latest firmware: 0.0.0_0030

Everything else looks good to me.
Thank you for your help!

### Steps to reproduce the behavior

Pretty much out of scale from very beginning

### Expected behavior

Expecting correct consumption totals

### Environment

- Host system: Raspberry Pi
- Running method: Raspbian
- Firmware version: 26780700
- deCONZ version: 2.28.1
- Device: ConBee II
- Do you use an USB extension cable: yes

SwoopX, 2024-09-05:
Hi,

Thanks for reporting. Apparently, the device already reports in kWh. The fix is rather easy as we just need to remove the value multiplication within the DDF.

I'll prepare a quick PR for that.

SwoopX, 2024-09-06:
@sch115 I have to say I'm getting doubts that your assumption is correct. Just checked the socket outlet I recently got and it reports the below:

```
22:55:24:555 	03 Device temperature 29 °C
22:55:24:555 	05 RSSI dB (?) 38 (0x0026)
22:55:24:555 	09 unknown 259 (0x0103)
22:55:24:555 	0a Parent NWK 0 (0x0000)
22:55:24:555 	0c unknown 10 (0x0A)
22:55:24:555 	0d firmware 0.0.0_029 (0x0000001D)
22:55:24:556 	11 unknown 1 (0x00000001)
22:55:24:556 	64 on/off 0
22:55:24:556 	65 on/off 0
22:55:24:556 	95 consumption 0,028400 (28)
22:55:24:556 	96 voltage 2346,399902 (235)
22:55:24:556 	98 power 0,000000 (0)
22:55:24:556 	97 current 0,000000 (0)
22:55:24:556 	9a unknown 0 (0x00)
```

The reported consumption value needs the multiplication by 1000 here to expose the correct value of 28 Wh (Sonos One for a couple of hours). The relay will most likely act no different 🤔

sch115, 2024-09-07:
> @sch115 I have to say I'm getting doubts that your assumption is correct. Just checked the socket outlet I recently got and it reports the below:
>
> ```
> 22:55:24:555 	03 Device temperature 29 °C [...]

Yeah I see your point. Definitely weird. I tried to re-pair the relay couple times but no difference. I even find this log in Homebridge:

[07/09/2024, 09:55:27] [deCONZ] On Off switch 13 History: warning: RangeError: The value of "value" is out of range. It must be >= 0 and <= 65535. Received 2334000
    at checkInt (node:internal/buffer:74:11)
    at writeU_Int16LE (node:internal/buffer:724:3)
    at Buffer.writeUInt16LE (node:internal/buffer:732:10)
    at TotalConsumptionValue.writeEntry (file:///var/lib/homebridge/node_modules/homebridge-deconz/node_modules/homebridge-lib/lib/ServiceDelegate/History.js:78:47)
    at History.entryToBuffer (file:///var/lib/homebridge/node_modules/homebridge-deconz/node_modules/homebridge-lib/lib/ServiceDelegate/History.js:643:21)
    at History._onGetEntries (file:///var/lib/homebridge/node_modules/homebridge-deconz/node_modules/homebridge-lib/lib/ServiceDelegate/History.js:729:24)
    at CharacteristicDelegate._onGet (file:///var/lib/homebridge/node_modules/homebridge-deconz/node_modules/homebridge-lib/lib/CharacteristicDelegate.js:362:26)
    at Characteristic.emit (node:events:519:28)
    at /var/lib/homebridge/node_modules/homebridge/node_modules/hap-nodejs/src/lib/Characteristic.ts:2468:14
    at new Promise (<anonymous>)

I also have Aqara T1 Single Switch Module in my setup and totals there are OK.

closed 2024-09-09 by manup