dresden-elektronik/deconz-rest-plugin #7917: Royal Gardineer NX-6476-919  Water Valve with Plug as Gateway

cartman10, 2024-09-07:
### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for my device

### Product name

Royal Gardineer NX-6476-919

### Manufacturer

dresden elektronik

### Model identifier

ConBee ll

### Device type to add

Other

### Node info

[image]

### Endpoints and clusters

[image]

### Basic

[image]

### Further relevant clusters

### Power Configuration

### On/Off
[image]
[image]

### Level Control

### Color Control

### Thermostat

### Simple Metering

### Electrical Measurement

### Any other cluster of relevance/interest

Mimiix, 2024-09-07:
Hi, 

This is the Conbee stick, not your device.

Please provide the device clusters

cartman10, 2024-09-07:
Hi, obviouslsly the deivce didn't pair...  how do I pair it?

Mimiix, 2024-09-07:
Just like you would with other devices. It should show up in deconz with a hex name and then you can make the correct screens.

cartman10, 2024-09-29:
not a zigbee device

closed 2024-09-29 by cartman10