dresden-elektronik/deconz-rest-plugin #7918: Battery not reported on DDF for Tuya temperature and humidity sensor (_TZ3210_ncw88jfq) #7908

RneeJ, 2024-09-08:
### Does the issue really belong here?

- [X] I definitively want to report a bug within deCONZ or its REST-API

### Is there already an existing issue for this?

- [X] I have searched the existing issues and there is none for the bug at hand

### Describe the bug

@Smanar see #7884 

In the .json i deliverd for this device the battery was reported correctly, now it is missing.

### Steps to reproduce the behavior

NA

### Expected behavior

report state of battery as before

### Screenshots

[image]

For before (in original .json) see #7884

### Environment

na

### deCONZ Logs

na

### Additional context

na

Smanar, 2024-09-08:
Hello, but on your capture here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/7884#issuecomment-2325166813
The battery is here too ? We can see config/battery.

So the problem is from Phoscon ?

RneeJ, 2024-09-08:
Hoi Smanar. 

Before the battery was reported correctly (at date 30.08.2024):  
[image]
then i send in the my working .json.

You made some changes to the .json, it was pulled in and a new DDF was created.

I downloaded the new DDF to my " production" machine and now the battery status is gone.

(But a version is shown correctly)

Mimiix, 2024-09-08:
Is it still in the api?

RneeJ, 2024-09-08:
Hi all,

I put this .json in my $HOME/.local/share/deCONZ/devices/ :
[_TZ3210_ncw88jfq_temp-hum-sensor-RNEEJ.json](https://github.com/user-attachments/files/16922762/_TZ3210_ncw88jfq_temp-hum-sensor-RNEEJ.json)

Now i have the battery status back again.

(Also in Domoticz)

Smanar, 2024-09-08:
But we can see the battery level on the API 
[image]

So if it's not visible in phoscon, the problem is from it.

RneeJ, 2024-09-08:
@Smanar , i think there is some confusion here, i apologize.
I have a seperate full size "test" and a headless Pi "production" machine. 
I used my " test"  machine for setting up Zigbee/Phoscon/deCONZ. 
I don't want to mess up Domoticz ;-) 

But the problem is the same. 
This new DDF does not show battery for this device :
[image]

This was Domoticz with the installed DDF :
[image]

This is now Domoticz with the mentioned [_TZ3210_ncw88jfq_temp-hum-sensor-RNEEJ.json](https://github.com/user-attachments/files/16922762/_TZ3210_ncw88jfq_temp-hum-sensor-RNEEJ.json) installed : 
[image]

So the DDF Bundle is wrong , because it does not show the battery.

Smanar, 2024-09-16:
Ha ok.
But I have checked the DDF https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/devices/tuya/_TZ3210_ncw88jfq_temp-hum-sensor.json
It's the good one `"uuid": "f490468c-83c1-4367-9168-4043f59e23cc",`

And I m seing the line for battery (even lighter than the working version)
```
        {
          "name": "config/battery"
        },
```

So IDK if it's missing or not updated (a null value for exemple)
Can you take a look on your "not working" version how it look direclty in the API ? (the json for the ZHATemperature for exemple) from Domoticz (on the Front end) or Phoscon (Help/API Information/sensor).

The code is lighter, there is only 1 line, but it need to work too.

RneeJ, 2024-09-29:
@Smanar , yours isn't working.

I did a complete new install  on a complete new machine (Rasp 5). (Domoticz ; it took ages...)
So a new fresh download of deCONZ etc, not fiddled anything by me.
it included "uuid": "f490468c-83c1-4367-9168-4043f59e23cc"

_Battery was NOT reported, not in the deCONZ app nor in Domoticz._

I installed my json and battery was reported correct :

 "schema": "devcap1.schema.json",
  "manufacturername": "_TZ3210_ncw88jfq",
  "modelid": "TS0201",
  "vendor": "Tuya",
  "product": "Temperature and humidity sensor",
  "sleeper": true,
  "status": "Gold",
  "path": "/devices/_TZ3210_ncw88jfq_temp-hum-sensor.json",
  "subdevices": [
    {
      "type": "$TYPE_TEMPERATURE_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0402"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "refresh.interval": 7265,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          },
          "default": 0
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "config/tuya_unlock",
          "public": false,
          "refresh.interval": 86400,
          "default": false
        },
        {
          "name": "state/lastupdated"
        },
        {
          "name": "state/temperature",
          "awake": true,
          "default": 0
        }
      ]
    },
    {
      "type": "$TYPE_HUMIDITY_SENSOR",
      "restapi": "/sensors",
      "uuid": [
        "$address.ext",
        "0x01",
        "0x0405"
      ],
      "items": [
        {
          "name": "attr/id"
        },
        {
          "name": "attr/lastannounced"
        },
        {
          "name": "attr/lastseen"
        },
        {
          "name": "attr/manufacturername"
        },
        {
          "name": "attr/modelid"
        },
        {
          "name": "attr/name"
        },
        {
          "name": "attr/swversion"
        },
        {
          "name": "attr/type"
        },
        {
          "name": "attr/uniqueid"
        },
        {
          "name": "config/battery",
          "refresh.interval": 7265,
          "read": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "fn": "zcl:attr"
          },
          "parse": {
            "at": "0x0021",
            "cl": "0x0001",
            "ep": 1,
            "eval": "Item.val = Attr.val / 2"
          },
          "default": 0
        },
        {
          "name": "config/offset",
          "default": 0
        },
        {
          "name": "config/on"
        },
        {
          "name": "config/reachable"
        },
        {
          "name": "state/humidity",
          "awake": true,
          "parse": {
            "at": "0x0000",
            "cl": "0x0405",
            "ep": 0,
            "eval": "Item.val = Attr.val * 10 + R.item('config/offset').val",
            "fn": "zcl:attr"
          },
          "default": 0
        },
        {
          "name": "state/lastupdated"
        }
      ]
    }
  ],
  "bindings": [
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0001"
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0402",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x29",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    },
    {
      "bind": "unicast",
      "src.ep": 1,
      "cl": "0x0405",
      "report": [
        {
          "at": "0x0000",
          "dt": "0x21",
          "min": 60,
          "max": 300,
          "change": "0x00000064"
        }
      ]
    }
  ]
}

RneeJ, 2024-09-29:
And my presence sensor 
ts0202_presence_sensor.json with "uuid":"13e89e44-1f42-4e5d-8d29-995d394451ac", has exactly the same problem. 

It does not show battery status.

This (by me) changed json does :
{
   "schema":"devcap1.schema.json",
   "uuid":"13e89e44-1f42-4e5d-8d29-995d394451ac",
   "manufacturername": [
      "_TYZB01_tv3wxhcz",
      "_TYZB01_hqbdru35",
      "_TZ3000_tiwq83wk",
      "_TZ3000_ykwcwxmz",
      "_TZ3000_hgu1dlak",
      "_TZ3000_hktqahrq",
      "_TZ3000_kmh5qpmb",
      "_TZ3000_jmrgyl7o",
      "_TZ3000_msl6wxk9",
      "_TZ3000_otvn3lne",
      "_TZ3000_6ygjfyll",
      "_TZ3040_6ygjfyll",
      "_TZ3000_mcxw5ehu",
      "_TYZB01_ef5xlc9q",
      "_TYZB01_vwqnz1sn",
      "_TYZB01_2b8f6cio",
      "_TZE200_bq5c8xfe",
      "_TYZB01_dl7cejts",
      "_TYZB01_qjqgmqxr",
      "_TZ3000_mmtwjmaq",
      "_TYZB01_zwvaj5wy",
      "_TZ3000_bsvqrxru",
      "_TZ3000_mg4dy6z6",
      "_TZ3040_bb6xaihh",
      "_TZ3000_lf56vpxj"
   ],
   "modelid":[
      "TS0202",
      "TS0202",
      "TS0202",
      "TS0202",
      "TS0202",
      "TS0202",
      "TS0202",
      "TS0202",
      "TS0202",
      "TS0202",
      "TS0202",
      "TS0202",
      "TS0202",
      "TS0202",
      "TS0202",
      "TS0202",
      "TS0202",
      "TS0202",
      "TS0202",
      "TS0202",
      "TS0202",
      "TS0202",
      "TS0202",
      "TS0202",
      "TS0202"
   ],
   "vendor": "Tuya",
   "product":"Presence sensor (TS0202)",
   "sleeper":true,
   "status":"Gold",
   "subdevices":[
      {
         "type":"$TYPE_PRESENCE_SENSOR",
         "restapi":"/sensors",
         "uuid":[
            "$address.ext",
            "0x01",
            "0x0500"
         ],
         "fingerprint": {
           "profile": "0x0104",
           "device": "0x0402",
           "endpoint": "0x01",
           "in": [
             "0x0000",
             "0x0001",
             "0x0500"
           ]
         },
         "items":[
            {
               "name":"attr/id"
            },
            {
               "name":"attr/lastannounced"
            },
            {
               "name":"attr/lastseen"
            },
            {
               "name":"attr/manufacturername"
            },
            {
               "name":"attr/modelid"
            },
            {
               "name":"attr/name"
            },
            {
              "name": "attr/swversion",                   
              "parse": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001", "script": "tuya_swversion.js"},
              "read": {"fn": "zcl:attr", "ep": 1, "cl": "0x0000", "at": "0x0001"}
            },
            {
               "name":"attr/type"
            },
            {
               "name":"attr/uniqueid"
            },
            {
               "name":"config/battery",
            "refresh.interval": 7265,
            "read":{
                "at":"0x0021",
                "cl":"0x0001",
                "ep":1,
                "fn":"zcl:attr"
             },
            "parse":{
                 "at":"0x0021",
                 "cl":"0x0001",
                 "ep":1,
                 "eval":"Item.val = Attr.val / 2"
             },
             "default":0 
            },
            {
               "name":"config/duration"
            },
            {
               "name":"config/enrolled",
               "public":false
            },
            {
               "name":"config/on"
            },
            {
               "name":"config/pending"
            },
            {
               "name":"config/reachable"
            },
            {
               "name":"state/lastupdated"
            },
            {
               "name":"state/lowbattery"
            },
            {
               "name":"state/presence",
               "awake":true,
               "parse":{
                  "fn":"ias:zonestatus",
                  "mask":"alarm1"
               }
            }
         ]
      }
   ],
   "bindings":[
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0500"
      },
      {
         "bind":"unicast",
         "src.ep":1,
         "cl":"0x0001",
         "report":[
            {
               "at":"0x0021",
               "dt":"0x20",
               "min":3600,
               "max":14400,
               "change":"0x00000001"
            }
         ]
      }
   ]
}

Smanar, 2024-09-29:
Ha yes, have checked the json

```
{
  "schema": "resourceitem1.schema.json",
  "id": "config/battery",
  "datatype": "UInt8",
  "access": "R",
  "public": true,
  "range": [
    0,
    100
  ],
  "description": "The current device battery level in 0&ndash;100&thinsp;%.",
  "default": 0
}
```

All is managed by legacy code, so can be buggy.

BTW do you want to make a PR to correct the 2 files or want I make it ?

RneeJ, 2024-09-29:
@Smanar thank you for your quick response.

please, you can make the PR. :+1:

Smanar, 2024-09-29:
I think this is fine ? https://github.com/dresden-elektronik/deconz-rest-plugin/pull/7951

RneeJ, 2024-09-29:
Oke , i will test them next week. Thank you!

closed 2024-09-30 by manup

RneeJ, 2024-10-05:
Can confirm this works as expected !

**In Phoscon App :** 
[image]

**also in Domoticz :** 
[image]

_Just one thing is not fully oke :_ 
**In Phoscon App** 
- Temperature is only in full digids : 21 deg 
- Humidity is in decimals : 54,1 % 

**In Domoticz however :** 
- Temperature is only in decimal : 21,3 deg 
- Humidity is in full digids : 54 % 
 
I mean the data is available so why are they not used in the applications ?

The ddf is oke, i can live with this , so if you want

Smanar, 2024-10-05:
I don't have impact on Phoscon, but on domoticz, the plugin use only the int value, it's perhaps because Domoticz use only int value for humidity (My other domoticz plugin do same), or a choice, temperature move a lot 12 °C is not same than 12.5 °C for regulation, for humidity 53 % or 53.9 %, I don't think it's  a big difference.